espressif/esp-idf #12922: gatt_server_service_table demo pairing

skanky-dev, 2024-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I've got a project successfully running using the "gatt_server_service_table demo" for the ESP32 C3. However there doesn't appear to be any support for Bluetooth pairing? Is it possible to pair to the ESP32 running this code to gain the additional security benefits of paired devices?

zhp0406, 2024-01-08:
@skanky-dev 
Yes, the[ GATT server service table](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server_service_table) demo does not support pairing. If you intend to leverage security features, please refer to the following example
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_server

skanky-dev, 2024-01-08:
Thanks very much for your answer

closed 2024-01-08 by skanky-dev