espressif/esp-idf #12923: ESP32S3 esp_ota_end() results in TG1WDT_SYS_RESET

rtborg, 2024-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am implementing firmware update on a custom board using ESP32S3 and USB HID. I am able to successfully download the firmware to the chip (confirmed with reading back the flash), but when I call `esp_ota_end()`, the chip reboots without printing a backtrace:

  ```
I (343392) ao_firmware_updater: Progress: 99.74%
  I (343732) ao_firmware_updater: Progress: 99.91%
  I (344012) ao_firmware_updater: Progress: 100.00%
  I (344012) ao_firmware_updater: Finished binary image write, validating...
  D (345012) esp_image: reading image header @ 0x110000
  D (345012) esp_image: image header: 0xe9 0x06 0x02 0x03 4037599c
  I (345012) esp_image: segment 0: paddr=00110020 vaddr=3c060020 size=20fbch (135100) map
  D (345012) esp_image: free data page_count 0x000001f7
  ESP-ROM:esp32s3-20210327
  Build:Mar 27 2021
  rst:0x8 (TG1WDT_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
  Saved PC:0x4037f94b
  0x4037f94b: panic_handler at /home/null/esp/esp-idf/components/esp_system/port/panic_handler.c:147 (discriminator 3)
  
  SPIWP:0xee
  mode:DIO, clock div:1
  load:0x3fce3810,len:0x1f84
  load:0x403c9700,len:0xd94
  load:0x403cc700,len:0x3194
  SHA-256 comparison failed:
  Calculated: 304a058dc63485772d4d2ceecf5007394c8dfbf8b004e575f7570f45121d3c95
  Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
  Attempting to boot anyway...
  entry 0x403c994c
  I (47) boot: ESP-IDF v5.0.4-265-gf333925342 2nd stage bootloader
```

Attaching a JTAG debugger shows where the crash happens:
[image]. 

 I am able to trace the code up to the loop in line 639 of [esp_image_format.c](https://github.com/espressif/esp-idf/blob/35556a6a858089f475f35cf7fefe4406fb1882c0/components/bootloader_support/src/esp_image_format.c#L602C3-L602C3). Reset happens during execution of said loop. With debugger attached, code ends in [panic_handlers.c](https://github.com/espressif/esp-idf/blob/35556a6a858089f475f35cf7fefe4406fb1882c0/components/esp_system/port/panic_handler.c#L147); otherwise the device reboots.

The IWDT timeout period is 1000 ms. The situation does not change when using the master branch as well.  
Attached is a boot log with DEBUG level.
[boot_log.txt](https://github.com/espressif/esp-idf/files/13818226/boot_log.txt)

kayzen9, 2024-01-03:
try reading flash from 0x110000 & compare with original OTA image. if the contents do not match then image was not write correctly. I was facing similar issue

rtborg, 2024-01-03:
I've done that using esptool.py, dumped flash and compared the data at 0x110000 with the binary I was sending, which is how I confirmed esp_ota_write does its job.

rtborg, 2024-01-04:
Issue fixed - I was doing `memcpy` to a string literal, which passed with no warnings. I rectified the offending `memcpy` and now `esp_ota_end` does not fail.

closed 2024-01-04 by rtborg