espressif/esp-idf #12924: SD card example with SPI has compilation errors

eismeraldo, 2024-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

The newest Version on GitHub

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I'm using an ESP32-S3-WROOM-1 N16R8, VSCode and PlatformIo. The SD card is
https://www.aliexpress.com/item/1005006028969168.html?spm=a2g0o.order_list.order_list_main.17.21ef1802D7InG1
The example "https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/sdspi/main/sd_card_example_main.c" compiles with errors.
[image]

[image]

[image]

[image]

The SD card is ok, it works fine with an Arduino.
Is there a working example for the processor mentioned above?

### What is the actual behavior?

compile errors

### Steps to reproduce.

1. Change the example to meet the Arduino Lib (setup and loop)
2. Change the GPIOs
#define PIN_NUM_MISO GPIO_NUM_15  // MISO (Grün)
#define PIN_NUM_MOSI GPIO_NUM_14  // MOSI (Blau)
#define PIN_NUM_CLK GPIO_NUM_16   // SCK  (Gelb)
#define PIN_NUM_CS GPIO_NUM_13    // SD_CS (Orange)

3. build
...

### Build or installation Logs.

```plain
Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)
------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.4.0+sha.516520f) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32-s3-devkitc-1\src\main.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:104:35: error: invalid conversion from 'int' to 'spi_host_device_t' [-fpermissive] 
     ret = spi_bus_initialize(host.slot, &bus_cfg, SDSPI_DEFAULT_DMA);
                              ~~~~~^~~~
In file included from C:/Users/adi/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/spi_master.h:12,
                 from C:/Users/adi/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/sdspi_host.h:14,
                 from C:/Users/adi/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/fatfs/vfs/esp_vfs_fat.h:20,
                 from src/main.cpp:16:
C:/Users/adi/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/spi_common.h:156:48: note:   initializing argument 1 of 'esp_err_t spi_bus_initialize(spi_host_device_t, const spi_bus_config_t*, spi_dma_chan_t)'
 esp_err_t spi_bus_initialize(spi_host_device_t host_id, const spi_bus_config_t *bus_config, spi_dma_chan_t dma_chan);
                              ~~~~~~~~~~~~~~~~~~^~~~~~~
src/main.cpp:114:32: error: invalid conversion from 'int' to 'spi_host_device_t' [-fpermissive] 
     slot_config.host_id = host.slot;
                           ~~~~~^~~~
src/main.cpp:171:11: error: 'esp_vfs_fat_sdcard_format' was not declared in this scope
     ret = esp_vfs_fat_sdcard_format(mount_point, card);
           ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:171:11: note: suggested alternative: 'esp_vfs_fat_sdcard_unmount'
     ret = esp_vfs_fat_sdcard_format(mount_point, card);
           ^~~~~~~~~~~~~~~~~~~~~~~~~
           esp_vfs_fat_sdcard_unmount
src/main.cpp:203:23: error: invalid conversion from 'int' to 'spi_host_device_t' [-fpermissive] 
     spi_bus_free(host.slot);
                  ~~~~~^~~~
In file included from C:/Users/adi/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/spi_master.h:12,
                 from C:/Users/adi/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/sdspi_host.h:14,
                 from C:/Users/adi/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/fatfs/vfs/esp_vfs_fat.h:20,
                 from src/main.cpp:16:
C:/Users/adi/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/spi_common.h:169:42: note:   initializing argument 1 of 'esp_err_t spi_bus_free(spi_host_device_t)'
 esp_err_t spi_bus_free(spi_host_device_t host_id);
                        ~~~~~~~~~~~~~~~~~~^~~~~~~
*** [.pio\build\esp32-s3-devkitc-1\src\main.cpp.o] Error 1
================================== [FAILED] Took 1.87 seconds ==================================
```

igrr, 2024-01-03:
The issue is that you are trying to compile the original C source code of the example as C++ (main.cpp). While often valid C code usually is also valid C++ code, this is not the case here, as C++ doesn't allow implicit casts from ints to enums.

You can either switch to using C (which might not be possible as you are using the Arduino framework) or modify the code to be valid C++, for example by adding a type cast to spi_host_device_t.

eismeraldo, 2024-01-03:
Thank you for the tips. The example from version 4.4 with the corresponding casts could be built without errors.

eismeraldo, 2024-01-03:
> The issue is that you are trying to compile the original C source code of the example as C++ (main.cpp). While valid C code usually is also valid C++ code, this is not the case here, as C++ doesn't allow implicit casts from ints to enums.
> 
> You can either switch to using C (which might not be possible as you are using the Arduino framework) or modify the code to be valid C++, for example by adding a type cast to spi_host_device_t.
> 
> One more note,
> 
> > IDF version.
> > The newest Version on GitHub
> 
> I think it's actually v4.4.x, since you are using Arduino-esp32 v2.0.11, and it uses the older release/v4.4 branch of ESP-IDF. However this doesn't matter for the issue you are facing. Just make sure you are taking the example code from release/v4.4 branch of IDF and not from `master`.

Thank you for the tips. The example from version 4.4 with the corresponding casts could be built without errors.

closed 2024-01-03 by eismeraldo