espressif/esp-idf #12925: USB SERIAL JTAG Echo Example Load Access Fault

ChocolateLoverRaj, 2024-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1194-g5d8fb281e7

### Espressif SoC revision.

ESP32D0WDQ6 (revision 0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3 Super-Mini

### Power Supply used.

USB

### What is the expected behavior?

The example should work as described in the [README](https://github.com/espressif/esp-idf/tree/5d8fb281e7b4ae3f3827c91aa6dc7ef969580b5a/examples/peripherals/usb_serial_jtag/usb_serial_jtag_echo#example-output)
> Type some characters in the terminal connected to the external serial interface. As result you should see echo in the same terminal which you used for typing the characters.

### What is the actual behavior?

I get this error, and then it reboots, and I get an error, over and over again. The actual echoing doesn't work.

### Steps to reproduce.

1. Clone this repo
2. Install `idf.py`
3. Go to `examples/peripherals/usb_serial_jtag/usb_serial_jtag_echo`
4. Run `idf.py set-target esp32c3`
5. Run `idf.py menuconfig`
6. Set `ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG` to `n`
7. Run `idf.py flash monitor`

### Debug Logs.

```plain
I (24) boot: ESP-IDF v5.3-dev-1194-g5d8fb281e7 2nd stage bootloader
I (24) boot: compile time Jan  2 2024 23:22:29
I (25) boot: chip revision: v0.4
I (28) boot.esp32c3: SPI Speed      : 80MHz
I (33) boot.esp32c3: SPI Mode       : DIO
I (38) boot.esp32c3: SPI Flash Size : 2MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
.I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a430h ( 42032) map
I (101) esp_image: segment 1: paddr=0001a458 vaddr=3fc8be00 size=01234h (  4660) load
I (103) esp_image: segment 2: paddr=0001b694 vaddr=40380000 size=04984h ( 18820) load
I (114) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=17b20h ( 97056) map
I (134) esp_image: segment 4: paddr=00037b48 vaddr=40384984 size=073a0h ( 29600) load
I (144) boot: Loaded app from partition at offset 0x10000
I (144) boot: Disabling RNG early entropy source...
I (144) cpu_start: Unicore app
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4004c48c  RA      : 0x4004c4b0  SP      : 0x3fcde280  GP      : 0x3fc8c600  
TP      : 0x00000000  T0      : 0x00000000  T1      : 0x00000000  T2      : 0x00000009  
S0/FP   : 0x000000ff  S1      : 0x00000050  A0      : 0x000000ff  A1      : 0x3fcde298  
A2      : 0x0012ad40  A3      : 0x03ff0000  A4      : 0x0000e000  A5      : 0x60ffe01c  
A6      : 0xfa000000  A7      : 0x00000003  S2      : 0x42000000  S3      : 0x3c020000  
S4      : 0x3fcde430  S5      : 0x0000a430  S6      : 0xffff0000  S7      : 0x40380250  
Stack dump detected
0x4004c48c: uart_tx_flush in ROM

0x4004c4b0: uart_tx_wait_idle in ROM

0x40380250: call_start_cpu0 at /var/home/rajas/Documents/esp-idf/components/esp_system/port/cpu_start.c:363

S8      : 0x00020020  S9      : 0x00010020  S10     : 0x00000005  S11     : 0x3fcde3c4  
T3      : 0x007d0000  T4      : 0x00030000  T5      : 0x00800000  T6      : 0x00800000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x60ffe01c  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x4004c48c in ?? ()
#0  0x4004c48c in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
ELF file SHA256: 

Rebooting...
```

### More Information.

I did not make any changes to the example

suda-morris, 2024-01-04:
How did you set up your board? The README said this example needs two USB cables:

> The board shall be connected to a computer with a single USB cable for flashing and monitoring with UART port. The usb_serial_jtag port on board can be be connected to computer with another USB cable to get the echo.

ChocolateLoverRaj, 2024-01-04:
I didn't realize that. I only used one USB cables connected to the USB-C port. Which pins do I need to connect the D+/D- of the second USB cable to, and then which cable is used to flash the program?

mythbuster5, 2024-02-21:
What about master. We encounter some usj problem on 5.2 which is fixing now.

frozen-eye, 2024-02-21:
> What about master. We encounter some usj problem on 5.2 which is fixing now.

Let me try and get back to you with the results

frozen-eye, 2024-02-21:
I've tried using the latest docker env image "espressif/idf:latest" and can confirm that issue has gone. UART is working well with one URB cable (esp32c3 - xiao)

closed 2024-11-22 by mythbuster5