espressif/esp-idf #12928: CONFIG_SPI_FLASH_ERASE_YIELD_TICKS and CONFIG_SPI_FLASH_ERASE_YIELD_TICKS do not work properly

derrick-senva, 2024-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.5

### Espressif SoC revision.

ESP32-PICO-V3-02 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-PICO-DevKitM-2

### Power Supply used.

USB

### What is the expected behavior?

I expect the call to esp_partition_erase_range() to properly yield CPU cycles by enabling and configuring CONFIG_SPI_FLASH_ERASE_YIELD_TICKS and CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS.

### What is the actual behavior?

Regardless of the values that CONFIG_SPI_FLASH_ERASE_YIELD_TICKS and CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS are set to, the task does not yield CPU cycles and the task watchdog is triggered. Probing the SPI pins shows that the ESP32 continuously polls the external flash until erase operations are complete.

### Steps to reproduce.

1. Start with the [storage/ext_flash_fatfs](https://github.com/espressif/esp-idf/tree/v5.0.3/examples/storage/ext_flash_fatfs) example using a ESP32-PICO-V3-02 dev kit.
2. Connect a 128-Mbit Winbond W25Q128JVS with the following IO# assignments: `MOSI 8, MISO 34, CLK 5, CS 2`
3. Configure the previous IO# assignments, SPI3_HOST (VSPI), SPI_FLASH_FASTRD (for single I/O), and ESP_FLASH_80MHZ.
4. Run the example.

### Debug Logs.

```plain
I (0) cpu_start: App cpu up.
I (311) cpu_start: Pro cpu start user code
I (311) cpu_start: cpu freq: 160000000 Hz
I (311) cpu_start: Application information:
I (316) cpu_start: Project name:     ext_flash_fatfs
I (321) cpu_start: App version:      c8c6d19-dirty
I (327) cpu_start: Compile time:     Jan  4 2024 12:42:52
I (333) cpu_start: ELF file SHA256:  c6fe898d9cd2c3b5...
I (339) cpu_start: ESP-IDF:          v5.0.5
I (344) cpu_start: Min chip rev:     v3.0
I (349) cpu_start: Max chip rev:     v3.99
I (353) cpu_start: Chip rev:         v3.0
I (358) heap_init: Initializing. RAM available for dynamic allocation:
I (366) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (371) heap_init: At 3FFB2928 len 0002D6D8 (181 KiB): DRAM
I (378) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (384) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (390) heap_init: At 4008C4AC len 00013B54 (78 KiB): IRAM
I (398) spi_flash: detected chip: generic
I (401) spi_flash: flash io: dio
W (405) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (414) app_start: Starting scheduler on CPU0
I (418) app_start: Starting scheduler on CPU1
I (418) main_task: Started on CPU0
I (428) main_task: Calling app_main()
I (428) example: Initializing external SPI Flash
I (428) example: Pin assignments:
I (438) example: MOSI:  8   MISO: 34   SCLK:  5   CS:  2
I (438) example: DMA CHANNEL: 3
I (448) spi_flash: detected chip: winbond
I (448) spi_flash: flash io: fastrd
I (458) example: Initialized external Flash, size=16384 KB, ID=0xef4018
I (458) example: Adding external Flash as a partition, label="storage", size=16384 KB
I (468) example: Erasing partition range, offset=0 size=16384 KB
E (5428) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (5428) task_wdt:  - IDLE0 (CPU 0)
E (5428) task_wdt: Tasks currently running:
E (5428) task_wdt: CPU 0: main
E (5428) task_wdt: CPU 1: IDLE1
E (5428) task_wdt: Aborting.
E (5428) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x40008541:0x3ffb4a30 0x40084c81:0x3ffb4a50 0x4008bddd:0x3ffb4a70 0x4008c117:0x3ffb4aa0 0x4008460d:0x3ffb4ac0 0x400de7d7:0x3ffb4b00 0x400d622d:0x3ffb4b20 0x400d6320:0x3ffb4b60 0x400f1af8:0x3ffb4c20 0x4008621d:0x3ffb4c50
0x40084c81: delay_us at C:/Espressif/frameworks/esp-idf-v5.0.5/components/spi_flash/spi_flash_os_func_app.c:169

0x4008bddd: spi_flash_chip_generic_wait_idle at C:/Espressif/frameworks/esp-idf-v5.0.5/components/spi_flash/spi_flash_chip_generic.c:471

0x4008c117: spi_flash_chip_winbond_erase_sector at C:/Espressif/frameworks/esp-idf-v5.0.5/components/spi_flash/spi_flash_chip_winbond.c:118

0x4008460d: esp_flash_erase_region at C:/Espressif/frameworks/esp-idf-v5.0.5/components/spi_flash/esp_flash_api.c:622

0x400de7d7: esp_partition_erase_range at C:/Espressif/frameworks/esp-idf-v5.0.5/components/esp_partition/partition_target.c:134

0x400d622d: example_add_partition at C:/derrick_workspace/eclipse_workspaces/emx-ip-esp32/ext_flash_fatfs/main/ext_flash_fatfs_example_main.c:189 (discriminator 13)

0x400d6320: app_main at C:/derrick_workspace/eclipse_workspaces/emx-ip-esp32/ext_flash_fatfs/main/ext_flash_fatfs_example_main.c:77

0x400f1af8: main_task at C:/Espressif/frameworks/esp-idf-v5.0.5/components/freertos/app_startup.c:208 (discriminator 13)

0x4008621d: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.0.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:149
```

### More Information.

Replacing [line 174](https://github.com/espressif/esp-idf/blob/e004907bd53b68e6db2e29ae335d8b38cbb39472/examples/storage/ext_flash_fatfs/main/ext_flash_fatfs_example_main.c#L174C37-L174C37) with a loop that erases in smaller chunks and calls vTaskDelay() in between calls works fine. The entire example runs to completion (creating a file, writing to it, reading it).

igrr, 2024-01-05:
The issue you are seeing is probably related to the fact that until commit https://github.com/espressif/esp-idf/commit/1b747867bd03695ee22c6d2593c2615ef9a3eb91, "yield during erase" functionality was only implemented for main flash (on SPI1). We didn't backport this change to older releases since it touched flash initialization code and looked a bit risky. Can you try upgrading to IDF 5.1.x and see if the that solves the issue?

derrick-senva, 2024-01-05:
@igrr I'll have to go through the migration guide to get our project up to v5.1.x. If the Breaking Changes are a little too time consuming, is it possible that I can backport this specific fix with some direction (cherry-picked commits)?

Also, when I was looking into the SPI Flash component a few months ago, it seemed like it was treating SPI3 as if it was SPI1 (internal instruction flash) by suspending all tasks during any read/write flash operations. Do you know if this is still the case? Our application is only using this external flash for data storage, so suspending tasks is unnecessary for us.

closed 2024-01-11 by derrick-senva

igrr, 2024-01-11:
> On a related note, when I was looking into the SPI Flash component a few months ago, it seemed like it was treating SPI3 as if it was SPI1 (internal instruction flash) by suspending all tasks during any flash read/write operations. Do you know if this is still the case?

I don't recall that being the case... At least in v5.0.5, seems like the only place where the cache/tasks get suspended is from spi1_start, and we didn't call that for SPI2/SPI3:

https://github.com/espressif/esp-idf/blob/03d7a3807bee18c7b9ffa0638af83b92fc23895f/components/spi_flash/spi_flash_os_func_app.c#L235-L237

If you see some behavior which indicates that other tasks get suspended during access to SPI2/SPI3, please open a new issue and we'll take a look.