espressif/esp-idf #12930: RAM assigned to Task is not displayed in the build output summary

eme-bennytang, 2024-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF version: v5.1.2

In my project, there are a lot of task created for different purpose.
But when I create another one for a new feature, somehow the task just didnt execute at all!
After some digging, I found that the task is created but never executes.

I have tried to change some options in the SDK config and find that "CONFIG_FREERTOS_PLACE_FUNCTIONS_INTO_FLASH" will make it work and the tasks will go back to normal.

But after reading the description of "CONFIG_FREERTOS_PLACE_FUNCTIONS_INTO_FLASH", it should just free 8KB IRAM, and this suggest my project is having a IRAM issue instead of a FREERTOS config issue.
(proven by enabling the config, but have another task consume the 8KB freed from the config on the stack, and issue occur again)

I have purposefully create a task with 8KB stack, and rebuild to create the same task with 16KB stack.
In the build summary, the used RAM is the same!

Is there anyway I can monitor the stack usage for each task and is the overall IRAM usage enough for everything?

sudeep-mohanty, 2024-01-05:
Hi @eme-bennytang,
I assume that you are creating tasks with dynamic allocation, i.e, you maybe using the [xTaskCreate()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html?highlight=xtaskcreate#_CPPv411xTaskCreate14TaskFunction_tPCKcK22configSTACK_DEPTH_TYPEPCv11UBaseType_tPC12TaskHandle_t) (Or its `..PinnedToCore()`alternatives) API to create tasks and not the [xTaskCreateStatic()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html?highlight=xtaskcreate#_CPPv417xTaskCreateStatic14TaskFunction_tPCKcK8uint32_tPCv11UBaseType_tPC11StackType_tPC12StaticTask_t) API. In such a case, the task stack is allocated from the heap memory by default. To see the affect of creating tasks with increasing stack sizes, I suggest to use the [esp_get_free_heap_size()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html?highlight=esp_get_free_heap_size#_CPPv422esp_get_free_heap_sizev) API.

To monitor statically allocated RAM for your app, I would suggest to use the IDF frontend [idf.py size](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-py.html#show-size-size) command (and its variations).

To monitor the stack usage of tasks, you may use the [uxTaskGetStackHighWaterMark()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html#_CPPv427uxTaskGetStackHighWaterMark12TaskHandle_t) API provided by FreeRTOS. 

Some general guidelines for minimizing RAM usage can be found [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html?highlight=stack#minimizing-ram-usage).

Let us know if this information is helpful. Please feel free to reach out if you have more questions.

eme-bennytang, 2024-01-15:
Hi @sudeep-mohanty 
I have tried "esp_get_free_heap_size()" and result is over 100k+ free heap

But according to my "trial and error" approach of the heap, it should be around 8K stack (which is 32K bytes I think).
Anyway I will keep an eye on it, and keep monitor my stack on the tasks with water mark API.

Thanks for the help

closed 2024-01-15 by eme-bennytang