espressif/esp-idf #12934: [MESH][v5.1.2]: Issue WIFI_REASON_ASSOC_TOOMANY when connect 20 Devices same range with Root Device

andy-danieal, 2024-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

With ESP-WIFI-MESH, we have used the **"No-Router"** setting and the **"Fixed Root"** device. However, I noticed that some random two or three devices had been disconnected. Even after restarting, the issue persists. In contrast, when only three devices are connected, all devices are successfully connected.

`ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x183c
load:0x4004b000,len:0x4
load:0x4004b004,len:0xab4
load:0x4004f000,len:0x31b0
entry 0x4004b1c4
I (23) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (24) boot: compile time Dec 13 2023 11:22:36
I (24) boot: chip revision: v1.0
I (28) boot.esp32s2: SPI Speed      : 80MHz
I (32) boot.esp32s2: SPI Mode       : DIO
I (37) boot.esp32s2: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 0000d000 00004000
I (66) boot:  1 otadata          OTA data         01 00 00011000 00002000
I (73) boot:  2 phy_init         RF data          01 01 00013000 00001000
I (81) boot:  3 edf              Custom data        01 02 00014000 00004000
I (88) boot:  4 storage          Unknown data     01 81 00018000 00020000
I (95) boot:  5 ota_0            OTA app          00 10 00040000 00170000
I (103) boot:  6 ota_1            OTA app          00 11 001b0000 00170000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=00040020 vaddr=3f000020 size=2d45ch (185436) map
I (160) esp_image: segment 1: paddr=0006d484 vaddr=3ffc9210 size=02b94h ( 11156) load
I (163) esp_image: segment 2: paddr=00070020 vaddr=40080020 size=b27b0h (731056) map
I (312) esp_image: segment 3: paddr=001227d8 vaddr=3ffcbda4 size=0108ch (  4236) load
I (313) esp_image: segment 4: paddr=0012386c vaddr=40024000 size=15210h ( 86544) load
I (350) boot: Loaded app from partition at offset 0x40000
I (350) boot: Disabling RNG early entropy source...
I (362) cpu_start: Unicore app
I (362) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (364) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (371) cpu_start: Pro cpu up.
I (394) cpu_start: Pro cpu start user code
I (394) cpu_start: cpu freq: 160000000 Hz
I (394) cpu_start: Application information:
I (402) cpu_start: App version:      04763cb-dirty
I (407) cpu_start: Compile time:     Dec 21 2023 19:45:41
I (413) cpu_start: ELF file SHA256:  2b81259c1d8e2a74...
I (419) cpu_start: ESP-IDF:          v5.1.2-dirty
I (425) cpu_start: Min chip rev:     v0.0
I (429) cpu_start: Max chip rev:     v1.99
I (434) cpu_start: Chip rev:         v1.0
I (439) heap_init: Initializing. RAM available for dynamic allocation:
I (446) heap_init: At 3FFD4078 len 00027F88 (159 KiB): DRAM
I (452) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (458) heap_init: At 3FF9E000 len 00001FE8 (7 KiB): RTCRAM
I (466) spi_flash: detected chip: generic
I (469) spi_flash: flash io: dio
I (474) coexist: coex firmware version: b6d5e8c
I (479) app_start: Starting scheduler on CPU0
I (00:00:00.090) main_task: Started on CPU0
I (00:00:00.095) main_task: Calling app_main()
I (00:00:00.110) wearleveling_helper: Mounting FAT filesystem
I (992) wifi:wifi driver task: 3ffe05b8, prio:23, stack:6656, core=0
[... 158 lines cut ...]
I (00:01:09.374) wearleveling_helper: Writing File: /spiflash/switch-data.txt
I (71500) wifi:new:<11,2>, old:<11,0>, ap:<11,2>, sta:<11,2>, prof:11
I (71921) wifi:state: init -> auth (b0)
I (71938) wifi:state: auth -> init (5c0)
I (71939) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<11,2>, prof:11
I (00:01:11.336) [MESH-EVENT]: <MESH_EVENT_PARENT_DISCONNECTED>reason:5
I (73549) wifi:new:<11,2>, old:<11,0>, ap:<11,2>, sta:<11,2>, prof:11
I (73970) wifi:state: init -> auth (b0)
I (73987) wifi:state: auth -> init (5c0)
I (73988) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<11,2>, prof:11
I (00:01:13.385) [MESH-EVENT]: <MESH_EVENT_PARENT_DISCONNECTED>reason:5
I (75597) wifi:new:<11,2>, old:<11,0>, ap:<11,2>, sta:<11,2>, prof:11
I (76018) wifi:state: init -> auth (b0)
I (76035) wifi:state: auth -> init (5c0)
I (76036) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<11,2>, prof:11
I (00:01:15.433) [MESH-EVENT]: <MESH_EVENT_PARENT_DISCONNECTED>reason:5
I (00:01:18.503) [MESH-EVENT]: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (79590) wifi:new:<11,2>, old:<11,0>, ap:<11,2>, sta:<11,2>, prof:11
I (80011) wifi:state: init -> auth (b0)
I (80073) wifi:state: auth -> init (5c0)
I (80074) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<11,2>, prof:11
I (00:01:19.470) [MESH-EVENT]: <MESH_EVENT_PARENT_DISCONNECTED>reason:5
`

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13842848/sdkconfig.txt)

andy-danieal, 2024-01-09:
Any update regrading that issue ?

zhangyanjiaoesp, 2024-01-19:
@andy-danieal 

> W (1161) wifi:Affected by the ESP-NOW encrypt num, set the max connection num to 1

this log indicates the max connection num for SoftAP has been changed to 1 due to EPSNOW, please change the [CONFIG_ESP_WIFI_ESPNOW_MAX_ENCRYPT_NUM](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s2/api-reference/kconfig.html#config-esp-wifi-espnow-max-encrypt-num) to 0 if you don't need ESP-NOW.

here is the doc : https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/wifi.html#ap-basic-configuration
[image]

andy-danieal, 2024-01-19:
Thanks for your replay. And issue has been resolved. But we need ESP-Now also. So, can we use same time?

```
    ESP_ERROR_CHECK(esp_mesh_set_ap_authmode(app_mesh_config->ap_mode));
    cfg.mesh_ap.max_connection         = 8;
    cfg.mesh_ap.nonmesh_max_connection = 2;
    memcpy((uint8_t *)&cfg.mesh_ap.password, app_mesh_config->ap_password, strlen(app_mesh_config->ap_password));
    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
```

```
   CONFIG_ESP_WIFI_ESPNOW_MAX_ENCRYPT_NUM=3
```

closed 2024-01-20 by andy-danieal