espressif/esp-idf #12935: clang build error

Jiu-xiao, 2024-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1 bbd8d13f930705051afa883691bc8389eefc48a1

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

idf_as_lib example, use $ENV{IDF_PATH}/tools/cmake/toolchain-clang-esp32c3.cmake

### What is the actual behavior?

ll/src -I/home/xiao/XRobot/lib/mini_shell/config -isystem /home/xiao/XRobot/src/system/FreeRTOS -isystem /home/xiao/XRobot/src/system --target=riscv32-esp-elf -march=rv32imc -mabi=ilp32  -g -std=gnu++17 -Werror -gdwarf-4 -MD -MT src/system/CMakeFiles/system.dir/FreeRTOS/timer.cpp.obj -MF src/system/CMakeFiles/system.dir/FreeRTOS/timer.cpp.obj.d -o src/system/CMakeFiles/system.dir/FreeRTOS/timer.cpp.obj -c /home/xiao/XRobot/src/system/FreeRTOS/timer.cpp
/tmp/timer-8a0ff0.s: Assembler messages:
/tmp/timer-8a0ff0.s:124: Error: non-constant .uleb128 is not supported
/tmp/timer-8a0ff0.s:206: Error: non-constant .uleb128 is not supported
/tmp/timer-8a0ff0.s:523: Error: non-constant .uleb128 is not supported
/tmp/timer-8a0ff0.s:526: Error: non-constant .uleb128 is not supported
clang-15: error: assembler command failed with exit code 1 (use -v to see invocation)
[1037/1061] Generating ld/sections.ld
ninja: build stopped: subcommand failed.

### Steps to reproduce.

1. install esp-idf and esp-clang
2. copy example idf_as_lib
3. export IDF_TOOLCHAIN=clang 
4. include $ENV{IDF_PATH}/tools/cmake/toolchain-clang-esp32c3.cmake
5. build

### Build or installation Logs.

```plain
[32/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/secure_boot.c.obj
[33/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_common.c.obj
[34/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_efuse.c.obj
[35/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/flash_encrypt.c.obj
[36/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_common_loader.c.obj
[37/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/flash_qio_mode.c.obj
[38/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/flash_partitions.c.obj
[39/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_clock_loader.c.obj
[40/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_console_loader.c.obj
[41/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_init.c.obj
[42/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32c3/bootloader_sha.c.obj
[43/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/bootloader_flash.c.obj
[44/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32c3/bootloader_soc.c.obj
[45/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_panic.c.obj
[46/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/bootloader_flash_config_esp32c3.c.obj
[47/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp_image_format.c.obj
[48/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_console.c.obj
[49/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_utility.c.obj
[50/96] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32c3/esp_efuse_table.c.obj
[51/96] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32c3/esp_efuse_fields.c.obj
[52/96] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32c3/bootloader_esp32c3.c.obj
[... 43 lines cut ...]
esptool.py v4.7.0
Creating esp32c3 image...
Successfully created esp32c3 image.
Generated /home/xiao/XRobot/build/bootloader/bootloader.bin
[96/96] cd /home/xiao/XRobot/build/bootloader/esp-idf/esptool_py && python /home/xiao/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/xiao/XRobot/build/bootloader/bootloader.bin
Bootloader binary size 0x5140 bytes. 0x2ec0 bytes (37%) free.
[1028/1061] Building CXX object src/system/CMakeFiles/system.dir/FreeRTOS/memory.cpp.obj
FAILED: src/system/CMakeFiles/system.dir/FreeRTOS/memory.cpp.obj 
/home/xiao/.espressif/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin/clang++ -DFREERTOS_TERM_TASK_STACK_DEPTH=1024 -DFREERTOS_TIMER_TASK_STACK_DEPTH=1024 -DFREERTOS_USB_TASK_STACK_DEPTH=1024 -DINIT_TASK_STACK_DEPTH=2048 -DMCU_DEBUG_BUILD -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DXROBOT_BOARD=esp32-c3-idf -I/home/xiao/XRobot/hw/mcu/default -I/home/xiao/XRobot/hw/bsp/esp32-c3-idf/drivers -I/home/xiao/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/xiao/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/xiao/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/xiao/esp/esp-idf/components/freertos/esp_additions/include -I/home/xiao/esp/esp-idf/components/freertos/esp_additions/arch/riscv/include -I/home/xiao/XRobot/build/config -I/home/xiao/esp/esp-idf/components/newlib/platform_include -I/home/xiao/esp/esp-idf/components/esp_hw_support/include -I/home/xiao/esp/esp-idf/components/esp_hw_support/include/soc -I/home/xiao/esp/esp-idf/components/esp_hw_support/include/soc/esp32c3 -I/home/xiao/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -I/home/xiao/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -I/home/xiao/esp/esp-idf/components/heap/include -I/home/xiao/esp/esp-idf/components/log/include -I/home/xiao/esp/esp-idf/components/soc/include -I/home/xiao/esp/esp-idf/components/soc/esp32c3 -I/home/xiao/esp/esp-idf/components/soc/esp32c3/include -I/home/xiao/esp/esp-idf/components/hal/esp32c3/include -I/home/xiao/esp/esp-idf/components/hal/include -I/home/xiao/esp/esp-idf/components/hal/platform_port/include -I/home/xiao/esp/esp-idf/components/esp_rom/include -I/home/xiao/esp/esp-idf/components/esp_rom/include/esp32c3 -I/home/xiao/esp/esp-idf/components/esp_rom/esp32c3 -I/home/xiao/esp/esp-idf/components/esp_common/include -I/home/xiao/esp/esp-idf/components/esp_system/include -I/home/xiao/esp/esp-idf/components/esp_system/port/soc -I/home/xiao/esp/esp-idf/components/esp_system/port/include/riscv -I/home/xiao/esp/esp-idf/components/esp_system/port/include/private -I/home/xiao/esp/esp-idf/components/riscv/include -I/home/xiao/esp/esp-idf/components/lwip/include -I/home/xiao/esp/esp-idf/components/lwip/include/apps -I/home/xiao/esp/esp-idf/components/lwip/include/apps/sntp -I/home/xiao/esp/esp-idf/components/lwip/lwip/src/include -I/home/xiao/esp/esp-idf/components/lwip/port/include -I/home/xiao/esp/esp-idf/components/lwip/port/freertos/include -I/home/xiao/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/xiao/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/xiao/esp/esp-idf/components/driver/include -I/home/xiao/esp/esp-idf/components/driver/deprecated -I/home/xiao/esp/esp-idf/components/driver/analog_comparator/include -I/home/xiao/esp/esp-idf/components/driver/dac/include -I/home/xiao/esp/esp-idf/components/driver/gpio/include -I/home/xiao/esp/esp-idf/components/driver/gptimer/include -I/home/xiao/esp/esp-idf/components/driver/i2c/include -I/home/xiao/esp/esp-idf/components/driver/i2s/include -I/home/xiao/esp/esp-idf/components/driver/ledc/include -I/home/xiao/esp/esp-idf/components/driver/mcpwm/include -I/home/xiao/esp/esp-idf/components/driver/parlio/include -I/home/xiao/esp/esp-idf/components/driver/pcnt/include -I/home/xiao/esp/esp-idf/components/driver/rmt/include -I/home/xiao/esp/ [... cut ...] er.cpp
/tmp/timer-d34779.s: Assembler messages:
/tmp/timer-d34779.s:124: Error: non-constant .uleb128 is not supported
/tmp/timer-d34779.s:206: Error: non-constant .uleb128 is not supported
/tmp/timer-d34779.s:523: Error: non-constant .uleb128 is not supported
/tmp/timer-d34779.s:526: Error: non-constant .uleb128 is not supported
clang-15: error: assembler command failed with exit code 1 (use -v to see invocation)
[1037/1061] Generating ld/sections.ld
ninja: build stopped: subcommand failed.
```

igrr, 2024-01-05:
It's a bit hard to say without seeing what's inside /home/xiao/XRobot/src/system/FreeRTOS/timer.cpp, do you think you can share a minimal example to reproduce the issue?

Jiu-xiao, 2024-01-05:
#include <timer.hpp>

#include "bsp_time.h"

using namespace System;

Timer* Timer::self_ = NULL;

Timer::Timer() {
  self_ = this;

  auto thread_fn = [](void* arg) {
    XB_UNUSED(arg);
    uint32_t last_online_time = bsp_time_get_ms();
    while (1) {
      Timer::self_->list_.Foreach(Timer::Refresh, NULL);
      Timer::self_->thread_.SleepUntil(1, last_online_time);
    }
  };

  this->thread_.Create(thread_fn, static_cast<void*>(NULL), "timer_task",
                       FREERTOS_TIMER_TASK_STACK_DEPTH, Thread::HIGH);
}

bool Timer::Refresh(ControlBlock& block, void* arg) {
  XB_UNUSED(arg);

  if (!block.running) {
    return true;
  }

  block.count++;
  if (block.cycle <= block.count) {
    block.count = 0;
    block.fun(block.type);
  }

  return true;
}

Jiu-xiao, 2024-01-05:
https://github.com/xrobot-org/XRobot/tree/master/src/system/FreeRTOS

Jiu-xiao, 2024-01-05:
build pass after remove `add_link_options(-fdump-rtl-dfinish)`

denizzzka, 2024-08-19:
Faced with same error for `nvs_flash` component on `v5.3`

denizzzka, 2024-08-19:
https://github.com/llvm/llvm-project/issues/57998