espressif/esp-idf #12937: Request for example showing blufi usage along side of another ble service

cmorganBE, 2024-01-05:
### Is your feature request related to a problem?

Yes, I'd like to expose blufi interface on my IoT device while also providing another service for control and configuration of the device, blufi is used to get the device on wifi but we'd like to be able to interact with the device via BLE as well, and avoid having the system be in two different modes, one for blufi and one for other things, so ideally we'd want simultaneous advertising of blufi and our other services.

### Describe the solution you'd like.

An example showing how to use blufi with another ble service, https://esp32.com/viewtopic.php?f=13&t=37634.

### Describe alternatives you've considered.

I've reviewed the ble_multi_adv example but this doesn't directly apply to blufi.

zhp0406, 2024-01-08:
Could you provide a more specific explanation of "providing a BLE interface to enable them to control and query my device"?

cmorganBE, 2024-01-08:
@zhp0406 I mean only that I'd like to advertise and be able to implement one or more BLE services simultaneously along side of blufi, like the ble_multi_adv example but with one of those being blufi.

cmorganBE, 2024-02-01:
Any feedback on this request? I'm back looking at how one might mix blufi with other services and struggling as I don't understand ble or nimble particularly well at the moment. Could use some pointers if you have any...

zhp0406, 2024-02-04:
@cmorganBE 
Blufi supports both Bluedroid host and Nimble host. Which one would you like to use?

zhp0406, 2024-02-04:
In Bluedroid host, the callback for GATT Server (GATTS) is in the 'blufi_profile_cb.' Within the GATTS callback, you can create services and add characteristics.

see:
https://github.com/espressif/esp-idf/blob/master/components/bt/common/btc/profile/esp/blufi/bluedroid_host/esp_blufi.c

and in nimble host:
https://github.com/espressif/esp-idf/blob/master/components/bt/common/btc/profile/esp/blufi/nimble_host/esp_blufi.c

cmorganBE, 2024-02-04:
@zhp0406 in my case I'm using nimble for lower memory overhead.

In that case would I have to copy/modify esp_blufi.c to integrate adding services/characteristics specific to my implementation? I had assumed that doing so would of course be possible but also complex for someone like myself starting with using nimble and ble.

rahult-github, 2024-02-05:
Hi @cmorganBE ,

I think, there need to be some application level changes that can be done to achieve this.

1. ble_multi_adv uses extended adv. So if there is any attempt to start blufi advertising, i suggest you can modify the application to have one more instance of adv data, which is blufi. 

2. And if legacy adv is to be used, once blufi is connected, you can start advertising again. So, as POC, i did bleprph advertising once blufi is connected. This way, we can connect over blufi  and device now advertises a new set and hence a third party central device can connect.    [advertise_after_connect.txt](https://github.com/espressif/esp-idf/files/14164746/advertise_after_connect.txt)

rahult-github, 2024-03-26:
Closing this issue. If anything is needed from stack, please feel free to reopen

closed 2024-03-26 by rahult-github

cmorganBE, 2024-05-15:
Thank you for responding @rahult-github. I ended up switching to esp-nimble-cpp when I started looking at simplifying the api with some helper functions and then dropped using blufi and just added a characteristic to take a json payload for wifi joining. Would be great if there was an example of blufi with esp-nimble-cpp but I know its not an espressif component. Imo it could be, its really good at simplifying the api space, there is SOOOO much boilerplate with the c nimble stuff.