espressif/esp-idf #12939: Multiple HTTP POST request fail

pyrotechnik, 2024-01-06:
Hello,
I have a problem with HTTP Post. ESP is like an HTTP server to which I send requests. The problem occurs when I want to send multiple POST requests in sequence. Only the first 5 always go through. I tried putting a delay between each request. For example 1 second and still nothing. As if there is some limit to POST requests. Do you know what they could be? I would need to occasionally send dozens of POST requests in a row.
I'm using the code from the example HTTP server simply.
Thank you.

Behn4m, 2024-01-11:
I have kind of the same problem with GET requests - I am using the provided example code snippet

hmalpani, 2024-01-22:
Hello @pyrotechnik 
Can you set `enable_so_linger = true` in `httpd_config_t` structure while setting up the server. Please let me know if it helps.

pyrotechnik, 2024-01-27:
@hmalpani 
Hi, thanks for the advice, but unfortunately it's unchanged. It only ever passes 5 requests 6 are sent but ignored.

hmalpani, 2024-01-29:
Hello @pyrotechnik 
Can you also try setting `lru_purge_enable = true` in `httpd_config_t`. If this doesn't help, can you share the sample application(with sdkconfig) and steps you used to reproduce this issue?

pyrotechnik, 2024-01-29:
My project is large (sqlite, wifi, SDMMC, SPI Lora), but there is simply code. In JS app I fetch HTTP_SET. And after 6 SET Request HTTP server stop receive. Until I reload app.

````
#include <stdio.h>
#include <stdbool.h>
#include <string.h>
#include <unistd.h>
#include "esp_timer.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_vfs_fat.h"
#include "sdmmc_cmd.h"
#include "driver/sdmmc_host.h"
#include <inttypes.h>
#include "esp_wifi.h"
#include <esp_http_server.h>

#include <esp_event.h>
#include <esp_log.h>
#include <esp_system.h>
#include <nvs_flash.h>
#include <sys/param.h>
#include "esp_netif.h"
#include "esp_eth.h"

#define HTTPD_CONFIG {                        \
        .task_priority      = tskIDLE_PRIORITY+5,       \
        .stack_size         = 20480,                     \
        .core_id            = tskNO_AFFINITY,           \
        .server_port        = 80,                       \
        .ctrl_port          = ESP_HTTPD_DEF_CTRL_PORT,  \
        .max_open_sockets   = 7,                        \
        .max_uri_handlers   = 20,                        \
        .max_resp_headers   = 10,                        \
        .backlog_conn       = 10,                        \
        .lru_purge_enable   = false,                    \
        .recv_wait_timeout  = 5,                        \
        .send_wait_timeout  = 5,                        \
        .global_user_ctx = NULL,                        \
        .global_user_ctx_free_fn = NULL,                \
        .global_transport_ctx = NULL,                   \
        .global_transport_ctx_free_fn = NULL,           \
        .enable_so_linger = false,                      \
        .linger_timeout = 0,                            \
        .keep_alive_enable = false,                     \
        .keep_alive_idle = 0,                           \
        .keep_alive_interval = 0,                       \
        .keep_alive_count = 0,                          \
        .open_fn = NULL,                                \
        .close_fn = NULL,                               \
        .uri_match_fn = NULL                            \
}

httpd_handle_t server_handle = NULL;

#define EXAMPLE_ESP_MAXIMUM_RETRY  5

const char *ssid = "Wi-Fi";
const char *pass = "password";

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static int retry_num = 0;

[... 129 lines cut ...]
    return server;
}

void app_main(void)
{
	esp_err_t ret,res;
	//Initialize NVS for WiFi
		   ret = nvs_flash_init();
		    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
		      ESP_ERROR_CHECK(nvs_flash_erase());
		      ret = nvs_flash_init();
		    }
		    ESP_ERROR_CHECK(ret);

		    ESP_LOGI("WiFi", "ESP_WIFI_MODE_STA");
		    wifi_connection();

		server_handle = start_webserver();

    while (true) {
        printf("Hello from app_main!\n");
        sleep(1);
    }
}
````

hmalpani, 2024-01-30:
Hello @pyrotechnik 
Can you please try setting `lru_purge_enable=true` in httpd_config_t once. When this option is enabled and no socket is available for new connection, esp_http_server will purge/close least recently used socket.

I think this will help in your use case.

pyrotechnik, 2024-01-30:
That didn't help either, I finally found that after each time I received a REQUEST. the session stayed open and got to max 7 quickly. So I added a command to the SET request handler to close the session and now it works as it should. Thank you for your help.

hmalpani, 2024-02-01:
Thanks for the update. Closing this issue as it has been fixed for you. Please feel free to reopen in case if you see the issue again.

closed 2024-02-01 by hmalpani