espressif/esp-idf #12940: esp32c6 - SHA-256 comparison failed

ds2k5, 2024-01-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

```
rustup target add riscv32imac-unknown-none-elf

cargo generate esp-rs/esp-idf-template cargo
cd esp32c6
. ~/export-esp.sh
cargo check
cargo run --release

```
main.rs

```
fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Hello, world!");
}

```Console:
```

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
0x4001974a - __umoddi3

SHA-256 comparison failed:
Calculated: aff89878a96cbff57c66c38aa6d1a422b7785040b5efea582babef6c357427c2
Expected: 0af544a033ab3492852b8232c904c578d5f07c9d4a423a64473f060db374ab32
Attempting to boot anyway...
entry 0x4086c410
0x4086c410 - _instruction_reserved_end
    at ??:??
I (41) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (41) boot: compile time Jun  7 2023 08:02:08
I (42) boot: chip revision: v0.0
I (46) boot.esp32c6: SPI Speed      : 40MHz
I (51) boot.esp32c6: SPI Mode       : DIO
I (55) boot.esp32c6: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (84) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 003f0000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=11ca0h ( 72864) load
E (112) esp_image: Segment 0 0x3f400020-0x3f411cc0 invalid: bad load address range
0x3f400020 - _rodata_start
    at ??:??
0x3f411cc0 - _lit4_end
    at ??:??
E (120) boot: Factory app partition is not bootable
E (125) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
0x4001974a - __umoddi3
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
0x4086c410 - _instruction_reserved_end
    at ??:??
load:0x4086e610,len:0x2d68
0x4086e610 - _instruction_reserved_end
    at ??:??
load:0x40875720,len:0x1800
0x40875720 - _instruction_reserved_end
    at ??:??
SHA-256 comparison failed:
Calculated: aff89878a96cbff57c66c38aa6d1a422b7785040b5efea582babef6c357427c2
Expected: 0af544a033ab3492852b8232c904c578d5f07c9d4a423a64473f060db374ab32
Attempting to boot anyway...
entry 0x4086c410
0x4086c410 - _instruction_reserved_end
    at ??:??
I (41) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (41) boot: compile time Jun  7 2023 08:02:08
I (42) boot: chip revision: v0.0
I (46) boot.esp32c6: SPI Speed      : 40MHz
I (51) boot.esp32c6: SPI Mode       : DIO
I (55) boot.esp32c6: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (84) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 003f0000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=11ca0h ( 72864) load
E (112) esp_image: Segment 0 0x3f400020-0x3f411cc0 invalid: bad load address range
0x3f400020 - _rodata_start
    at ??:??
0x3f411cc0 - _lit4_end
    at ??:??
E (120) boot: Factory app partition is not bootable
E (125) boot: No bootable app partitions in the partition table

```

What did I wrong ?

ds2k5, 2024-01-08:
@MabezDev 

thank you!

ivmarkov, 2024-01-09:
**Suggest to close**, as upstream fix just got merged. Tomorrow's nightly should be buildable again.
(And this is not an issue with ESP IDF per se anyway, but with the Rust wrappers.)

closed 2024-01-09 by Alvin1Zhang