espressif/esp-idf #12946: Failed to build example project on windows(both vscode & expressif-ide)

zengbiao, 2024-01-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

successfully build example project

### What is the actual behavior?

C:\xxx\blink>idf.py -v build
Checking Python dependencies...
Python requirements are satisfied.
Constraint file: C:\app\06\Espressif\espidf.constraints.v5.1.txt
Requirement files:
 - C:\Users\Administrator\esp\esp-idf\tools\requirements\requirements.core.txt
Python being checked: C:\app\06\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Executing action: all (aliases: build)
Running cmake in directory C:\xxx\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\app\06\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\xxx\zbblink1\blink"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.2")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/app/06/Espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - **failed**
-- Check for working C compiler: C:/app/06/Espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/app/06/Espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - broken
CMake Error at C:/app/06/Espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "C:/app/06/Espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/xxx/blink/build/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/app/06/Espressif/tools/ninja/1.10.2/ninja.exe cmTC_d9659 && [1/2] Building C object CMakeFiles/cmTC_d9659.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_d9659
    FAILED: cmTC_d9659
    cmd.exe /C "cd . && C:\app\06\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -mlongcalls -Wno-frame-address  CMakeFiles/cmTC_d9659.dir/testCCompiler.c.obj -o cmTC_d9659   && cd ."
    The system cannot find the path specified.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/Administrator/esp/esp-idf/tools/cmake/project.cmake:448 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/xxx/zbblink1/blink/build/CMakeFiles/CMakeOutput.log".
See also "C:/xxx/zbblink1/blink/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1, output of the command is in the C:\xxx\zbblink1\blink\build\log\idf_py_stderr_output_13460 and C:\xxx\zbblink1\blink\build\log\idf_py_stdout_output_13460

### Steps to reproduce.

1. same error occured when I used vscode
2. same error occured when I used espressif-ide
3. same error occured when I used cmd
issue :  esp-idf  version >=4.0

### Build or installation Logs.

```plain
xxx
```

### More Information.

**我试过好几个版本，报错都一样，ABI 检测失败，ninja 执行失败，不清楚到底哪里的问题，时间比较紧急，麻烦程序员小哥赶紧看下**

brianignacio5, 2024-01-08:
How is this related to vscode or Espressif-ide if the user is building from command line terminal ?

Could you provide steps you used to build with vscode or Espressif-ide ? Have you review your environment to see if it is working properly ? Could you try deleting current build directory and try again ?

zengbiao, 2024-01-08:
attached CMakeError.log
'''
Run Build Command(s):C:/app/06/Espressif/tools/ninja/1.10.2/ninja.exe cmTC_7d693 && [1/2] Building C object CMakeFiles/cmTC_7d693.dir/CMakeCCompilerABI.c.obj

Using built-in specs.

COLLECT_GCC=C:\app\06\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe

Target: xtensa-esp32-elf

Configured with: /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-host_w64-mingw32 --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --exec_prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-12.2.0_20230208' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-libstdcxx-verbose --with-gmp=/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/buildtools/complibs-host --with-mpfr=/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/buildtools/complibs-host --with-mpc=/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/buildtools/complibs-host --with-isl=/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/buildtools/complibs-host --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes --with-gnu-ld

Thread model: posix

Supported LTO compression algorithms: zlib

gcc version 12.2.0 (crosstool-NG esp-12.2.0_20230208) 

COLLECT_GCC_OPTIONS='-mlongcalls' '-Wno-frame-address' '-v' '-o' 'CMakeFiles/cmTC_7d693.dir/CMakeCCompilerABI.c.obj' '-c' '-dumpdir' 'CMakeFiles/cmTC_7d693.dir/'

 c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/12.2.0/cc1.exe -quiet -v -iprefix c:\app\06\espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\../lib/gcc/xtensa-esp32-elf/12.2.0/ C:/app/06/Espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeCCompilerABI.c -quiet -dumpdir CMakeFiles/cmTC_7d693.dir/ -dumpbase CMakeCCompilerABI.c.c -dumpbase-ext .c -mlongcalls -Wno-frame-address -version -o C:\Users\ADMINI~1\AppData\Local\Temp\ccpKssm9.s

GNU C17 (crosstool-NG esp-12.2.0_20230208) version 12.2.0 (xtensa-esp32-elf)

	compiled by GNU C version 6.3.0 20170516, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.1, isl version isl-0.24-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072

ignoring duplicate directory "c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/lib/gcc/../../lib/gcc/xtensa-esp32-elf/12.2.0/include"

ignoring duplicate directory "c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/lib/gcc/../../lib/gcc/xtensa-esp32-elf/12.2.0/include-fixed"

ignoring duplicate directory "c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/lib/gcc/../../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/sys-include"

ignoring duplicate directory "c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/lib/gcc/../../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/include"

#include "..." search starts here:

#include <...> search starts here:

 c:\app\06\espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\../lib/gcc/xtensa-esp32-elf/12.2.0/include

 c:\app\06\espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\../lib/gcc/xtensa-esp32-elf/12.2.0/include-fixed

 c:\app\06\espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/sys-include

 c:\app\06\espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/include

End of search list.

GNU C17 (crosstool-NG esp-12.2.0_20230208) version 12.2.0 (xtensa-esp32-elf)

	compiled by GNU C version 6.3.0 20170516, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.1, isl version isl-0.24-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072

Compiler executable checksum: 4c06f819f3a69a83cffbb10c8d3d9a0e

COLLECT_GCC_OPTIONS='-mlongcalls' '-Wno-frame-address' '-v' '-o' 'CMakeFiles/cmTC_7d693.dir/CMakeCCompilerABI.c.obj' '-c' '-dumpdir' 'CMakeFiles/cmTC_7d693.dir/'

 c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/as.exe --traditional-format --longcalls -o CMakeFiles/cmTC_7d693.dir/CMakeCCompilerABI.c.obj C:\Users\ADMINI~1\AppData\Local\Temp\ccpKssm9.s

COMPILER_PATH=c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/12.2.0/;c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../libexec/gcc/;c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/ [... cut ...] ensa-esp32-elf/lib CMakeFiles/cmTC_7d693.dir/CMakeCCompilerABI.c.obj -lgcc -lc -lnosys -lc -lgcc c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/crtend.o c:/app/06/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/crtn.o

COLLECT_GCC_OPTIONS='-mlongcalls' '-Wno-frame-address' '-v' '-o' 'cmTC_7d693' '-dumpdir' 'cmTC_7d693.'

ninja: build stopped: subcommand failed.
'''

zengbiao, 2024-01-08:
> How is this related to vscode or Espressif-ide if the user is building from command line terminal ?
>
> Could you provide steps you used to build with vscode or Espressif-ide ? Have you review your [...]

Try to build official example project in 3 ways with the same error...
1: vscode "ESP-IDF: build your project"
2:espressif-ide: build button
3:windows terminal: export.bat  $ idf.py --no-ccache -v build

**Guess this error caused from CMakeTestCCompile.cmake or something with ninja?**

`  # Clear result from normal variable.
  unset(CMAKE_C_COMPILER_WORKS)
  # Puts test result in cache variable.
  try_compile(CMAKE_C_COMPILER_WORKS ${CMAKE_BINARY_DIR}
    ${CMAKE_BINARY_DIR}${CMAKE_FILES_DIRECTORY}/CMakeTmp/testCCompiler.c
    OUTPUT_VARIABLE __CMAKE_C_COMPILER_OUTPUT)
  # Move result from cache to normal variable.
  set(CMAKE_C_COMPILER_WORKS ${CMAKE_C_COMPILER_WORKS})
  unset(CMAKE_C_COMPILER_WORKS CACHE)
  __TestCompiler_restoreTryCompileTargetType()
  if(NOT CMAKE_C_COMPILER_WORKS)
    PrintTestCompilerResult(CHECK_FAIL "broken")
    file(APPEND ${CMAKE_BINARY_DIR}${CMAKE_FILES_DIRECTORY}/CMakeError.log
      "Determining if the C compiler works failed with "
      "the following output:\n${__CMAKE_C_COMPILER_OUTPUT}\n\n")
    string(REPLACE "\n" "\n  " _output "${__CMAKE_C_COMPILER_OUTPUT}")
    message(FATAL_ERROR "The C compiler\n  \"${CMAKE_C_COMPILER}\"\n"
      "is not able to compile a simple test program.\nIt fails "
      "with the following output:\n  ${_output}\n\n"
      "CMake will not be able to correctly generate this project.")
  endif()
  PrintTestCompilerResult(CHECK_PASS "works")
  file(APPEND ${CMAKE_BINARY_DIR}${CMAKE_FILES_DIRECTORY}/CMakeOutput.log
    "Determining if the C compiler works passed with "
    "the following output:\n${__CMAKE_C_COMPILER_OUTPUT}\n\n")
endif()`

zengbiao, 2024-01-08:
> 检查你的 cmd.exe 是否正常。参见：
> 
> https://superuser.com/questions/727316/error-in-command-line-the-system-cannot-find-the-path-specified

牛逼，确实是我之前改过注册表导致的。谢谢

zengbiao, 2024-01-08:
注册表中加了错误的自动运行命令，影响了编译器

closed 2024-01-08 by zengbiao