espressif/esp-idf #12948: ESP32-C3 Pixel NIMBLE connection error 133

vsh-ektos, 2024-01-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-C3FN4 (rev. v0.3)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I am using the BLE Heart Rate Measurement Example on the ESP32-C3FN4. The microcontroller is flashed successfully and BLE works correctly with Android and iPhone devices.

### What is the actual behavior?

But it does not work with Pixel smartphones (tested on Pixel 6a and Pixel 7). When I try to connect to the ESP32-C3 via nRF-connect using a Pixel, I get an error Error 133 (0x85): GATT ERROR in about 95% of the cases (in about 5% of the attempts I manage to connect successfully). In the case when the connection is not successful, I do not receive any logs on the ESP-32C3.

### Steps to reproduce.

1. Create firmware using the BLE Heart Rate Measurement Example 
2. Flash to ESP32-C3FN4. 
3. On Pixel 6a or Pixel 7, install nRF Connect
4. Try to connect to the device using nRF Connect

### More Information.

If I use Extended advertise, everything works correctly, but I need BLE 4.0 support

Singh-Sumeet, 2024-01-10:
Hi @vsh-ektos,
I ran the `ble_htp_prph` application on ESP32-C3 (IDF version 5.1.2) and connected it to my Pixel 7.
On my side, there is a 100% connection rate, and I do not observe the error you mentioned.
To help you further I will need debug enabled logs of `ble_htp_prph` application and the nRF Connect app.

For `ble_htp_prph`, before doing `idf.py build`, do these steps:
1. `idf.py menuconfig` -> Component Config -> Bluetooth -> NimBLE Options -> NimBLE Host log verbosity -> Select Debug logs
2. `idf.py menuconfig` -> Component Config -> Log output -> Default log verbosity -> Select Debug
In both cases press `S` to *save* or press `Esc` to *exit* and `y` to *save and exit*
This will debug logs to show in `idf.py monitor`

For nRF Connect,
1. Show logs using the menu on the top left or drag the screen right.
2. Select DEBUG in the drop down menu at bottom left.
3. Press save.

vsh-ektos, 2024-01-10:
Pixel 7
Log from nRF Connect
[Log 2024-01-10 14_15_01.txt](https://github.com/espressif/esp-idf/files/13887469/Log.2024-01-10.14_15_01.txt)

Log from idf.py monitor
[ble_htp_prph_pixel_log.txt](https://github.com/espressif/esp-idf/files/13887493/ble_htp_prph_pixel_log.txt)

vsh-ektos, 2024-01-10:
When I use a Samsung smartphone, no problems at all

Samsung Galaxy M21
Log from nRF Connect
[Log 2024-01-10 14_20_22.txt](https://github.com/espressif/esp-idf/files/13887594/Log.2024-01-10.14_20_22.txt)

Log from idf.py monitor
[ble_htp_prph_samsung.txt](https://github.com/espressif/esp-idf/files/13887605/ble_htp_prph_samsung.txt)

vsh-ektos, 2024-01-10:
I checked additionally on ESP32-M5STAMPC3, and everything works correctly with Pixel 7

It seems the problem only affects the ESP32-C3FN4

vsh-ektos, 2024-01-11:
Most likely, the problem is in the hardware. 
Thank you for your help.
Could you tell me who I can contact for help with Hardware?

Singh-Sumeet, 2024-01-11:
Hi @vsh-ektos ,
The issue has already been routed to the concerned team, someone should get back to you once there is an update.