espressif/esp-idf #12949: Bootloader Version

diplfranzhoepfinger, 2024-01-09:


ESP-Marius, 2024-01-10:
It should have the same info, see [bootloader image format](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/bootloader_image_format.html)

or read it directly with e.g. `esptool.py --chip esp32 image_info build/bootloader/bootloader.bin --version 2`

ESP-Marius, 2024-01-15:
Marking this one as closed, if you feel what I linked to do not cover the feature you requested then feel free to reopen the issue.

closed 2024-01-15 by ESP-Marius