espressif/esp-idf #12951: Problems with source file "xtensa_api.h" (dependency of "freertos/FreeRTOS.h")

MarcelloDelDino, 2024-01-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hey community,

at the moment I am having some trouble with the xtensa libraries.
As soon as I include for example freertos/FreeRTOS.h it is giving me the following errors in VSCode.

#include errors detected. Consider updating your compile_commands.json or includePath. Squiggles are disabled for this translation unit (/home/marcel/esp_projects/esp1/main/main.c).

cannot open source file "xtensa_api.h" (dependency of "freertos/FreeRTOS.h")

This is how the c_cpp_properties.json looks like.

`{
"configurations": [
{
"name": "ESP-IDF",
"compilerPath": "${config:idf.toolsPath}/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc",
"includePath": [
"${config:idf.espIdfPath}/components/**",
"${config:idf.espIdfPathWin}/components/**",
"${config:idf.espAdfPath}/components/**",
"${config:idf.espAdfPathWin}/components/**",
"/home/marcel/esp/esp-idf/components/**",
"${workspaceFolder}/**"
],
"browse": {
"path": [
"${config:idf.espIdfPath}/components",
"${config:idf.espIdfPathWin}/components",
"${config:idf.espAdfPath}/components/**",
"${config:idf.espAdfPathWin}/components/**",
"${workspaceFolder}",
"/home/marcel/esp/esp-idf/components/**"
],
"limitSymbolsToIncludedHeaders": false
},
"cStandard": "c17",
"cppStandard": "c++17",
"compileCommands": "${workspaceFolder}/build/compile_commands.json"
}
],
"version": 4
}`

I am using EndeavourOS (Arch Linux) with VSCodium and ESP-IDF v5.3-dev-1288-g5524b692ee, but I am getting the same error with VSCode in Windows 10.

When I include xtensa_api.h manually it is not giving me that error anymore, but then it starts to show up other dependencies missing.

Do you need more information to help me with my problem?

Best regards,
Marcel

brianignacio5, 2024-01-10:
This is related t your C/C++ extension configuration.

Have your build your project to generate the `compile_commands.json` file ?

More info in https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md

MarcelloDelDino, 2024-01-10:
Yes, I build the project already.
Also I can see the file 'compile_commands.json' in my build folder.
There are a lot of directories.

Also When I include something else from the ESP-IDF components directory, for exapmple "freertos/task.h", it will not create any error.

It seems to be specifically with xtensa, but I found the header files from xtensa in the components folder.

MarcelloDelDino, 2024-01-18:
Is there anything else I can do to solve the issue?
Do you need more information from me?

martmalo, 2024-02-08:
I have the same problem

MarcelloDelDino, 2024-05-21:
Hey everybody, 
I managed to fix this issue by changing the compilerPath in c_cpp_properties.json to the following value.
"${config:idf.toolsPath}/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc"
This is needed, because the ESP32C3 is using the Risc-V architecture.

When you create a new project with the getting started sample_project it will always be as follows.
"${config:idf.toolsPath}/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc"
But this is not suitable for Risc-V.

Best regards,
Marcel