espressif/esp-idf #12955: Crash on wifi stack after calling esp_restart

QuentinFarizon, 2024-01-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-DevkitM1

### Power Supply used.

USB

### What is the expected behavior?

I am calling esp_restart, and expecting it to restart properly

### What is the actual behavior?

There is a crash in wifi stack, apparently due to logging :

```
abort() was called at PC 0x40377183 on core 0
0x40377183: lock_acquire_generic at /home/quentin/esp/esp-idf/components/newlib/locks.c:130

Backtrace: 0x40375b62:0x3fc9c290 0x403806e9:0x3fc9c2b0 0x40386806:0x3fc9c2d0 0x40377183:0x3fc9c340 0x40377279:0x3fc9c380 0x40377348:0x3fc9c3a0 0x420b2d72:0x3fc9c3d0 0x420b5ee1:0x3fc9c6f0 0x420c63e5:0x3fc9c720 0x403859a1:0x3fc9c750 0x4207431d:0x3fc9c7a0 0x420834b6:0x3fc9c810 0x4207a575:0x3fc9c880 0x4207a5f8:0x3fc9c8a0 0x4207b78c:0x3fc9c8c0 0x42084c3a:0x3fc9c8e0 0x4207a1dd:0x3fc9c900 0x4207b638:0x3fc9c920 0x42079cee:0x3fc9c940 0x4207c78a:0x3fc9c960 0x42072c31:0x3fc9c980 0x420041a5:0x3fc9c9b0 0x4200db63:0x3fc9c9e0 0x40377629:0x3fc9ca00 0x4037768d:0x3fc9ca30 0x40377991:0x3fc9ca60 0x40376f75:0x3fc9ca80 0x4208d385:0x3fcb7ed0 0x40387ff2:0x3fcb7ef0 0x4003d021:0x3fcb7f10 |<-CORRUPTED
0x40375b62: panic_abort at /home/quentin/esp/esp-idf/components/esp_system/panic.c:452

0x403806e9: esp_system_abort at /home/quentin/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40386806: abort at /home/quentin/esp/esp-idf/components/newlib/abort.c:38

0x40377183: lock_acquire_generic at /home/quentin/esp/esp-idf/components/newlib/locks.c:130

0x40377279: _lock_acquire_recursive at /home/quentin/esp/esp-idf/components/newlib/locks.c:158

0x40377348: __retarget_lock_acquire_recursive at /home/quentin/esp/esp-idf/components/newlib/locks.c:314 (discriminator 3)

0x420b2d72: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:849 (discriminator 2)

0x420b5ee1: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x420c63e5: esp_log_writev at /home/quentin/esp/esp-idf/components/log/log.c:200

0x403859a1: esp_log_write at /home/quentin/esp/esp-idf/components/log/log.c:210

0x4207431d: wifi_log at ??:?

0x420834b6: ieee80211_sta_new_state at ??:?

0x4207a575: ieee80211_sta_disconnect at ??:?

0x4207a5f8: ieee80211_sta_disconnect at ??:?

0x4207b78c: wifi_disconnect_process at ??:?

0x42084c3a: wifi_station_stop at ??:?

0x4207a1dd: _do_wifi_stop at ??:?

0x4207b638: wifi_stop_process at ??:?

0x42079cee: ieee80211_ioctl_process at ??:?

0x4207c78a: ieee80211_ioctl at ??:?

0x42072c31: esp_wifi_stop at ??:?

0x420041a5: esp_restart at /home/quentin/esp/esp-idf/components/esp_system/esp_system.c:52

0x4200db63: applicative_watchdog_handler at /home/quentin/Code/Boks/boks-link/main/applicative_watchdog/applicative_watchdog.c:20

0x40377629: timer_process_alarm at /home/quentin/esp/esp-idf/components/esp_timer/src/esp_timer.c:451

0x4037768d: timer_alarm_handler at /home/quentin/esp/esp-idf/components/esp_timer/src/esp_timer.c:497

0x40377991: timer_alarm_isr at /home/quentin/esp/esp-idf/components/esp_timer/src/esp_timer_impl_systimer.c:93

0x40376f75: _xt_lowint1 at /home/quentin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240

0x4208d385: pm_coex_tbtt_process at ??:?

0x40387ff2: pm_tbtt_process at ??:?

0x4003d021: pm_on_tbtt in ROM
```

### Steps to reproduce.

i call esp_restart from a FreeRTOS task

Xiehanxin, 2024-01-11:
hi @QuentinFarizon could you provide some detail about the reproduce steps? you call the eap_restart after somo codes? or you can give a simple reproduce  example.

AxelLin, 2024-01-26:
@QuentinFarizon 
Any update for https://github.com/espressif/esp-idf/issues/12955#issuecomment-1886184336 ?
BTW, does it work if test using v5.0.5 release?

KaeLL, 2024-01-26:
Upper layers of the call stack look similar to issues I was having that were fixed by #10649

closed 2024-03-22 by Alvin1Zhang