espressif/esp-idf #12956: Make it possible to idf.py flash into ota_1/ota_2 partitions  after build

damian-kurek-wizzdev, 2024-01-10:
### Is your feature request related to a problem?

I have a factory partition that is smaller than ota_1 and ota_2. When I compile a project with esp-matter it is too big for factory partition. The factory table is already resized that why ota_0 and ota_1 have different size 

# Name | Type | SubType | Offset | Size | Flags
-- | -- | -- | -- | -- | --
# Note: if you have increased the bootloader size | make sure to update the offsets to avoid overlap |   |   |   |  
otadata | data | ota | 0xd000 | 0x2000 |  
phy_init | data | phy | 0xf000 | 0x1000 |  
factory | app | factory | 0x10000 | 0x1F0000 |  
ota_0 | app | ota_0 | 0x200000 | 0x2D0000 |  
ota_1 | app | ota_1 | 0x4D0000 | 0x2E0000 |  
nvs | data | nvs | 0x7B0000 | 0x30000 |  
zb_storage | data | fat | 0x7E0000 | 16K |  
[partition_table.csv](https://github.com/espressif/esp-idf/files/13890290/partition_table.csv)

zb_fct | data | fat | 0x7E4000 | 1K |  

</body>

</html>

### Describe the solution you'd like.

I would like to idf.py flash into ota_1.
idf.py .... build flash --partition ota_1 monitor

### Describe alternatives you've considered.

I need to manually resize partitions during development

### Additional context.

Maybe it works already and I can not find it?

mahavirj, 2024-01-12:
@damian-kurek-wizzdev You may use `parttool.py` for this, please check the command line interface docs [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#command-line-interface)

adokitkat, 2024-01-15:
Hello, you should be able to use `parttool.py` or `esptool.py` to write directly to specified offset in memory.

damian-kurek-wizzdev, 2024-01-17:
@mahavirj 
I checked it, parttool.py works. I think that is annoying to use multiple programs. 
I think that adding it to the idf.py is a good idea and should be considered as a feature to add.

boarchuz, 2024-01-17:
+1, developing for non-factory partitions can be a PITA.
I made a [component](https://github.com/boarchuz/bo_cflash) for flash task customisation, maybe it fits in your workflow @damian-kurek-wizzdev.

mahavirj, 2024-01-18:
@damian-kurek-wizzdev I am checking through our tools team on potential improvements here..