espressif/esp-idf #12958: Channel for console output 修改为 USB Serial/JTAG Controller 后启动

zgjsntsm, 2024-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1288-g5524b692ee

### Espressif SoC revision.

ESP32-C3 (QFN32) (revision v0.3)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

custom borad

### Power Supply used.

USB

### What is the expected behavior?

项目中两个串口均被占用，希望LOG从USB Serial/JTAG 输出。

### What is the actual behavior?

启动报错。

### Steps to reproduce.

1. 克隆 hello_world 示例
2. `idf.py set-target esp32c3`
3. `idf.py menuconfig`  修改 `Channel for console output` 为` USB Serial/JTAG Controller `
4. `idf.py all`
5. `idf.py flash monitor`

### Debug Logs.

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x42001428
0x42001428: esp_panic_handler at D:/Espressif/frameworks/esp-idf-master/components/esp_system/panic.c:452

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xaf0
load:0x403ce710,len:0x2e2c
entry 0x403cc71a
I (24) boot: ESP-IDF v5.3-dev-1288-g5524b692ee-dirty 2nd stage bootloader
I (24) boot: compile time Jan 11 2024 13:54:57
I (26) boot: chip revision: v0.3
I (29) boot.esp32c3: SPI Speed      : 80MHz
I (34) boot.esp32c3: SPI Mode       : DIO
I (39) boot.esp32c3: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=094f0h ( 38128) map
I (101) esp_image: segment 1: paddr=00019518 vaddr=3fc8aa00 size=01234h (  4660) load
I (104) esp_image: segment 2: paddr=0001a754 vaddr=40380000 size=058c4h ( 22724) load
I (116) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=14a98h ( 84632) map
I (134) esp_image: segment 4: paddr=00034ac0 vaddr=403858c4 size=04f88h ( 20360) load
I (141) boot: Loaded app from partition at offset 0x10000
I (141) boot: Disabling RNG early entropy source...
I (143) cpu_start: Unicore app
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x4004c48c  RA      : 0x4004c4b0  SP      : 0x3fcde280  GP      : 0x3fc8b200
0x4004c48c: uart_tx_flush in ROM

0x4004c4b0: uart_tx_wait_idle in ROM

TP      : 0x00000000  T0      : 0x00000000  T1      : 0x00000000  T2      : 0x00000009
S0/FP   : 0x000000ff  S1      : 0x00000050  A0      : 0x000000ff  A1      : 0x3fcde298
A2      : 0x0012ad40  A3      : 0x03ff0000  A4      : 0x0000e000  A5      : 0x60ffe01c
A6      : 0xfa000000  A7      : 0x00000003  S2      : 0x42000000  S3      : 0x3c020000
S4      : 0x3fcde430  S5      : 0x000094f0  S6      : 0xffff0000  S7      : 0x40380254
0x40380254: call_start_cpu0 at D:/Espressif/frameworks/esp-idf-master/components/esp_system/port/cpu_start.c:367

S8      : 0x00020020  S9      : 0x00010020  S10     : 0x00000005  S11     : 0x3fcde3c4
T3      : 0x007d0000  T4      : 0x00030000  T5      : 0x00800000  T6      : 0x00800000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x60ffe01c
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000
```

### More Information.

在idfv5.1和更早版本中将 `Channel for console output` 修改为` USB Serial/JTAG Controller ` 后即可实现 但在master中失败。

mythbuster5, 2024-01-11:
Interesting, will take a look. Btw, have you ever tried 5.2?

mythbuster5, 2024-01-11:
Ok. I already found the issue, fixing..

FL0WL0W, 2024-01-15:
I got this issue with esp-idf 5.2 but not with esp-idf 5.1

I spent alot of time thinking this was a hardware issue. @mythbuster5 any idea when this will be fixed?

mythbuster5, 2024-01-17:
Yes, It's fixing now. The solution will come out in the near future.

closed 2024-01-30 by espressif-bot