espressif/esp-idf #12961: During the RF test, the packet sent by Bluetooth could not be received.

H478417, 2024-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-release-v5.1

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32s3

### Power Supply used.

USB

### What is the expected behavior?

During the RF test, a packet sent by Bluetooth was received

### What is the actual behavior?

No packet is received. phy: rx_num: 0 rx_rssi: 0 is displayed

### Steps to reproduce.

1. The sending device is LE, the rate is 1M, the power is 10dbm, the payload length is 37 byte, the pattern type is 10101010, the channel is 1, the syncw is 0x71746129, and the data is sent in cycles.
2. The receiving end executes the instruction:esp_ble_rx -n 1 -s 0x71764129 -r 0    output : phy: RW LE V9 RX PER
3. Wait a while
4. Run the cmdstop command
expected hehavior:  rx_num and rx_rssi   can receive data.
 actual behavior: rx_num and rx_rssi   are both 0.
 

### Debug Logs.

```plain
phy>
phy> esp_ble_rx -n 1 -s 0x71764129 -r 0
phy> I (17897) phy: RW LE V9 RX PER

phy>
phy>
phy> cmdstop
phy> I (27167) phy: rx_num: 0 rx_rssi: 0

phy>
phy>  esp_ble_rx -n 1 -s 0x71764129 -r 0
phy> I (47417) phy: RW LE V9 RX PER

phy>
phy>
phy> get_rx_result
I (55877) cmd_phy: Desired: 1, Correct: 0, RSSI: 0, flag: 0
phy> get_rx_result
I (61727) cmd_phy: Desired: 1, Correct: 0, RSSI: 0, flag: 0
phy>cmdstop
phy> I (70397) phy: rx_num: 0 rx_rssi: 0

phy>
phy> esp_ble_rx -n 1 -s 0x71764129 -r 0
phy> I (149447) phy: RW LE V9 RX PER

phy>
phy>
phy> cmdstop
phy> I (167987) phy: rx_num: 0 rx_rssi: 0
aa
Unrecognized command
phy>
phy>  esp_ble_rx
W (454097) cmd_phy: Default channel is 1
W (454107) cmd_phy: Default syncw is 0x71764129
W (454107) cmd_phy: Default rate is PHY_BLE_RATE_1M
phy> I (454117) phy: RW LE V9 RX PER

phy>
phy> cmdstop
phy> I (598427) phy: rx_num: 0 rx_rssi: 0

phy>
phy>
phy>

Unrecognized command
phy>
phy>
phy>
phy>
phy>
phy>
phy>
phy> esp_ble_rx
W (975867) cmd_phy: Default channel is 1
W (975867) cmd_phy: Default syncw is 0x71764129
W (975867) cmd_phy: Default rate is PHY_BLE_RATE_1M
phy> I (975887) phy: RW LE V9 RX PER

phy>
```

zhp0406, 2024-01-15:
It seems like tx channel and rx channel are not the same one，in phy_cert test case：

BLE TX channel mapping：https://github.com/espressif/esp-idf/blob/master/components/esp_phy/include/esp_phy_cert_test.h/#L154

BLE RX channel mapping：https://github.com/espressif/esp-idf/blob/master/components/esp_phy/include/esp_phy_cert_test.h/#L166

H478417, 2024-01-16:
@zhp0406  Thank you for your comments, my test V5.1.1 is able to receive data。
 
[image]

H478417, 2024-01-16:
Data can be received using V5.1.1 tests.

closed 2024-01-16 by H478417