espressif/esp-idf #12966: detect esp32 by usb PID and VID in idf.py

thewh1teagle, 2024-01-12:
### Is your feature request related to a problem?

Instead of selecting the right COM port it should find it automatically like raspberry picotool does

### Describe the solution you'd like.

find the usb by PID / VID by default if no COM port provided

### Describe alternatives you've considered.

`-a` auto argument

Dazza0, 2024-01-12:
@thewh1teagle That won't be possible for all of the ESP chips that don't have the integrated USB Serial JTAG peripheral. For example, the ESP32 WROVER Kit uses FTDI UART to USB bridge, thus the COM port is coming form that chip (instead of the ESP32 itself) and users will need to specify the VID/PID of the FTDI chip instead.

Furthermore, the FTDI chip exposes two COM ports (one for UART, one for JTAG) via separate USB interfaces, so users will also need to specify an interface number in addition to the VID/PID of the FTDI.

dobairoland, 2024-01-22:
Based on the available information I think this is already addressed if port is not specified. I'm closing it for inactivity.

closed 2024-01-22 by dobairoland