espressif/esp-idf #12967: use API: esp_log_set_vprintf to redirect logs to the file system and print them may cause system blocking.

Jcc99, 2024-01-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

tag v5.1.1

### Espressif SoC revision.

ESP32C6

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C6 DevKitC-1

### Power Supply used.

External 3.3V

### What is the expected behavior?

system running as normal

### What is the actual behavior?

system was blocked and never resume

### Steps to reproduce.

[fatfsgen_example_main.zip](https://github.com/espressif/esp-idf/files/13916055/fatfsgen_example_main.zip)

I have reproduce the problem using fatfsgen_example code, and the problematic code has been uploaded above.

Jcc99, 2024-01-12:
In my code, when the issue occurs, most tasks cannot be scheduled normally.

igrr, 2024-01-12:
I think this is a duplicate of https://github.com/espressif/esp-idf/issues/5933.

RichFalk, 2024-02-25:
I ran into a similar problem when trying to log to Flash.  The system would lock up and the watchdog timer would not work to restart.  I don't think this is a recursive call problem since I added a semaphore to prevent recursion and it still hung.  It seems that there are some ESP subsystems that can't be called from the esp_log_set_vprintf callback function.

My workaround was to send to a ringbuffer and then store these logging strings into Flash in my main loop in the main thread.

I've found, as a general rule, that callbacks in ESP code are very limited in terms of what you can do from them.  The same limitation of not being able to log to flash also occurs from event handlers (such as when using the WiFi library) and the solution is the same which is to use a ringbuffer and do the Flash logging in the main thread processing loop.

It appears that doing printf to serial works without a problem, but writing to Flash either directly (say via esp_partition_write) or via a file system as you tried to do is a bridge too far.

haberturdeur, 2024-06-03:
Hi @Jcc99 , thank you for the report.
I tried replicating the issue with the provided code, but wasn't able to.
All the tasks got scheduled correctly.
(I was testing on C3, but I assume the problem isn't caused by C6 specifically, but I will test that later.)
Does the problem occur immediately or only after some time?.

Jcc99, 2024-06-03:
As you said, this problem occurs over a period of time, but it is bound to happen.  @haberturdeur