espressif/esp-idf #12970: failing to open gattc connection to ble device (`ESP_GATT_CONN_FAIL_ESTABLISH` and `ESP_GATT_ERROR`)

iniw, 2024-01-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

- IDF version: release/v5.2
- Board: ESP-WROOM-32D

I am trying to connect my board to a DiFluid Microbalance scale that uses BLE. I used the bluedroid BLE gatt_client example as the base for my code with minor adaptations (I have ensured that the problem also happens on the original example), the only slightly substantial one being that I wait for `ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT` to arrive after calling `esp_ble_gap_stop_scanning` before connecting to the device, because i read in a potentially relevant [github issue](https://github.com/espressif/esp-idf/issues/4562#issuecomment-578377457) that it might help.

When calling `esp_ble_gattc_open` it sends back two events: `ESP_GATTC_DISCONNECT_EVT` with `reason` set to `ESP_GATT_CONN_FAIL_ESTABLISH` and `ESP_GATTC_OPEN_EVT` with `status` set to `ESP_GATT_ERROR`, in that order.

Here's a log with relevant traces set to `DEBUG`:

```
I (1595) LUCAS: ESP_GATTC_REG_EVT
I (1605) LUCAS: registering gatt client, app_id: 0, if: 4
I (1605) main_task: Returned from app_main()
I (1615) LUCAS: ble scan started
I (1695) BT_BTM: BTM_CheckAdvData type=0x01
I (1695) BT_BTM: BTM_CheckAdvData type=0x19
I (1695) BT_BTM: BTM_CheckAdvData type=0x03
I (1795) BT_BTM: BTM_CheckAdvData type=0x01
I (1795) BT_BTM: BTM_InqDbRead: bd addr [f412fa10e846]

I (1795) BT_BTM: BTM_CheckAdvData type=0x01
I (1805) BT_BTM: BTM_CheckAdvData type=0x19
I (1805) BT_BTM: BTM_CheckAdvData type=0x03
I (1815) BT_BTM: BTM_CheckAdvData type=0x09
I (1815) LUCAS: found device "Microbalance 302069"
I (1825) LUCAS: ble scan stopped
I (1825) BT_GATT: GATT_Connect gatt_if=4
I (1835) BT_L2CAP: L2CA_ConnectFixedChnl() CID: 0x0004  BDA: f412fa10e846
I (1845) BT_BTM: BTM_InqDbRead: bd addr [f412fa10e846]

I (1845) BT_GATT: GATT_GetConnIdIfConnected status=0

W (2025) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
W (2115) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
W (2205) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
I (2295) BT_GATT: GATT_GetConnIdIfConnected status=0

W (2295) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e
W (2305) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x3e
W (2305) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
E (2315) LUCAS: ESP_GATTC_DISCONNECT_EVT, reason = 0x3e
E (2315) LUCAS: ESP_GATTC_OPEN_EVT - status = 0x85
```

zhp0406, 2024-01-16:
@iniw 
thanks for report the issue，
Is this issue occurring randomly or consistently? How can I reproduce this issue?

zhp0406, 2024-01-17:
@iniw 

I am using version 5.1.2 and have not been able to reproduce the issue. On the master side, I am using the code you provided, and on the slave side, I am using the example 'examples/bluetooth/bluedroid/ble/gatt_server.' The master can initiate connections normally without reporting 0x3e.

If you use the 'gatt_server' example for testing, can you successfully establish a connection? If using 'gatt_server' results in a successful connection, consider checking the following:

1. Verify whether the address and address type match when initiating a connection using `esp_ble_gattc_open`.
2. Check if the slave has any whitelisting or similar configurations. If the slave is set to use a whitelist, only devices on the whitelist may connect.

Additionally, providing logs of the air packets would be beneficial for a more thorough analysis of the issue. Below are my logs:"

```
I (31) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (31) boot: compile time Jan 17 2024 15:27:40
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.1
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1fee4h (130788) map
I (152) esp_image: segment 1: paddr=0002ff0c vaddr=3ffbdb60 size=0010ch (   268) load
I (152) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=80710h (526096) map
I (348) esp_image: segment 3: paddr=000b0738 vaddr=3ffbdc6c size=04710h ( 18192) load
I (355) esp_image: segment 4: paddr=000b4e50 vaddr=40080000 size=187bch (100284) load
I (409) boot: Loaded app from partition at offset 0x10000
I (410) boot: Disabling RNG early entropy source...
I (421) cpu_start: Multicore app
I (421) cpu_start: Pro cpu up.
I (422) cpu_start: Starting app cpu, entry point is 0x40081378
0x40081378: call_start_cpu1 at /home/zhanghaipeng/esp/esp-idf/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (439) cpu_start: Pro cpu start user code
I (439) cpu_start: cpu freq: 160000000 Hz
I (439) cpu_start: Application information:
I (444) cpu_start: Project name:     gatt_client_demo
I (450) cpu_start: App version:      v5.1.2-dirty
I (455) cpu_start: Compile time:     Jan 17 2024 15:27:34
I (461) cpu_start: ELF file SHA256:  8923d23c...
I (467) cpu_start: ESP-IDF:          v5.1.2-dirty
I (472) cpu_start: Min chip rev:     v0.0
I (477) cpu_start: Max chip rev:     v3.99 
I (481) cpu_start: Chip rev:         v3.1
I (486) heap_init: Initializing. RAM available for dynamic allocation:
I (494) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (499) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (506) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (512) heap_init: At 3FFC6EB8 len 00019148 (100 KiB): DRAM
I (518) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (524) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (531) heap_init: At 400987BC len 00007844 (30 KiB): IRAM
I (538) spi_flash: detected chip: gd
I (541) spi_flash: flash io: dio
W (545) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (559) coexist: coex firmware version: b6d5e8c
I (564) app_start: Starting scheduler on CPU0
I (569) app_start: Starting scheduler on CPU1
mI (569) main_task: Started on CPU0
I (579) main_task: Calling app_main()
I (599) BTDM_INIT: BT controller compile version [ec4ac65]
I (599) BTDM_INIT: Bluetooth MAC: 08:b6:1f:ed:4e:42
I (599) phy_init: phy_version 4780,16b31a7,Sep 22 2023,20:42:16
I (989) REPRO: ESP_GATTC_REG_EVT
I (989) REPRO: registering gatt client, app_id: 0, if: 3
I (989) main_task: Returned from app_main()
W (999) REPRO: unhandled ble gap event: 7
I (1009) REPRO: found device "ESP_GATTS_DEMO"
W (1009) REPRO: unhandled ble gap event: 18
I (1109) REPRO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (1109) REPRO: ESP_GATTC_OPEN_EVT - status = 0x0
I (1109) REPRO: successfully opened
W (1309) REPRO: unhandled ble gap event: 20
W (2509) REPRO: unhandled gattc event: 46
W (2589) REPRO: unhandled gattc event: 18
```

zhp0406, 2024-01-17:
@iniw 
Please use gatt_server for further testing to determine if the connection issue is caused by the slave device.

iniw, 2024-01-17:
> @iniw Please use gatt_server for further testing to determine if the connection issue is caused by the slave device.

I am able to open a connection between two esp32's using the repro code and the `gatt_server` example, which leads me to believe that the microbalance server does in fact require some sort of special handling that i am not aware of.

Thank you for your help.

zhp0406, 2024-01-18:
Okay, if there are no other questions, this issue can be closed.

iniw, 2024-01-18:
I spoke to the developers of the device and they told me there are no special rules/limitations to establish a connection, it's supposed to be a normal BLE server like any other.

Using the 'EFR Connect' app on my phone i am able to establish a connection with the device, that app is not associated at all with the scale and so it wouldn't have any special privileges. Here's what the app reports:
[image]
I find the parsing error interesting, perhaps that could be related?

zhp0406, 2024-01-19:
@iniw 
An error with code 0x3e occurred during the connection establishment, and it's also possible that, after initiating the connection, the server did not provide any response. Do you have a way to capture airborne packets?

zhp0406, 2024-01-19:
Have you tested with other versions and other chips of ESP?

iniw, 2024-01-19:
> @iniw 
> An error with code 0x3e occurred during the connection establishment, and it's also possible that, after initiating the connection, the server did not provide any response. Do you have a way to capture airborne packets?

Do you have any recommendations for a program that can sniff bluetooth packets like that?

> Have you tested with other versions and other chips of ESP?

I have tested on two ESP32s, but they were the same model

zhp0406, 2024-01-20:
@iniw 
Do you have a way to capture airborne packets?This process requires the use of a sniffer device.

zhp0406, 2024-01-24:
@iniw 
Any progress? Do you have the ESP32-C3 or ESP32-S3 chip? Can you try using other ESP chips? I suspect the issue may be caused by different frequency-hopping algorithms.

closed 2024-02-28 by Alvin1Zhang

EdwardDeaver, 2025-03-21:
Hello I'm reviving this thread.  I'm using the latest version with an esp32-s3 and can't get the bluetooth ancs example code to work. It flashes fine but will not pair a device. 

"
W (325410) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3d
W (325420) BT_APPL: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 1c:57:dc:XX:XX:XX
W (325430) BT_BTC: btc_dm_ble_auth_cmpl_evt, remove bond in flash bd_addr: 1c57dcXXXXX
I (325440) addr: 1c 57 dc XX XX XX
I (325440) BLE_ANCS: pair status = fail
I (325440) BLE_ANCS: fail reason = 0x66
I (325440) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x3d
I (325450) BLE_ANCS: advertising start success"