espressif/esp-idf #12973: Allow I2C to send empty commands

5p4k, 2024-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

`i2c_master_transmit` explicitly forbids sending 0-length buffers. However, there is in principle nothing wrong with doing so, and it was possible with the legacy driver.

In fact, we relied on this to wake up a PN532 controller (a popular NFC reader module by NXP), since the chip wakes up when it detects its address on I2C, but takes some time before being ready to receive the command body (https://www.nxp.com/docs/en/user-guide/141520.pdf, page 57, paragraph named "variant").

I tried disable the check line below and transmit an empty buffer, and everything worked fine, producing an I2C write packet with no data.

https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/components/esp_driver_i2c/i2c_master.c#L950

So I wonder wethere there is any reason to forbid an empty buffer, and wheter this behavior could be restored?

mythbuster5, 2024-01-15:
If you know when to wake up the sensor, you can  use `i2c_master_probe(i2c_master_bus_handle_t bus_handle, uint16_t address, int xfer_timeout_ms)` instead. It will only address and no buffer.

5p4k, 2024-01-19:
You are right, in fact that was my first attempt, but I did not manage to make that work. So I've looked a bit further into `i2c_master_probe`, and I cannot get it to produce any signal on the line. I'm not sure if it's a duplicate of https://github.com/espressif/esp-idf/issues/12159 ; here is what I have been trying:

```c++
#include <cstdint>
#include <array>
#include <esp_log.h>
#include <driver/i2c_master.h>

namespace pinout {
    constexpr gpio_num_t i2c_scl = GPIO_NUM_17;
    constexpr gpio_num_t i2c_sda = GPIO_NUM_18;

    constexpr std::uint16_t dev_addr = 0x24;

    constexpr i2c_device_config_t dev_config{
            .dev_addr_length = I2C_ADDR_BIT_LEN_7,
            .device_address = pinout::dev_addr,
            .scl_speed_hz = 400'000};

    constexpr i2c_master_bus_config_t bus_config = {
            .i2c_port = -1 /* autoselect */,
            .sda_io_num = pinout::i2c_sda,
            .scl_io_num = pinout::i2c_scl,
            .clk_source = I2C_CLK_SRC_DEFAULT,
            .glitch_ignore_cnt = 7 /* typical */,
            .intr_priority = 0 /* default */,
            .trans_queue_depth = 10,
            .flags = {.enable_internal_pullup = true}};
}

void attempt_probe(i2c_master_bus_handle_t bus, std::uint16_t dev_addr) {
    const auto r = i2c_master_probe(bus, dev_addr, 500);
    if (r == ESP_OK) {
        ESP_LOGI("PROBE", "Successful probe at 0x%x", dev_addr);
    } else {
        ESP_LOGE("PROBE", "Failed probe at 0x%x: %s", dev_addr, esp_err_to_name(r));
    }
}

extern "C" int app_main() {
    i2c_master_bus_handle_t bus_handle = nullptr;
    i2c_master_dev_handle_t dev_handle = nullptr;

    ESP_ERROR_CHECK(i2c_new_master_bus(&pinout::bus_config, &bus_handle));
    ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &pinout::dev_config, &dev_handle));

    ESP_LOGI("PROBE", "Attempting probe.");

    attempt_probe(bus_handle, pinout::dev_addr << 1);
    attempt_probe(bus_handle, (pinout::dev_addr << 1) | 1);
    attempt_probe(bus_handle, pinout::dev_addr);

    ESP_LOGI("PROBE", "Sending a command to see if anything changes.");

    std::array<std::uint8_t, 5> dummy = {0x10, 0x20, 0x30, 0x40, 0x50};
    ESP_ERROR_CHECK(i2c_master_transmit(dev_handle, dummy.data(), dummy.size(), 500));

    ESP_LOGI("PROBE", "Reattempting probe.");

    attempt_probe(bus_handle, pinout::dev_addr << 1);
    attempt_probe(bus_handle, (pinout::dev_addr << 1) | 1);
    attempt_probe(bus_handle, pinout::dev_addr);

    return 0;
}
```

I tested with a logic analyzer but the only signal I get is from the `i2c_master_transmit` call.
I was not sure whether I should have been using the 7 bit device address or not, so I tried all combinations.
Other things I tried:

- probing without installing a device on all addresses
- changing the order in which I probe the different addresses
- probe only after transmitting

No matter what, the only signal produced is the `i2c_master_transmit` (there is nothing else in this recording even though it certainly was started before the probes and ended after returning):

[image]

The `i2c_master_probe` command, however, is always successful:
[image]

Those `DEBUG` messages come from the following modifications to `s_i2c_send_commands`:

```diff
diff --git a/components/esp_driver_i2c/i2c_master.c b/components/esp_driver_i2c/i2c_master.c
index 91e44952d5..c3dbb51871 100644
--- a/components/esp_driver_i2c/i2c_master.c
+++ b/components/esp_driver_i2c/i2c_master.c
@@ -390,2 +390,3 @@ static void s_i2c_send_commands(i2c_master_bus_handle_t i2c_master, TickType_t t
             i2c_master->trans_idx = 0;
+            ESP_LOGW("DEBUG", "Exiting because I cannot take semaphore");
             return;
@@ -398,2 +399,3 @@ static void s_i2c_send_commands(i2c_master_bus_handle_t i2c_master, TickType_t t
             xSemaphoreGive(i2c_master->cmd_semphr);
+            ESP_LOGW("DEBUG", "Exiting because timeout");
             return;
@@ -402,2 +404,3 @@ static void s_i2c_send_commands(i2c_master_bus_handle_t i2c_master, TickType_t t
         if (i2c_master->status == I2C_STATUS_ACK_ERROR) {
+            ESP_LOGW("DEBUG", "Exiting because ack error");
             const i2c_ll_hw_cmd_t hw_stop_cmd = {
@@ -413,6 +416,9 @@ static void s_i2c_send_commands(i2c_master_bus_handle_t i2c_master, TickType_t t
         if (i2c_operation->hw_cmd.op_code == I2C_LL_CMD_WRITE) {
+            ESP_LOGW("DEBUG", "Writing command");
             s_i2c_write_command(i2c_master, i2c_operation, &fifo_fill, &address_fill, NULL);
         } else if (i2c_operation->hw_cmd.op_code == I2C_LL_CMD_READ) {
+            ESP_LOGW("DEBUG", "Reading command");
             s_i2c_read_command(i2c_master, i2c_operation, &fifo_fill, NULL);
         } else {
+            ESP_LOGW("DEBUG", "Ending command");
             s_i2c_start_end_command(i2c_master, i2c_operation, &address_fill, NULL);
```

I'm not entirely sure what this means, but I would expect `i2c_master_probe` to always call `s_i2c_start_end_command`, not at the first call only; either way, `s_i2c_start_end_command` does not produce any switch in the line (although I'm not entirely sure whether that depends on my configuration?).

I'm on an ESP32S3, and esp-idf is on b3f7e2c8a4d354df8ef8558ea7caddc07283a57b.

ogimenezb, 2024-07-01:
@5p4k have you been able to make PN532 work with new I2C implementation?
 have been trying for a week, even tried 5.3-beta2. I'm about to go back to old implementation.

mythbuster5, 2025-04-11:
Long time passed, and we solve lots of issue on new i2c driver and make it robust. Don't know if you still encounter issue on latest master branch. Glad to wait your response, thanks~