espressif/esp-idf #12974: TWAI code is blocked by FreeRTOS task

augustjaubert, 2024-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using the [twai self test code](https://github.com/espressif/esp-idf/tree/v5.1.1/examples/peripherals/twai/twai_self_test) from Espressif on a custom ESP32-S3 board (WROOM module).

### Platform: 
The code has been re-arranged to be used with Arduino, it is just simpler for me than the ESP-IDF or Platformio.

### Hardware: 
I'm using the TCAN1462V-Q1 transceiver, and have gotten it working with the ESP32-C3 XIAO microcontrollers.

### Issue: 
The code installs the driver and starts it in the control task function, but stops at the line

`xSemaphoreTake(ctrl_sem, portMAX_DELAY);  //Wait for TX and RX tasks to finish iteration`

inside the `static void twai_control_task(void *arg)` function, and thus does not reach `Serial.println("checkpoint 2");`.

I put checkpoints (4 total yet) at various places in the code just to get an understanding of where it reaches.

What I think is that the TX and RX RTOS tasks don't finish their iteration, but I don't know why nor what else I can test to better understand what is happening.

I am hoping there is someone that has had similar behaviour with the WROOM module that can tell me what else I can test to figure out the root of the problem.

### **Current output** (with the ESP32-S3 Wroom module):

```c++
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x42021ee6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9a8
load:0x403cc700,len:0x2920
entry 0x403c98b8
[    92][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): XTAL: 40 / 1 = 40 Mhz, APB: 80000000 Hz
Transceiver awake
[   127][I][self_test_twai_receiver.ino:135] setup(): [TWAI Self Test] Driver installed
[   629][I][self_test_twai_receiver.ino:98] twai_control_task(): [TWAI Self Test] Driver started
checkpoint 1
checkpoint 3
``` 

**The expected output** (with the ESP32-C3 XIAO):
```c++
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x4201018c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710
Transceiver awake
checkpoint 1
checkpoint 3
Msg received - Data = 0
Msg received - Data = 1
Msg received - Data = 2
Msg received - Data = 3
Msg received - Data = 4
Msg received - Data = 5
Msg received - Data = 6
Msg received - Data = 7
[...]
Msg received - Data = 95
Msg received - Data = 96
Msg received - Data = 97
Msg received - Data = 98
Msg received - Data = 99
checkpoint 2
``` 
Now I can tell that the XIAO is not running the ESP_LOG commands on the serial monitor and only runs the Serial.print.

The code:

```c++
/*
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

/*
 * The following example demonstrates the self testing capabilities of the TWAI
 * peripheral by utilizing the No Acknowledgment Mode and Self Reception Request
 * capabilities. This example can be used to verify that the TWAI peripheral and
 * its connections to the external transceiver operates without issue. The example
 * will execute multiple iterations, each iteration will do the following:
 * 1) Start the TWAI driver
 * 2) Transmit and receive 100 messages using self reception request
 * 3) Stop the TWAI driver
 */

#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "esp_err.h"
#include "esp_log.h"
#include "driver/twai.h"
#include "soc/gpio_periph.h"
#include "hal/gpio_hal.h"

/* --------------------- Definitions and static variables ------------------ */

//Example Configurations
#define NO_OF_MSGS 100
#define NO_OF_ITERS 3
#define TX_GPIO_NUM GPIO_NUM_41
#define RX_GPIO_NUM GPIO_NUM_42

#define sleepPin 2
#define TX_TASK_PRIO 8    //Sending task priority
#define RX_TASK_PRIO 9    //Receiving task priority
#define CTRL_TSK_PRIO 10  //Control task priority
#define MSG_ID 0x1        //11 bit standard format ID
#define EXAMPLE_TAG "TWAI Self Test"

static const twai_timing_config_t t_config = TWAI_TIMING_CONFIG_50KBITS();
//Filter all other IDs except MSG_ID
static const twai_filter_config_t f_config = { .acceptance_code = (MSG_ID << 21),
                                               .acceptance_mask = ~(TWAI_STD_ID_MASK << 21),
                                               .single_filter = true };
//Set to NO_ACK mode due to self testing with single module
static const twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(TX_GPIO_NUM, RX_GPIO_NUM, TWAI_MODE_NO_ACK);

static SemaphoreHandle_t tx_sem;
static SemaphoreHandle_t rx_sem;
static SemaphoreHandle_t ctrl_sem;
static SemaphoreHandle_t done_sem;

/* --------------------------- Tasks and Functions -------------------------- */

static void twai_transmit_task(void *arg) {
  twai_message_t tx_msg = { .self = 1, .identifier = MSG_ID, .data_length_code = 1 };
  for (int iter = 0; iter < NO_OF_ITERS; iter++) {
    xSemaphoreTake(tx_sem, portMAX_DELAY);
    for (int i = 0; i < NO_OF_MSGS; i++) {
      //Transmit messages using self reception request
      tx_msg.data[0] = i;
      ESP_ERROR_CHECK(twai_transmit(&tx_msg, portMAX_DELAY));
      vTaskDelay(pdMS_TO_TICKS(10));
      delay(1000);
    }
  }
  vTaskDelete(NULL);
}

static void twai_receive_task(void *arg) {
  twai_message_t rx_message;
  
  for (int iter = 0; iter < NO_OF_ITERS; iter++) {
    xSemaphoreTake(rx_sem, portMAX_DELAY);
    for (int i = 0; i < NO_OF_MSGS; i++) {
      //Receive message and print message data
      ESP_ERROR_CHECK(twai_receive(&rx_message, portMAX_DELAY));
      Serial.print("Msg received - Data = ");
      Serial.println(rx_message.data[0]);
      ESP_LOGI(EXAMPLE_TAG, "Msg received - Data = %d", rx_message.data[0]);
    }
    //Indicate to control task all messages received for this iteration
    xSemaphoreGive(ctrl_sem);
  }
  vTaskDelete(NULL);
}

static void twai_control_task(void *arg) {
  xSemaphoreTake(ctrl_sem, portMAX_DELAY);
  for (int iter = 0; iter < NO_OF_ITERS; iter++) {
    //Start TWAI Driver for this iteration
    ESP_ERROR_CHECK(twai_start());
    //Serial.println("Driver started");
    ESP_LOGI(EXAMPLE_TAG, "Driver started");

    //Trigger TX and RX tasks to start transmitting/receiving
    xSemaphoreGive(rx_sem);
    xSemaphoreGive(tx_sem);
    Serial.println("checkpoint 1");
    xSemaphoreTake(ctrl_sem, portMAX_DELAY);  //Wait for TX and RX tasks to finish iteration
    Serial.println("checkpoint 2");
    ESP_ERROR_CHECK(twai_stop());  //Stop the TWAI Driver
                                   //Serial.println("Driver stopped");
    ESP_LOGI(EXAMPLE_TAG, "Driver stopped");
    vTaskDelay(pdMS_TO_TICKS(100));  //Delay then start next iteration
  }
  xSemaphoreGive(done_sem);
  vTaskDelete(NULL);
}
void setup() {
  //gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[TX_GPIO_NUM], PIN_FUNC_GPIO);
  //gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[RX_GPIO_NUM], PIN_FUNC_GPIO);

  Serial.begin(115200);
  pinMode(sleepPin, OUTPUT);
  digitalWrite(sleepPin, LOW);
  //Create tasks and synchronization primitives

  Serial.println("Transceiver awake");
  tx_sem = xSemaphoreCreateBinary();
  rx_sem = xSemaphoreCreateBinary();
  ctrl_sem = xSemaphoreCreateBinary();
  done_sem = xSemaphoreCreateBinary();

  xTaskCreatePinnedToCore(twai_control_task, "TWAI_ctrl", 4096, NULL, CTRL_TSK_PRIO, NULL, tskNO_AFFINITY);
  xTaskCreatePinnedToCore(twai_receive_task, "TWAI_rx", 4096, NULL, RX_TASK_PRIO, NULL, tskNO_AFFINITY);
  xTaskCreatePinnedToCore(twai_transmit_task, "TWAI_tx", 4096, NULL, TX_TASK_PRIO, NULL, tskNO_AFFINITY);
  //Install TWAI driver
  ESP_ERROR_CHECK(twai_driver_install(&g_config, &t_config, &f_config));
  //Serial.println("Driver installed");
  ESP_LOGI(EXAMPLE_TAG, "Driver installed");
  gpio_pulldown_en(RX_GPIO_NUM);
  delay(500);
  //Start control task
  xSemaphoreGive(ctrl_sem);
  //Wait for all iterations and tasks to complete running
  Serial.println("checkpoint 3");
  xSemaphoreTake(done_sem, portMAX_DELAY);
  Serial.println("checkpoint 4");
  //digitalWrite(sleepPin, HIGH);
  Serial.println("Transceiver sleeping");
  //Uninstall TWAI driver
  ESP_ERROR_CHECK(twai_driver_uninstall());
  //Serial.println("Driver uninstalled");
  ESP_LOGI(EXAMPLE_TAG, "Driver uninstalled");

  //Cleanup
  vSemaphoreDelete(tx_sem);
  vSemaphoreDelete(rx_sem);
  vSemaphoreDelete(ctrl_sem);
  vQueueDelete(done_sem);
}
void loop() {
}
``` `

Dazza0, 2024-01-15:
> What I think is that the TX and RX RTOS tasks don't finish their iteration, but I don't know why nor what else I can test to better understand what is happening.

Because your RX task isn't receiving any messages. Since this is a loopback test and no messages are being received, the TX task is filling up the TX queue and thus blocking indefinitely. Thus, both TX and RX tasks are blocked indefinitely.

Please check that your transceiver is connected properly to your designated TX and RX GPIOs.

augustjaubert, 2024-01-20:
Thank you so much for your help! @Dazza0 @wanckl 

> What I think is that the TX and RX RTOS tasks don't finish their iteration, but I don't know why nor what else I can test to better understand what is happening.
>
> Because your RX task isn't [...]
This is very interesting, I am glad it had an explanation.

> Please check that your transceiver is connected properly to your designated TX and RX GPIOs.
>
You're completely right, I thought all my connections were fine as I had tested them multiple times with a continuity test. Nothing problematic came up, but then yesterday I just decided to heat up all the solder joints and add a bit of flux just to make it a bit cleaner, and that seems to have fixed it. I am a bit embarrassed but also relieved that that was the problem, because that means the ESP module does not have a defect.

Thank you!

closed 2024-01-20 by augustjaubert