espressif/esp-idf #12975: wifi station: Disconnect reason : 16

zzzh, 2024-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Espressif SoC revision.

ESP32-C3 (revision 4)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3-CORE

### Power Supply used.

USB

### What is the expected behavior?

wifi station keep working or recover after break

### What is the actual behavior?

connection broke and not recover

### Steps to reproduce.

Did not find a way to reproduce

### Debug Logs.

```plain
I (13136665) wifi:state: run -> init (10c0)
I (13136665) wifi:pm stop, total sleep time: 11764344979 us / 13135727690 us

W (13136665) wifi:<ba-del>idx
W (13136665) wifi:<ba-del>idx
I (13136675) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
E (13136685) wifi station: Disconnect reason : 16
W (13146925) wifi:Haven't to connect to a suitable AP now!
W (13166925) wifi:Haven't to connect to a suitable AP now!
W (13186925) wifi:Haven't to connect to a suitable AP now!
W (13206925) wifi:Haven't to connect to a suitable AP now!
W (13226925) wifi:Haven't to connect to a suitable AP now!
W (13246925) wifi:Haven't to connect to a suitable AP now!
W (13266925) wifi:Haven't to connect to a suitable AP now!
W (13286925) wifi:Haven't to connect to a suitable AP now!
W (13306925) wifi:Haven't to connect to a suitable AP now!
W (13326925) wifi:Haven't to connect to a suitable AP now!
W (13346925) wifi:Haven't to connect to a suitable AP now!
W (13366925) wifi:Haven't to connect to a suitable AP now!
W (13386925) wifi:Haven't to connect to a suitable AP now!
W (13406925) wifi:Haven't to connect to a suitable AP now!
W (13426925) wifi:Haven't to connect to a suitable AP now!
```

### More Information.

esp_wifi_sta_get_ap_info() was used to check wifi status every 20 seconds

nishanth-radja, 2024-01-17:
@zzzh how often does this issue occur and can we have wireless sniffer capture when the issue occurs ?
what is the security which is being used for connection and which AP model is being used?

AxelLin, 2024-03-01:
@zzzh Any feedback for the comment https://github.com/espressif/esp-idf/issues/12975#issuecomment-1895470374 ?

closed 2025-04-07 by Alvin1Zhang