espressif/esp-idf #12977: Why is the DS peripheral configured for rsa_alt rather than rsa?

owen63, 2024-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp-idf v5.12, mbedtls v3.5.0
I have had issues connecting ESP32-S3 to ThingsBoard using TLS 1.3 mutual authentication and DS. The problem was that the DS peripheral is configured with rsa_alt (see esp_mbedtls_init_pk_ctx_for_ds) and rsa_alt only supports MBEDTLS_PK_RSA, not MBEDTLS_PK_RSASSA_PSS.
This conflicts with mbedtls which does not support MBEDTLS_PK_RSA (legacy?) in mbedtls_ssl_tls13_check_sig_alg_cert_key_match() and mbedtls_ssl_tls13_sig_alg_for_cert_verify_is_supported().
Can the  DS peripheral be configured for rsa (which supports MBEDTLS_PK_RSA and MBEDTLS_PK_RSASSA_PSS)?

AdityaHPatwardhan, 2024-01-29:
Hi @owen63,
Thanks for the issue.
Can you share any error logs that you observed while trying the connection with TLS 1.3?
If I understand correctly, for mbedTLS the ALT layers are to be replaced in future by the PSA Crypto layer but the current layer are supposed to be functioning correctly.
If posible can you share a small reproducible code that I can quickly try at my end?
Thanks,
Aditya

owen63, 2024-02-01:
Hi Aditya,

I am putting together a sample project now...

owen63, 2024-02-01:
Cannot reproduce.

closed 2024-02-01 by owen63