espressif/esp-idf #12978: Connect ESP32-C3 to STM32

shlomo-edu, 2024-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I installed the ESP-AT binary on my ESP32-C3-DevKit1 and tested it from a YAT terminal.
It works great.
Then I connected its module to my STM32 via TX and RX pins:
   PD2     ------> USART5_RX : GPIO_PIN_2
   PC10    ------> USART4_TX : GPIO_PIN_5
Now I send a command: AT_GMR from the STM via the UART using the following line:
`HAL_UART_Transmit_IT(&huart4, buff, Len);`
And I expect to get the result (same result that I see when I connected the terminal via the DevKit) in the callback function:
`HAL_UART_RxCpltCallback`
But I don't get anything.
Also, I don't understand why I can't place a breakpoint in the callback function.
I am using IAR workbench IDE, and I can place breakpints everywhere, but not in the callbacks.

Dazza0, 2024-01-15:
> Now I send a command: AT+GMR from the STM via the UART using the following line:
> HAL_UART_Transmit_IT(&huart4, buff, Len);
> And I expect to get the result (same result that I see when I connected the terminal via the DevKit) in the callback function:
> HAL_UART_RxCpltCallback

- Since you are transmitting from the STM32, shouldn't you be expecting the `HAL_UART_TxCpltCallback` to be called?

> But I don't get anything.

- Please use a logic analyzer to check what bytes are actually being sent/received between the STM32 and the ESP

> Also, I don't understand why I can't place a breakpoint in the callback function.
> I am using IAR workbench IDE, and I can place breakpints everywhere, but not in the callbacks.

You will need to check with STM regarding their tooling and IDE

ustccw, 2024-01-16:
@shlomo-edu could you please double check if there a `\r\n` terminator appending after `AT+GMR` string. (You can check this by connecting another Dupont Line from the UART RX of ESP32-C3 to the `USB to serial converter`, and see what happened in the ESP32-C3 RX)