espressif/esp-idf #12979: Ways to configure multiple key for secure boot

eme-bennytang, 2024-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi 

I am trying to set up a multi-boot environment where I can use the secure boot feature and accept both image from Key A / Key B.

Because I don't have much units on my hand and as you know the secure boot feature involves some settings being brunt to efuse. I want to confirm the setup beforehand so that I can go smoothly.

ESP-IDF: v5.1.2

Enable below options in the SDK config in the project
[image]
CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS=y  (to avoid all slots being revoke after 1st time boot up)

Steps:
1. Sign the bootloader/application from the project with Key A
2. Program the Key A signed bootloader/application to a default unit
3. 1st time power up after programming (should automatically burn the Key A's public key digest inside the signature block to the efuse)
4. Sign the same bootloader/application with Key B
5. Program the Key B's public key digest to efuse manually by espefuse command
6. Program the key B signed bootloader/application to the unit
7. The unit should be able to power up nice and smoothly

Can you help to confirm if this is the procedures needed?

Thanks

Harshal5, 2024-01-16:
Hello @eme-bennytang,

You could refer to the documentation guide that mentions the usage of multiple keys in secure boot v2: [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/security/secure-boot-v2.html#multiple-keys).

> CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS=y (to avoid all slots being revoked after 1st time boot up)

This config is only important if you plan to boot up the device when the binaries are not signed with all the keys.
For example, if you boot up the device by flashing binaries signed using just one key but need to support multiple (> 1) secure boot keys in the future.

If the "first" boot (or booting before the binaries are signed with all the required keys) is important to your case, you need to build the binaries by enabling the config `CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS`, else, I would prefer to first sign the binaries with all the required keys and thus you do not need to enable the config `CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS` or burn the key digests for other keys in the efuses manually.

Below are the steps to enable secure boot v2 by signing the binaries with multiple keys before the first boot-up:

1. Build the project. (You do not need to enable the config `CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS`)
2. The build system will already generate signed binaries for the one secure boot key that was mentioned during the build (`CONFIG_SECURE_BOOT_SIGNING_KEY`).
3. Sign the bootloader and application binaries with all the other keys required. 
```
espsecure.py sign_data --version 2 --k secure_boot_signing_key_1.pem secure_boot_signing_key_2.pem --append_signatures build/bootloader/bootloader.bin
```
and
```
espsecure.py sign_data --version 2 --k secure_boot_signing_key_1.pem secure_boot_signing_key_2.pem --append_signatures build/app.bin
```
4. Finally you can flash the signed binaries to the device. The device will boot up and burn the efuses by itself.

<hr/>

Recently we have also introduced a command `idf.py qemu` in the ESP-IDF (master branch) which enable users to emulate the target on the host machine. You could check the above workflow using `qemu` just to be sure.

1. Enable the config `CONFIG_SECURE_BOOT_FLASH_BOOTLOADER_DEFAULT`, this would flash the bootloader binary with along with the app and partition-table binaries using the default flashing command (This is needed just for the QEMU use case).
2. Build and sign the binaries (Steps 1-3 as mentioned above).
3. Run `idf.py qemu`

You could check the logs to confirm the efuse key blocks used.
Example:
If the binaries are signed with three keys:
> I (80) efuse: Writing EFUSE_BLK_KEY0 with purpose 9
I (81) efuse: Writing EFUSE_BLK_KEY1 with purpose 10
I (81) efuse: Writing EFUSE_BLK_KEY2 with purpose 11

In case you would like to check the efuses summary for the above emulation as well, you need to run the following command:

```
qemu-system-riscv32 -nographic -machine esp32c3 -global driver=esp32c3.gpio,property=strap_mode,value=0x02 -drive file=build/qemu_efuse.bin,if=none,format=raw,id=efuse -global driver=nvram.esp32c3.efuse,property=drive,value=efuse -serial tcp::5555,server,nowait
```

Check efuse summary:
```
espefuse.py --port socket://localhost:5555 --before=no_reset summary
```

Harshal5, 2024-01-17:
> I think that should do the trick, correct?

Correct, just make sure that in your case, you'll need to build your firmware/binaries with the config `CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS` enabled.

Also for a production use case, make sure you revoke all the unused key slots, i.e. even if you program a development key, you are still left with one empty key slot, as we support up to three secure boot keys, so you should revoke the last key slot.

> Thanks for the introduction for the qemu, is that for simulating the efuse stuff? Since I think lots of people have my problems.... (not enough units to try out the secure boot/flash encryption stuff....)
My IDF is v5.1.2 and seems not support yet, I may try that if we get a chance to upgrade the IDF further.

Yes, you can use QEMU to emulate the target ESP32-C3 completely, including the efuses use-case.
The command `idf.py qemu` has been recently integrated into ESP-IDF, so you would need to check out the master branch to try it out.

Thank you!

eme-bennytang, 2024-01-19:
Hi @Harshal5,

I am back with sad news.
Even with CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS=y
After 1st power up, the DUT stil revoke 2nd/3rd slots for the secure boot keys.

Can you help to confirm this way is not actually viable?
Or did I miss some config?

Harshal5, 2024-01-19:
Hi @eme-bennytang,

Ohh got it. So sorry, but seems like I missed out to mention that you will need to enable secure boot externally, in case you do not want the revocation bits to get set as enabling the config `CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS` ([ref](https://github.com/espressif/esp-idf/blob/master/components/bootloader/Kconfig.projbuild#L984)) just makes sure that the **application** (not the bootloader) does not revoke the unused key blocks and in your case secure boot gets enabled by the bootloader during the first boot up, the unused key blocks are revoked ([ref](https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/secure_boot_v2/secure_boot.c#L295-L304)).

I have verified the below steps using QEMU:

1. Build the project with the config enabled.
2. Enable Secure Boot externally ([ref](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/security/host-based-security-workflows.html#enable-secure-boot-v2-externally)) by programming efuses externally.
   1. Generate public key digest:
`espsecure.py digest_sbv2_public_key --keyfile secure_boot_signing_key.pem --output digest.bin`
    2. Burn key digest in the efuses:
`espefuse.py --port $ESPPORT --chip esp32c3 burn_key BLOCK_KEY0 digest.bin SECURE_BOOT_DIGEST0`
    3. Enable secure boot v2
`espefuse.py --port $ESPPORT --chip esp32c3 burn_efuse SECURE_BOOT_EN`

5. Boot the chip and you should get the below logs:

> W (94) secure_boot: Unused SECURE_BOOT_DIGEST1 should be revoked. It will not be fixed due to the config
W (95) secure_boot: Unused SECURE_BOOT_DIGEST2 should be revoked. It will not be fixed due to the config.

KaeLL, 2024-01-19:
Seems something worthy of being added to the docs.

eme-bennytang, 2024-01-22:
Hi @Harshal5,

Just so I can understand my failed case correctly:
1. Unused key slots revoke action will be done in BOTH **Bootloader** and **App**
2. **App** will not revoke the unused key slots due to CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS
3. **Bootloader** will somehow NOT recognise this config and STILL revoke the unused key slots
4. Unused key slots all revoked in **Bootloader** stage

I think I may go with your previous suggested route then
1. Sign both signature in the initial bootloader
2. Revoke one of the key later stage (by script espefuse.py, or by in code actions, I believe both should be viable)

Thanks so much with the great help. 
Sometimes the IDF document doesn't provide such details and those details are exactly what we need.

Harshal5, 2024-01-22:
@eme-bennytang,

Right, you could first sign the binaries with all the keys needed and then the keys not needed could be revoked using `espefuse.py` or in-code actions.

> Bootloader will somehow NOT recognise this config and STILL revoke the unused key slots

The bootloader intentionally does not recognize it because keeping the unused key slots un-revoked is a security hazard. But yes this seems to be missing in the documentation and is important to be mentioned. Will raise an internal request to fix this.
Closing this MR, feel free to re-open if needed!
Thanks!