espressif/esp-idf #12981: Error: esp_https_ota: Complete headers were not received

CiroL1, 2024-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am attempting to perform an Over-The-Air (OTA) update with an ESP32 DevKit V1 (4 MB flash) using Cloudflare R2 as an S3 Bucket for the .bin file and Cloudflare Workers as an intermediary. However, I consistently encounter the following error:

`` E (10769) esp_https_ota: Complete headers were not received
  E (10779) simple_ota_example: Firmware update failed``
 
I have tried both with and without SSL, as well as with HTTP/HTTPS, but the error remains.

Configuration:
ESP32 DevKit V1 with 4 MB flash
170 KB .bin file
ESP-IDF version 5.1.1
  
Partitions:
Name,     Type, SubType, Offset,  Size,   Flags
nvs,        data, nvs,     0xb000,  0x4000,
otadata,    data, ota,      ,        0x2000,
phy_init,   data, phy,      ,        0x1000,
factory,    app,  factory,  ,        1M,
ota_0,      app,  ota_0,    ,        1M,
ota_1,      app,  ota_1,    ,        1M,
spiffs,     data, spiffs,   ,        512K

Code: 
#include "esp_system.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_ota_ops.h"
#include "esp_http_client.h"
#include "esp_https_ota.h"
#include "esp_event.h"
#include "esp_crt_bundle.h"

#include "mqtt.h"

char* URL_BIN;

static const char *TAG = "simple_ota_example";

esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
    switch (evt->event_id) {
    case HTTP_EVENT_ERROR:
        ESP_LOGD(TAG, "HTTP_EVENT_ERROR");
        break;
    case HTTP_EVENT_ON_CONNECTED:
        ESP_LOGD(TAG, "HTTP_EVENT_ON_CONNECTED");
        break;
    case HTTP_EVENT_HEADER_SENT:
        ESP_LOGD(TAG, "HTTP_EVENT_HEADER_SENT");
        break;
    case HTTP_EVENT_ON_HEADER:
        ESP_LOGD(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
        break;
    case HTTP_EVENT_ON_DATA:
        ESP_LOGD(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
        break;
    case HTTP_EVENT_ON_FINISH:
        ESP_LOGD(TAG, "HTTP_EVENT_ON_FINISH");
        break;
    case HTTP_EVENT_DISCONNECTED:
        ESP_LOGD(TAG, "HTTP_EVENT_DISCONNECTED");
        break;
    case HTTP_EVENT_REDIRECT:
        ESP_LOGD(TAG, "HTTP_EVENT_REDIRECT");
        break;
    }
    return ESP_OK;
}

static esp_err_t _http_client_init_cb(esp_http_client_handle_t http_client)
{
    esp_err_t err = esp_http_client_set_header(http_client, "X-API-Key", "test-api-key");
    return err;
}

static void rollback_and_reboot()
{
    esp_ota_img_states_t ota_state;
    if (esp_ota_get_state_partition(esp_ota_get_running_partition(), &ota_state) == ESP_OK) {
        ESP_LOGW(TAG, "Rolling back to previous version");
        const esp_partition_t* next_partition = esp_ota_get_next_update_partition(NULL);
        if (next_partition != NULL) {
            esp_ota_set_boot_partition(next_partition);
            esp_restart();
        } else {
            ESP_LOGE(TAG, "Failed to get information");
        }
    } else {
        ESP_LOGE(TAG, "Failed to get OTA state for rollback");
    }
}

static void OTA_init()
{
    esp_http_client_config_t config = {
        .url = URL_BIN, 
        .event_handler = _http_event_handler,
        .keep_alive_enable = true,
        .transport_type = HTTP_TRANSPORT_OVER_SSL,
        .crt_bundle_attach = esp_crt_bundle_attach
    };

    esp_https_ota_config_t ota_config = {
        .http_config = &config,
        .http_client_init_cb = _http_client_init_cb
    };

    ESP_LOGI(TAG, "Attempting to download update from %s", config.url);

    esp_err_t ota_status = esp_https_ota(&ota_config);

    if (ota_status == ESP_OK) {
        ESP_LOGI(TAG, "OTA Succeed, Rebooting...");
        esp_restart();
    } else {
        ESP_LOGE(TAG, "Firmware update failed");
        char error_message[50];
        snprintf(error_message, sizeof(error_message), "OTA failed with error %d", ota_status);
        //MQTT_publish_errors("test", "test", error_message);

        rollback_and_reboot();
    }
    while (1) {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void OTA_start(char* url_bin)
{
    URL_BIN = url_bin;
    OTA_init();
}

kayzen9, 2024-01-18:
can download file using curl from server? correct permission set on server?

see https://github.com/espressif/esp-idf/issues/5736

kayzen9, 2024-01-19:
$ hexdump -C firmware.bin 
00000000  5b 6f 62 6a 65 63 74 20  4f 62 6a 65 63 74 5d     |[object Object]|
0000000f

file is broken ???

CiroL1, 2024-01-19:
Do you have a working .bin file to try?

kayzen9, 2024-01-20:
u wanna try ota function and do not have working bin ??.

build any app from idf like hello world and upload its binary to server

hmalpani, 2024-01-22:
Hello @CiroL1 
While trying to download the firmware from the URL you provided, i got HTTP status 403.
```
$ curl -k https://updates-dev.autoplants.workers.dev/AP-CRW-20/firmware.bin
{"status":403,"message":"AUTH_REQUIRED"
```

I think you have not provided the necessary permissions.

closed 2024-02-05 by mahavirj