espressif/esp-idf #12982: lwip_debug.c has log format issues resulting in compilation failure

atanisoft, 2024-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.2 tip -- e49823f10c6b6b8dadbad376002ea2bc531b7b0f

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

With `LWIP_DEBUG` disabled I'd expect no compilation failures from LWIP in lwip_debug.c.

### What is the actual behavior?

Compilation fails due to the following:

```
In file included from /scratch/esp-idf-v5.2/components/lwip/port/debug/lwip_debug.c:16:
/scratch/esp-idf-v5.2/components/lwip/port/debug/lwip_debug.c: In function 'dbg_lwip_tcp_pcb_one_show':
/scratch/esp-idf-v5.2/components/lwip/port/debug/lwip_debug.c:93:67: error: format '%d' expects argument of type 'int', but argument 6 has type 'tcpwnd_size_t' {aka 'long unsigned int'} [-Werror=format=]
   93 |     ESP_LWIP_LOGI("cwnd=%d ssthreash=%d", pcb->cwnd, pcb->ssthresh);
      |                                                                   ^
/scratch/esp-idf-v5.2/components/log/include/esp_log.h:308:59: note: in definition of macro 'LOG_FORMAT'
  308 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                                           ^~~~~~
/scratch/esp-idf-v5.2/components/log/include/esp_log.h:464:41: note: in expansion of macro 'ESP_LOG_LEVEL'
  464 |         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                         ^~~~~~~~~~~~~
/scratch/esp-idf-v5.2/components/log/include/esp_log.h:368:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  368 | #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
/scratch/esp-idf-v5.2/components/lwip/port/include/lwipopts.h:1561:41: note: in expansion of macro 'ESP_LOGI'
 1561 | #define ESP_LWIP_LOGI(...)              ESP_LOGI("lwip", __VA_ARGS__)
      |                                         ^~~~~~~~
/scratch/esp-idf-v5.2/components/lwip/port/debug/lwip_debug.c:93:5: note: in expansion of macro 'ESP_LWIP_LOGI'
   93 |     ESP_LWIP_LOGI("cwnd=%d ssthreash=%d", pcb->cwnd, pcb->ssthresh);
      |     ^~~~~~~~~~~~~
/scratch/esp-idf-v5.2/components/lwip/port/debug/lwip_debug.c:93:67: error: format '%d' expects argument of type 'int', but argument 7 has type 'tcpwnd_size_t' {aka 'long unsigned int'} [-Werror=format=]
   93 |     ESP_LWIP_LOGI("cwnd=%d ssthreash=%d", pcb->cwnd, pcb->ssthresh);
      |                                                                   ^
/scratch/esp-idf-v5.2/components/log/include/esp_log.h:308:59: note: in definition of macro 'LOG_FORMAT'
  308 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                                           ^~~~~~
/scratch/esp-idf-v5.2/components/log/include/esp_log.h:464:41: note: in expansion of macro 'ESP_LOG_LEVEL'
  464 |         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                         ^~~~~~~~~~~~~
/scratch/esp-idf-v5.2/components/log/include/esp_log.h:368:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  368 | #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
/scratch/esp-idf-v5.2/components/lwip/port/include/lwipopts.h:1561:41: note: in expansion of macro 'ESP_LOGI'
 1561 | #define ESP_LWIP_LOGI(...)              ESP_LOGI("lwip", __VA_ARGS__)
      |                                         ^~~~~~~~
/scratch/esp-idf-v5.2/components/lwip/port/debug/lwip_debug.c:93:5: note: in expansion of macro 'ESP_LWIP_LOGI'
   93 |     ESP_LWIP_LOGI("cwnd=%d ssthreash=%d", pcb->cwnd, pcb->ssthresh);
      |     ^~~~~~~~~~~~~
/scratch/esp-idf-v5.2/components/lwip/port/debug/lwip_debug.c:93:1: error: format '%d' expects argument of type 'int', but argument 6 has type 'tcpwnd_size_t' {aka 'long unsigned int'} [-Werror=format=]
[... 432 lines cut ...]
     ESP_LWIP_LOGI("local_port=%d, remote_port=%d", pcb->local_port, pcb->remote_port);
     ESP_LWIP_LOGI("flags=%x", pcb->flags);
     ESP_LWIP_LOGI("pooltmr=%d pollinterval=%d, last_tmr=%d tmr=%" PRIu32 " rtmer=%d", pcb->polltmr, pcb->pollinterval, pcb->last_timer, pcb->tmr, pcb->rtime);
-    ESP_LWIP_LOGI("recv_nxt=%" PRIu32 " recv_wnd=%d recv_ann_wnd=%d recv_ann_right_edge=%" PRIu32, pcb->rcv_nxt, pcb->rcv_wnd, pcb->rcv_ann_wnd, pcb->rcv_ann_right_edge);
+    ESP_LWIP_LOGI("recv_nxt=%" PRIu32 " recv_wnd=%" PRIu32 " recv_ann_wnd=%" PRIu32 " recv_ann_right_edge=%" PRIu32, pcb->rcv_nxt, pcb->rcv_wnd, pcb->rcv_ann_wnd, pcb->rcv_ann_right_edge);
     ESP_LWIP_LOGI("mss=%d", pcb->mss);
     ESP_LWIP_LOGI("rttest=%" PRIu32 " rtseq=%" PRIu32 " sa=%d sv=%d", pcb->rttest, pcb->rtseq, pcb->sa, pcb->sv);
     ESP_LWIP_LOGI("rto=%d nrtx=%d", pcb->rto, pcb->nrtx);
@@ -90,11 +90,11 @@ static void dbg_lwip_tcp_pcb_one_show(struct tcp_pcb* pcb)
 #if ESP_PER_SOC_TCP_WND
     ESP_LWIP_LOGI("per_soc_window=%d per_soc_snd_buf=%d", pcb->per_soc_tcp_wnd, pcb->per_soc_tcp_snd_buf);
 #endif
-    ESP_LWIP_LOGI("cwnd=%d ssthreash=%d", pcb->cwnd, pcb->ssthresh);
+    ESP_LWIP_LOGI("cwnd=%" PRIu32 " ssthreash=%" PRIu32, pcb->cwnd, pcb->ssthresh);
     ESP_LWIP_LOGI("snd_next=%" PRIu32 " snd_wl1=%" PRIu32 " snd_wl2=%" PRIu32, pcb->snd_nxt, pcb->snd_wl1, pcb->snd_wl2);
-    ESP_LWIP_LOGI("snd_lbb=%" PRIu32 " snd_wnd=%d snd_wnd_max=%d", pcb->snd_lbb, pcb->snd_wnd, pcb->snd_wnd_max);
+    ESP_LWIP_LOGI("snd_lbb=%" PRIu32 " snd_wnd=%" PRIu32 " snd_wnd_max=%" PRIu32, pcb->snd_lbb, pcb->snd_wnd, pcb->snd_wnd_max);
     //ESP_LWIP_LOGI("acked=%d", pcb->acked);
-    ESP_LWIP_LOGI("snd_buf=%d snd_queuelen=%d", pcb->snd_buf, pcb->snd_queuelen);
+    ESP_LWIP_LOGI("snd_buf=%" PRIu32 " snd_queuelen=%d", pcb->snd_buf, pcb->snd_queuelen);
     ESP_LWIP_LOGI("unsent_oversize=%d", pcb->unsent_oversize);
     ESP_LWIP_LOGI("keep_idle=%" PRIu32 " keep_intvl=%" PRIu32 " keep_cnt=%" PRIu32, pcb->keep_idle, pcb->keep_intvl, pcb->keep_cnt);
     ESP_LWIP_LOGI("persist_cnt=%d persist_backoff=%d", pcb->persist_cnt, pcb->persist_backoff);
```
and now it compiles cleanly.

atanisoft, 2024-01-15:
This appears to be related to https://github.com/espressif/esp-idf/commit/c510560890de8631352fb5aa0d0361064f9c6d46 with some of the data types changing since this was merged.

atanisoft, 2024-01-15:
@david-cermak can you review?

mhdong, 2024-01-16:
Hi @atanisoft 
The reason is 
`#if LWIP_WND_SCALE
typedef u32_t tcpwnd_size_t;
#else
typedef u16_t tcpwnd_size_t;
#endif`
The patch code you provided is correct. Thank you very much and we will fix it as soon as possible.

atanisoft, 2024-01-16:
Thanks @mhdong, this will need some special care as with `LWIP_WND_SCALE` disabled the current code is correct and working. I'm not sure which approach should be used here.

Also if possible, when this is fixed can it be backported to 5.2?

closed 2024-02-04 by espressif-bot