espressif/esp-idf #12983: Websocket server lost connect to ESP32 module

HuyBui91, 2024-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.1.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I'm using ESP WROOM 32 with "esp_websocket_client.h" to make connection to websocket server. I assume that after ESP32 make connect to websocket server, Module and Server always keep connection and available receive data from Server.

### What is the actual behavior?

After establish connection, ESP32 already recieved msg from Server which noitice that connection is successfull (Json string) and Server record connection under device ID of ESP32. But after about ~5s later, I checked at server by end point through postman and saw that lost connection ( Device ID dissappare) and I can't recieve msg from Server. This issue occur randomly, sometime I try press button "RST" or disconnect USB and connect again, The connection always keep.

### Steps to reproduce.

`void websocket_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
{
    TAG = "WEBSOCKET_EVENT";
    esp_websocket_event_data_t *data = (esp_websocket_event_data_t *)event_data;
    switch (event_id)
    {
    case WEBSOCKET_EVENT_CONNECTED:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_CONNECTED");
        break;
    case WEBSOCKET_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_DISCONNECTED");
        break;
    case WEBSOCKET_EVENT_DATA:
        if (data->op_code == 1)
        {
            ESP_LOGI(TAG, "WEBSOCKET_EVENT_DATA");
            ESP_LOGI(TAG, "Received opcode=%d", data->op_code);
            ESP_LOGW(TAG, "WEBSOCKET Received=%.*s", data->data_len, (char *)data->data_ptr);
            ESP_LOGW(TAG, "Total payload length=%d, data_len=%d, current payload offset=%d\r\n", data->payload_len, data->data_len, data->payload_offset);
            // Parse json funct ion
            parse_json((char *)data->data_ptr);
        }

        break;
    case WEBSOCKET_EVENT_ERROR:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_ERROR");
        break;
    }
}
esp_websocket_client_handle_t client_Socket;
esp_err_t Websocket_Start()
{
    TAG = "WEBSOCKET_START";
    esp_err_t ret;
    char websocket_url[100];
    snprintf(websocket_url, sizeof(websocket_url), "%s%s", BASE_WEBSOCKET_URL, CHIP_ID);
    ESP_LOGI(TAG, "Connecting to %s", websocket_url);
    esp_websocket_client_config_t websocket_cfg = {
        .uri = websocket_url,
        .disable_auto_reconnect = false,
        .pingpong_timeout_sec = 10,
        .disable_pingpong_discon = true,
        .reconnect_timeout_ms = 10000,
        .network_timeout_ms = 30000};
    client_Socket = esp_websocket_client_init(&websocket_cfg);
    ret = esp_websocket_register_events(client_Socket, WEBSOCKET_EVENT_ANY, websocket_event_handler, (void *)client_Socket);
    ret = esp_websocket_client_start(client_Socket);
    while (1)
    {
        vTaskDelay(500 / portTICK_PERIOD_MS); // Adjust as needed
    }
    return ret;
}`

gabsuren, 2024-04-17:
@HuyBui91 thank you for your question and sorry for a late response.
Have you tried to check error codes, something like this?
```
case WEBSOCKET_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_DISCONNECTED");
        log_error_if_nonzero("HTTP status code",  data->error_handle.esp_ws_handshake_status_code);
        if (data->error_handle.error_type == WEBSOCKET_ERROR_TYPE_TCP_TRANSPORT) {
            log_error_if_nonzero("reported from esp-tls", data->error_handle.esp_tls_last_esp_err);
            log_error_if_nonzero("reported from tls stack", data->error_handle.esp_tls_stack_err);
            log_error_if_nonzero("captured as transport's socket errno",  data->error_handle.esp_transport_sock_errno);
        }
        break;
 ```
You can find an example below:
  https://github.com/espressif/esp-protocols/blob/bd6e12050928464b4357015c7c8c1b215432d580/components/esp_websocket_client/examples/target/main/websocket_example.c#L79C3-L87C15
  
  
Regards,
Suren

gabsuren, 2024-05-29:
@HuyBui91 closing the issue. Please feel free to re-open it if you have any other questions

closed 2024-05-29 by gabsuren