espressif/esp-idf #12984: Compile error when CONFIG_ESP_CONSOLE_NONE=y

AxelLin, 2024-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1353-gb3f7e2c8a4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build should work with CONFIG_ESP_CONSOLE_NONE=y.

### What is the actual behavior?

/home/axel/esp/esp-idf/components/vfs/vfs_console.c: In function 'esp_vfs_console_register':
/home/axel/esp/esp-idf/components/vfs/vfs_console.c:210:35: error: 'primary_path' undeclared (first use in this function); did you mean 'primary_vfs'?
  210 |     err = esp_vfs_register_common(primary_path, strlen(primary_path), primary_vfs, NULL, &primary_vfs_index);
      |                                   ^~~~~~~~~~~~
      |                                   primary_vfs
/home/axel/esp/esp-idf/components/vfs/vfs_console.c:210:35: note: each undeclared identifier is reported only once for each function it appears in
[514/1450] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/deprecated/rmt_legacy.c.obj
ninja: build stopped: subcommand failed.

### Steps to reproduce.

Build for ESP32C6 with CONFIG_ESP_CONSOLE_NONE=y.

AxelLin, 2024-01-16:
BTW, I also got below build warnings when CONFIG_ESP_CONSOLE_NONE=y

[429/938] Building C object esp-idf/console/CMakeFiles/__idf_console.dir/esp_console_repl.c.obj
/home/axel/esp/esp-idf/components/console/esp_console_repl.c:476:13: warning: 'esp_console_repl_task' defined but not used [-Wunused-function]
  476 | static void esp_console_repl_task(void *args)
      |             ^~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/console/esp_console_repl.c:372:18: warning: 'esp_console_common_init' defined but not used [-Wunused-function]
  372 | static esp_err_t esp_console_common_init(size_t max_cmdline_length, esp_console_repl_com_t *repl_com)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/console/esp_console_repl.c:351:18: warning: 'esp_console_setup_history' defined but not used [-Wunused-function]
  351 | static esp_err_t esp_console_setup_history(const char *history_path, uint32_t max_history_len, esp_console_repl_com_t *repl_com)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/console/esp_console_repl.c:326:18: warning: 'esp_console_setup_prompt' defined but not used [-Wunused-function]
  326 | static esp_err_t esp_console_setup_prompt(const char *prompt, esp_console_repl_com_t *repl_com)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~

Dazza0, 2024-01-16:
@AxelLin Thanks for reporting this issue, will take a look.

AxelLin, 2024-02-13:
@Dazza0 
I thought a compile fix should be applied ASAP.
Do you have target timeline to fix this?

AxelLin, 2024-02-27:
@Dazza0 @igrr 
Is it really difficult to fix this compile error?
I don't expect such long time for a compile fix.

closed 2024-02-29 by espressif-bot