espressif/esp-idf #12985: AT模式加主从一体模式互相通信问题

monkey-qiu, 2024-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我这边使用AT模式和主从一体模式通信的时候遇到两个问题，一个是主从一体的主机连接其他从机的时候，设置mtu需要将近1分钟，二个就是AT模式的从机和主从一体的主机互相发送数据时，AT模式的从机一直提示AT指令ERROR，只能接收主机发送过来的数据

zhp0406, 2024-01-16:
@monkey-qiu 
可以给出完整的 log 吗？

zhp0406, 2024-01-16:
@monkey-qiu 
你这里所说的 mtu 是指的 BLE 设置 mtu 吗？

zhp0406, 2024-01-16:
@monkey-qiu 
“二个就是AT模式的从机和主从一体的主机互相发送数据时，AT模式的从机一直提示AT指令ERROR，只能接收主机发送过来的数据”

可以给出更加具体的描述吗？（建议不同的问题分开报，这样子方便找到负责具体模块的人）

monkey-qiu, 2024-01-16:
> @monkey-qiu “二个就是AT模式的从机和主从一体的主机互相发送数据时，AT模式的从机一直提示AT指令ERROR，只能接收主机发送过来的数据”
> 
> 可以给出更加具体的描述吗？（建议不同的问题分开报，这样子方便找到负责具体模块的人）

就是AT模式的从机支持多连接么，我现在的使用需求是，从机需要被多个主机连接，现在测试的时候，有一个主机连接上来，然后比较高频的给AT从机发数据，但是AT从机回复数据的时候发送AT+BLEGATTSNTFY指令一直报ERROR，或者没响应

monkey-qiu, 2024-01-16:
> @monkey-qiu 你这里所说的 mtu 是指的 BLE 设置 mtu 吗？
主从一体的主机连接别的从机时，参数设置调用那个接口，例如通信带宽、通信频率等

jamesxwx, 2024-01-16:
@monkey-qiu  AT  支持 3个连接，可以参考 AT  下面的示例文档
另外，AT  的问题可以到 esp-at 仓库下面提 issue 
https://github.com/espressif/esp-at/issues

monkey-qiu, 2024-01-17:
> @monkey-qiu 你这里所说的 mtu 是指的 BLE 设置 mtu 吗？

你好，你这边能给我一个demo么，就是主从一体模式的主机和从机都支持3个连接