espressif/esp-idf #12986: ESP32-S3 SDMMC example fails, sdmmc_init_ocr: send_op_cond (1) returned 0x107

SneezingCactus, 2024-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1353-gb3f7e2c8a4

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

The SD card should've been initialized and mounted in SDMMC mode normally, as it did in SPI mode.

### What is the actual behavior?

The SD card appears to give responses to some commands during initialization, as the logs notify that the card present is an SDHC/SDXC card. However, initialization later fails with a 0x107, suggesting that the SD card did not respond to a command (or several).

### Steps to reproduce.

1. Hook up the SD card to the board. GPIO choice shouldn't matter given it's an ESP32-S3, but these are the GPIOs I used during testing:

<table>
<tr>
 <td>SD card
 <td>GPIO
<tr>
 <td>CLK
 <td>14
<tr>
 <td>CMD
 <td>13
<tr>
 <td>D0
 <td>12
<tr>
 <td>D1
 <td>11
<tr>
 <td>D2
 <td>10
<tr>
 <td>D3
 <td>9
</table>

[... 140 lines cut ...]
D (762) sdmmc_periph: slot=1 host_div=10 card_div=20 freq=400kHz (max 400kHz)
D (762) sdmmc_periph: slot=1 width=1
D (772) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (792) sdmmc_sd: SDHC/SDXC card
D (792) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (792) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x107; not IO card
D (1792) sdmmc_common: send_op_cond timeout, trying MMC
D (1792) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (1792) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (1792) sdmmc_req: process_command_response: error 0x107  (status=00000100)
D (1802) sdmmc_cmd: sdmmc_send_cmd_send_op_cond: sdmmc_send_app_cmd err=0x107
E (1812) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
D (1822) sdmmc_init: sdmmc_card_init: sdmmc_init_ocr returned 0x107
E (1822) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
E (1832) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
I (1842) main_task: Returned from app_main()
```

### More Information.

* I have tried both 1-bit and 4-bit SDMMC mode, both of which do not work. SPI mode works just fine.
* I have pull-up resistors in every line. Removing the pull-ups doesn't make a difference in the results. Note that I'm using 4.7k resistors instead of 10k as I currently don't have enough 10k resistors for all of the lines.
* I have tried two different SD cards, both of which generate the same results. One is 2GB, the other one is 8GB, and they're both formatted as FAT32.
* I have tried using different GPIOs.
* I have tried changing the max_freq_khz to SDMMC_FREQ_PROBING, SDMMC_FREQ_DEFAULT and SDMMC_FREQ_HIGHSPEED.

SneezingCactus, 2024-01-20:
Seems like the real issue all this time was the power supply. I hooked up an external 3.3V power supply to the SD card and it started working perfectly. 4-bit won't work (for some reason, sdmmc_init_ocr still timeouts if DAT1 and DAT2 aren't grounded, even when trying to initialize the card as 4-bit), but 1-bit is more than enough for my project.

I will be closing the issue now.

closed 2024-01-20 by SneezingCactus

pacucha42, 2024-01-31:
Nice to hear you have solved the issue. Thanks for the update