espressif/esp-idf #12987: Wifi Aware / NAN support on C3/S3+?

telamon, 2024-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello!
I recently discovered the Wifi Aware specifications and was very happy to see that the ESP32 had support for it.
Is the NAN-feature for the current C3/S3 series planned to be added or dropped?
| Test                        | Note                                                           |
| ---                         | ---                                                            |
| ESP32 <-> ESP32             | Works ⭐                                                       |
| ESP32 <-> OpenNAN/Wireshark | Reports invalid beacon interval of 200, should be 100? |
| ESP32 c3                    | Fails to compile                                               |
| ESP32 s3                    | Fails to compile                                               |

Xiehanxin, 2024-01-17:
hi @telamon , the NaN is supported on ESP32 and ESP32-S2, and we don't have plan to support it on other chips.

telamon, 2024-01-17:
@Xiehanxin thank you for the reply, as far as i am aware, WIFI NAN is the only protocol that allows for direct anonymous and promiscuous connections between devices. (No pairing)

Are there any plans to discontinue production of  ESP32 or ESP32-S2 chips?

nachiketkukade, 2024-01-22:
> ESP32 <-> OpenNAN/Wireshark | Reports invalid beacon interval of 200, should be 100?

@telamon , yes, this should be 100, we'll fix this.

> Are there any plans to discontinue production of ESP32 or ESP32-S2 chips?

We have no such plans yet, latest IDF releases continue to support these chips.

telamon, 2024-01-25:
@nachiketkukade while exploring the API's i also discovered that you cannot subscribe and publish to the same service. It throws "Service is already in use" error - I don't think that is limitation in the NAN protocol, is it an bug?

Second error which is a bit more of a hindrance, is that if you have initiated an datapath with a peer, then disconnect and deinit the datapath; then all subsequent NDP requests with the same peer fails.
It seems to me that datapaths only work once and then require a hard reset of the device.
I might be wrong, please confirm?

AxelLin, 2024-01-27:
> hi @telamon , the NaN is supported on ESP32 and ESP32-S2, and we don't have plan to support it on other chips.

@Xiehanxin @nachiketkukade 
Is this hardware limitation on S3/C3/C6? Or for other reasons the newer SoC cannot support NaN?

nachiketkukade, 2024-01-30:
@telamon ,

>  i also discovered that you cannot subscribe and publish to the same service. It throws "Service is already in use" error - I don't think that is limitation in the NAN protocol, is it a bug?

The NAN specs do indeed define Service Name as "A string uniquely identifying the service. The designer of the service selects the name and ensures that it is unique for the service." [From section 1.3.3].
Also I'm not able to understand  this scenario from a use-case point of view, a device subscribing to(meaning seeking) a particular service also publishing(meaning providing) the same service doesn't make much sense. Can you elaborate your use-case a bit more?

> Second error which is a bit more of a hindrance, is that if you have initiated an datapath with a peer, then disconnect and deinit the datapath; then all subsequent NDP requests with the same peer fails.

This seems like an older issue that is fixed already. Can you please share the IDF version you're using? A set of console logs from both sides will also be helpful.

nachiketkukade, 2024-01-30:
@AxelLin , 

> Is this hardware limitation on S3/C3/C6? Or for other reasons the newer SoC cannot support NaN?

S3/C3 have many memory optimisations. Due to these, supporting NAN would have greater impact on memory which would affect other features. This is why NAN support was dropped. We're still making efforts to support it on C6, we will take the final call after evaluating its full impact.

telamon, 2024-03-28:
@nachiketkukade Apologies for late response.

> Also I'm not able to understand this scenario from a use-case point of view, a device subscribing to(meaning seeking) a particular service also publishing(meaning providing) the same service doesn't make much sense. Can you elaborate your use-case a bit more?

My use-case is a relay based swarm-net, all peers publish and subscribe a globally "known" state-reconciliation service:
1. Discover a device within signal range.
2. Establish link.
3. Exchange State.
4. Disconnect.

That is my reason to consider NAN bidirectional Pub/Sub for discovery and DataPaths communication.

AxelLin, 2024-05-19:
> > ESP32 <-> OpenNAN/Wireshark | Reports invalid beacon interval of 200, should be 100?
> 
> @telamon , yes, this should be 100, we'll fix this.

@nachiketkukade Is this fixed?

nachiketkukade, 2024-05-20:
@AxelLin , Yes this is fixed in master commit 029748bfadd3bfa9f84c99f9cfae297df9e94b64

nachiketkukade, 2024-05-20:
Closing since original questions/issues are addressed

closed 2024-05-20 by nachiketkukade