espressif/esp-idf #12988: Expose stop bit in modbus communication config

mattiaVDM, 2024-01-16:
### Is your feature request related to a problem?

I have a specification in my project which is asking to change stop bit configuration of ModBus slave stack at runtime. 
I see no option to do so in mb_communication_info_t struct.
```
/**
     * @brief Device communication structure to setup Modbus controller
     */
    typedef union
    {
        // Serial communication structure
        struct
        {
            mb_mode_type_t mode;  /*!< Modbus communication mode */
            uint8_t slave_addr;   /*!< Modbus slave address field (dummy for master) */
            uart_port_t port;     /*!< Modbus communication port (UART) number */
            uint32_t baudrate;    /*!< Modbus baudrate */
            uart_parity_t parity; /*!< Modbus UART parity settings */
            uint16_t dummy_port;  /*!< Dummy field, unused */
        };
        // TCP/UDP communication structure
        struct
        {
            mb_mode_type_t ip_mode;          /*!< Modbus communication mode */
            uint16_t ip_port;                /*!< Modbus port */
            mb_tcp_addr_type_t ip_addr_type; /*!< Modbus address type */
            void *ip_addr;                   /*!< Modbus address table for connection */
            void *ip_netif_ptr;              /*!< Modbus network interface */
        };
    } mb_communication_info_t;
```
   

### Describe the solution you'd like.

Expose the option to set stop bits when configuring the modbus stack.

### Describe alternatives you've considered.

I tried to read the source code but could not figure out how to set it by myself after slave/master initialization. Is there a workaround to do it?

atanisoft, 2024-01-16:
Looks to be a duplicate of https://github.com/espressif/esp-idf/issues/8003 which has a solution for changing the stop bits

alisitsyn, 2024-01-17:
Here is additional duplicate issue against esp-modbus:
https://github.com/espressif/esp-modbus/issues/25

See also the options in [v2.0.0_beta release](https://github.com/espressif/esp-modbus/discussions/45): 
[serial options example](https://github.com/espressif/esp-modbus/blob/release/v2.0/examples/serial/mb_serial_master/main/serial_master.c#L274)

closed 2024-05-24 by alisitsyn