espressif/esp-idf #12989: Add Example on BLE client + Wifi coex (IDFGH-11907)

gavalierm, 2024-01-16:
### Is your feature request related to a problem?

I tried to find sutable example, but there is non.

### Describe the solution you'd like.

I will like to see working solution how BLE Client can scan/connect/exchange passcode and coexist with WIFI.

On ESP32 C3 S3 boards.

### Describe alternatives you've considered.

My second choice will be Arduino wrapper, but i hope thats can be done.

### Additional context.

My main goal is interconnect BLE Device with Wifi Device.

In scenario BLE device - BT - ESP logic - Wifi - Wifi AP - Wifi Clients

gavalierm, 2024-01-19:
Hi @esp-lrh 

Thank you so much for your answer. 

Your response make sense. 

My scenario is 

Scan/Connecting/Connected: BLE and WIFI not at the same time - First wifi, after OK, MQTT after OK, BLE.

This should work.

My main question is about if can WIFI and BLE coexists on receiving and sending data (sorry for not clear question at very first post)

The Goal is write BLE to WIFI bridge (just forwarding Buffer data nothing fancy).

I look at almost all examples, but i can not found the right for me.

The example in ` examples/bluetooth/nimble/bleprph_wifi_coex` is PRPH, but PHRP do not do a scan and do not connect to the Advertised devices. So i think i need CENT example - but i can find any in my scenario.

(If i understand right that CENT is "client" who connect to the advertised device, and adverthise device show passcode on the sceen).

I do this once in past in arduino, but this time i switch to pure IDF only (and i thinks is better approach) but i struggle with basics (like it is CENT what i need?)

Thank you again for your response.

esp-lrh, 2024-01-19:
> Hi @esp-lrh
> 
> Thank you so much for your answer.
> 
> Your response make sense.
> 
> My scenario is
> 
> Scan/Connecting/Connected: BLE and WIFI not at the same time - First wifi, after OK, MQTT after OK, BLE.
> 
> This should work.
> 
> My main question is about if can WIFI and BLE coexists on receiving and sending data (sorry for not clear question at very first post)
> 
> The Goal is write BLE to WIFI bridge (just forwarding Buffer data nothing fancy).
> 
> I look at almost all examples, but i can not found the right for me.
> 
> The example in ` examples/bluetooth/nimble/bleprph_wifi_coex` is PRPH, but PHRP do not do a scan and do not connect to the Advertised devices. So i think i need CENT example - but i can find any in my scenario.
> 
> (If i understand right that CENT is "client" who connect to the advertised device, and adverthise device show passcode on the screen and client have to type and send the passcode to the adv device).
> 
> I do this once in past in arduino, but this time i switch to pure IDF only (and i thinks is better approach) but i struggle with basics (like it is CENT what i need?)
> 
> Thank you again for your response.

`CENT is "client" who connect to the advertised device` you can think so. 

From Bluetooth Core Specification:
```
When entered from the Initiating state, the Connection state shall be in the Central Role.
When entered from the Advertising state, the Connection state shall be in the Peripheral Role.

The Link Layer in the Central Role will communicate with a device in the Peripheral Role and defines the timings of transmissions. 
```

RF time-division multiplexing between BLE and WIFI is implemented in lower level, has almost no impact on the upper-layer APIs. All you need to do is simply receive data from BLE and call the WIFI APIs to send the data, and it will work as if you are receiving data from BLE and sending data using WIFI at the same time.

There is a BLE Central example in `examples/bluetooth/nimble/blecent`, you can try to merge it's code with example `bleprph_wifi_coex' (remember to remove the origin BLE Peripheral code).

gavalierm, 2024-01-19:
I wil try.

In fact this is exactly what is this issue about. To provide "merged" code for CENT+COEX scenario :)

I hope that i will have success.

If i will have more question i will ask you.

Thank you again.

gavalierm, 2024-01-19:
I have to say that example `examples/bluetooth/nimble/blecent/tutorial/blecent_walkthrough.md ` have many thing not cleared. I think many functions mentioned in this tutorial is not part of main.c and i was strugling what is about then.

I missing the part when CENT exchange the passcode with PRPH. So i have no clue how to setup and handle Security events.

The `esp_central.h` is dependency, but is not mentioned in walkthrough.  And is not defined if i need to use "examples" path to include (via idf_component.yml which i did) or is better approach to handle this.

Can you help me with those issues? Thank you.

gavalierm, 2024-01-19:
I m reallz strugling. I cannot move forward, because of PASSKEY.

The default CENT example to not count with this scenario... so i m done...

If you will have some time can you help?

Thank you.

gavalierm, 2024-01-19:
I can not pass this:

```
    /* Read the supported-new-alert-category characteristic. */
    chr = peer_chr_find_uuid(peer, BLE_UUID16_DECLARE(ServiceUUID), CameraService);
    if (chr == NULL) {
        MODLOG_DFLT(ERROR, "Error: Peer doesn't support CameraManufacturerUUID and CameraModelUUID\n");
        goto err;
    }
```

According to the protocol of Camera to i want connect

```
Blackmagic Bluetooth Camera Control
Blackmagic cameras with Bluetooth LE implement a variety of features and commands that
allow users to control their cameras wirelessly. Developers have full access to these features for
their custom applications.
The following services and characteristics describe the full range of communication options that
are available to the developer.
Service: Device Information Service
UUID: 180A
Characteristics
Camera Manufacturer
UUID: 2A29
Read the name of the manufacturer (always “Blackmagic Design”).
Camera Model
UUID: 2A24
Read the name of the camera model (eg. “URSA Mini Pro”).
Service: Blackmagic Camera Service
UUID: 291D567A-6D75-11E6-8B77-86F30CA893D3
Characteristics
Outgoing Camera Control (encrypted)
UUID: 5DD3465F-1AEE-4299-8493-D2ECA2F8E1BB
Send Camera Control messages.
These messages are identical to those described in the Blackmagic SDI Camera Control
Protocol section above. Please read that section for a list of supported messages and required
formatting information.
For an example of how packets are structured, please see the ‘example protocol packets’
section in this document.
Incoming Camera Control (encrypted)
UUID: B864E140-76A0-416A-BF30-5876504537D9
Request notifications for this characteristic to receive Camera Control messages from
the camera.
These messages are identical to those described in the Blackmagic SDI Camera Control
Protocol section above. Please read that section for a list of supported messages and required
formatting information.
```

I tried think i tried everythink but i m missing something.

gavalierm, 2024-01-20:
Ok i make progress.

The whole issue was with configuration of peer. Needs to be: `ble_hs_cfg.sm_io_cap = BLE_HS_IO_KEYBOARD_ONLY;` 

But this is what examples is good for :) There is no info about SM_IO_CAP.

I found this in `esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h`

gavalierm, 2024-01-20:
@esp-lrh 

OK i just give up. I m done.

With no solid example i m only guessing, flashing, guessing, flashing, almost for days now. With no progress.

Please add an example which will contains:
1. setup CENT device for accepting passkey (i think i have this done correctly)
2. scan a filter device with target service UUID (0x1800 in my case, i have this done)
3. establish connection and initiate secure connection (i have this done, kind of)
4. notify that PHPR needs to send passkey (on_passkey event, i have this done)
5. receive passkey from PHPR (!!! i was searching for this almost 5 hours!!!) i found ble_sm_inject_io works but if this is correct? I have no clue
6. create pair and bonded device and store (!!! my PRPH device still force me to enter password, again and again...)
7. subscribe to the custom characteristics (128) and starts receiving data (on event) NOT DONE

I do not know if my subscription is working... i have no clue if i do not receive data because wrong encryption, wrong subscriptions, wrong

I tried establish connection with PRPH device from my phone and is done with remembering the connection, never ask me again for passkey, and reads data like expected.

Please add example to the Examples.

Thank you.

esp-lrh, 2024-01-22:
Hi, @gavalierm 
Here are some explanations for your issues:
1. `CONFIG_EXAMPLE_EXTENDED_ADV`
This is a feature introduced in BLE 5.0, you can just ignore it

2. PASSKEY and `SM_IO_CAP` 
`SM_IO_CAP`  is a properties defined by the Bluetooth protocol stack and it will affect the pairing process.
see https://www.bluetooth.com/blog/bluetooth-pairing-part-1-pairing-feature-exchange/
The default value in `blecent` example is `BLE_HS_IO_NO_INPUT_OUTPUT`, which means the device has no ability to input or output anything, the pairing and encryption process will just work.
And that is the reason why you  'missing the part when CENT exchange the passcode with PRPH. So i have no clue how to setup and handle Security events'.
If you still want to entry a passkey when paring, you should change `SM_IO_CAP` and  listen to GAP event `BLE_GAP_EVENT_PASSKEY_ACTION` then respond appropriately.
example usage see `examples/bluetooth/nimble/bleprph/main/main.c`
 
3.  Add example to the `examples`
Maybe not now, we don't have much time to add example of such a specific scenario.

I think you could simply set `SM_IO_CAP` as `BLE_HS_IO_NO_INPUT_OUTPUT`, it will work and link key will be automatically save in device. Please try again.

gavalierm, 2024-01-22:
Hi @esp-lrh 

Thank you for your time.

Yes i figure out the the `BLE_GAP_EVENT_PASSKEY_ACTION` and `SM_IO_CAP`.

But:

the SM have to be enabled in menuconfig, store persist in NVS have to be enabled in menuconfig. I figure out those things too.

The connection have to be encrypted. This is my code from Arduino which works, but i not figure out how this i can setup in esp idf nimble.

```
  // Enable encryption
  _pBLEDevice.setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);  // i do not know how
  BLESecurity *pSecurity = new BLESecurity(); // not relevant
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND); // i do not know how
  pSecurity->setCapability(ESP_IO_CAP_IN); // done - ble_hs_cfg.sm_io_cap = BLE_HS_IO_KEYBOARD_ONLY;
  pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);   // i do not know how
```
Because connection have to be encrpzted i do not know that i do not receive the nofitifaction because wrong encryption or why.

and missing something like this `_incomingCameraControl->registerForNotify(_receiveNotifyCallback, false);`

I found in example the "send 2 bytes to seubscribe" so i try to write 2 bytes 0x01 and 0x00 to the CHR but i got Err code 6...

esp-lrh, 2024-01-22:
Hi,
The example `blecent` listen to `BLE_GAP_EVENT_CONNECT` event and call `ble_gap_security_initiate`, after that, the encryption of the connection should be enabled.

To register a service notification, you can use `esp_ble_gattc_register_for_notify`, see 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/bluetooth/esp_gattc.html

esp-lrh, 2024-01-22:
For nimble host, you can refer to the function `blecent_read_write_subscribe` in `blecent`.

Actually, the question now is how to use nimble host, maybe you can read nimble official document first.
https://mynewt.apache.org/latest/network/ble_hs/ble_hs.html

gavalierm, 2024-01-22:
I have `blecent_read_write_subscribe` tried a thousend times :)

It is possible that i need differend write data? 

My CHR is write only or notify only, so writing to them cause ERR 6.

I have bluetooth explorer app on my computer and i can see all CHR and roles of CHR. This explorer works as expected, he can read what is readable, he receive notifications from notif only CHR and he can write to the writable CHR.

I m missing something and i do not know what it is.

in fact i think i m connected and secured, because i can receive RX events. **But just from one single CHR**. I have to receive 3 CHR at the same time.

When i tried send same "subscribe" data to the others events with "no_rsp_write" i do not receive any RX events at all.

Like i said i have old Arduino code and after "Register notify" code i start imediatelly receivin bunch of data ... 

So something not working properly with my nimble code.

gavalierm, 2024-01-22:
@esp-lrh  can you tell me please what is error code 6 when i m trying to write_flat ?

esp-lrh, 2024-01-22:
`ble_gattc_write_flat` return error code 6 (BLE_HS_ENOMEM)? 
It means no memory to store and send the data, can you provide your code for me to check?

gavalierm, 2024-01-22:
I just try it again (not in repo now)

in function blecent_subscribe_to instead of `ble_gattc_write_flat`  instead of `ble_gattc_write_no_rsp_flat` add callback, and NULL and i m geting

`W (13414) ESP: Service ERR 6`

```
        const uint8_t notifyOn[] = {1, 0};
        /*** Write 0x00 and 0x01 (The subscription code) to the CCCD ***/

        rc = ble_gattc_write_flat(peer->conn_handle, dsc->dsc.handle, &notifyOn, sizeof(notifyOn), after_write, NULL); //ble_gattc_write_no_rsp_flat
        if (rc != 0) {
            ESP_LOGW(TAG,"Service ERR %d\n",rc);
            return rc;
        }
```

gavalierm, 2024-01-22:
I want to go step by step.

First i need to send "my name" to the camera.

This CHR is not encrypted at all. So need to work.

This is what i use to to that. (i like i said i know that this is possible, because on Arduino is done by chr->writeValue)

Camera just show up the name of connected device in the menu. Now is ADDR only. To this step is first step to get done.

   ```
 chr = peer_chr_find_uuid(peer, CameraService_UUID, DeviceName_UUID);
    if (chr == NULL) {
        MODLOG_DFLT(ERROR, "Error: Peer doesn't support DeviceName_UUID\n");
        goto err;
    }else{
        MODLOG_DFLT(WARN, "OK: Peer HAVE DeviceName_UUID\n");
        const char *string_to_write = ble_svc_gap_device_name();
        ESP_LOGW(TAG, "Device name %s",string_to_write);
        uint16_t data_len = strlen(string_to_write);
        rc = ble_gattc_write_no_rsp_flat(peer->conn_handle, chr->chr.val_handle, &string_to_write, sizeof(string_to_write));
        if (rc != 0) {
            MODLOG_DFLT(ERROR, "Error: Failed to WRITE characteristic; DeviceName_UUID");
            goto err;
        }
    }
```

gavalierm, 2024-01-22:
OH YEAH, after 5 days of trying. I was able to set the name.

The problem was the "reading" part before "write".

So question is: When i perform "read" operation - this somhow block the other steps?

This is ma read part (just read the data and log)
```

    chr = peer_chr_find_uuid(peer, BLE_UUID16_DECLARE(DeviceInformation_UUID), BLE_UUID16_DECLARE(CameraModel_UUID));
    if (chr == NULL) {
        MODLOG_DFLT(ERROR, "Error: Peer doesn't support CameraModel_UUID\n");
        goto err;
    }else{
        MODLOG_DFLT(WARN, "OK: Peer HAVE CameraModel_UUID\n");
        rc = ble_gattc_read(peer->conn_handle, chr->chr.val_handle, blecent_read_log, NULL);
        if (rc != 0) {
            MODLOG_DFLT(ERROR, "Error: Failed to read characteristic; CameraManufacturer_UUID");
            goto err;
        }
    }
```

gavalierm, 2024-01-23:
This part failed always with error code 6

```
    chr = peer_chr_find_uuid(peer, CameraService_UUID, OutgoingCameraControl_UUID);
    if (chr == NULL) {
        MODLOG_DFLT(ERROR, "Error: Peer doesn't support OutgoingCameraControl_UUID\n");
        goto err;
    }else{
        MODLOG_DFLT(WARN, "OK: Peer HAVE OutgoingCameraControl_UUID\n");
        int8_t value = {0xFF,0x05,0x00,0x00,0x01,0x0A,0x01,0x00,0x02,0x00,0x00,0x00};
        rc = ble_gattc_write_flat(peer->conn_handle, chr->chr.val_handle, &value, sizeof(value), NULL, NULL);
        if (rc != 0) {
            MODLOG_DFLT(ERROR, "Error: Failed to WRITE characteristic; OutgoingCameraControl_UUID rc=%d", rc);
            goto err;
        }
    }
```

esp-lrh, 2024-01-23:
Hi, @gavalierm 
Pairing and encryption do not always require user to input PASSKEY, just set `sm_io_cap` to `BLE_HS_IO_NO_INPUT_OUTPUT` first.

In addition, I recommend to disable other functions, such as wifi and MQTT, when debugging Bluetooth.

I made some small changes and run your code, it can connect to example `bleprph` and enable encryption, subscribe notification successfully.

```
W (5102) NimBLE: START: blecent_subscribe_to

I (5102) NimBLE: INFO: Service HAVE DESCRIPTOR

  conn_handle=0
  conn_handle=20

I (5112) NimBLE: GATT procedure initiated: write no rsp; 
I (5122) NimBLE: att_handle=20 len=2

```
Maybe you can connect to a `bleprph`  and debug first.

esp-lrh, 2024-01-23:
> code 6

code 6 means no memory, it is strange, disable wifi and try again ?
error code definition see `components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h`

gavalierm, 2024-01-23:
Yes i can disable WIFI an MQTT because i already have PASSKEY exchanged.

But with BLE_HS_IO_NO_INPUT_OUTPUT the camera never fire the PASSKEY and i can not make the encrypted connection.

According to the protocol i think PASSKEY is essential and Camera do not work with BLE_HS_IO_NO_INPUT_OUTPUT.

I think i have Security done. I can WRITE to the camera command which do what i need to do. BUT JUST ONCE.

After i write command second command is returned as ERR=6. 

My goal is simple and i do not understand why is so hard to achieve this.

Just read data from MQTT (hex buffer) and PASS it to the CHR on BLE (same buffer, same data, same strucutre, do nothig just pass it).

And vice versa. 

READ data from Notification and just PASS it to the MQTT as buffer.

And i hoped that will be simple, with 2 callbacks on both sides for each direction and done. No storing not manipulating just pass the data and free the buffer.

Why is this so hard to do - i m so confused.

gavalierm, 2024-01-23:
> > code 6
> 
> code 6 means no memory, it is strange, disable wifi and try again ? error code definition see `components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h`

I disable WIFI and MQTT in main.c by commenting it out so WIFI MQTT will not start.

And still getting ERR 6 on 4th operation.

gavalierm, 2024-01-23:
I found this in Menuconfig and looks like the 3 is the number what i m looking for.

```
(3) Maximum number of concurrent connections
(3) Maximum number of bonds to save across reboots
(8) Maximum number of CCC descriptors to save across reboots
(4) Maximum number of connection oriented channels
```

esp-lrh, 2024-01-23:
Event if you do not input a PASSKEY, the connection **will also be encrypted** after `ble_gap_security_initiate`

if you got ERR 6 after 3 operation, it means the data of previous 3 operation had not been sent.

I think your connection is stuck encryption process so it can't send data.

When I run your code first time (`sm_io_cap=BLE_HS_IO_KEYBOARD_ONLY`), I got a encryption timeout, it means you do not response `BLE_GAP_EVENT_PASSKEY_ACTION` event appropriately.

Bluetooth is not a so simple protocol, Bluetooth Core Specification has more than 3000 pages and there are many other profile, **don't guess its behaviors**, please read document and search in internet. 

The method you response to `BLE_GAP_EVENT_PASSKEY_ACTION` event is not good, please refer to the example.

gavalierm, 2024-01-23:
`When I run your code first time (sm_io_cap=BLE_HS_IO_KEYBOARD_ONLY), I got a encryption timeout, it means you do not response BLE_GAP_EVENT_PASSKEY_ACTION event appropriately.`

Can you point me to the EXAMPLE when is BLE_GAP_EVENT_PASSKEY_ACTION on BLECENT done?

Me esp is in CENT mode and the PHPR send the BLE_GAP_EVENT_PASSKEY_ACTION event, i need the catch this event and SEND BACK THE PASSKEY.

This is how i send back the PASSKEY if it is wrong approach i will be really glad to see the right one, can you point me to the exmaple when CENT device response to the PHPR device with the passkey?

```
void sendBLE_Passkey(uint32_t passkey){
            // Prepare the structure with the passkey
        struct ble_sm_io io;
        io.action = BLE_SM_IOACT_INPUT;
        io.passkey = passkey;

        ble_sm_inject_io(conn_handle, &io);
    //esp_ble_passkey_reply(mqtt_client->ble_security.ble_req.bd_addr, true, passkey);
}
```

`if you got ERR 6 after 3 operation, it means the data of previous 3 operation had not been sent.`

Thats not true, because PHPR device CHANGE THE VALUE. So my data was send and PHPR device (in my case Camera) receive the data and change the value (change settings to what i want). So still this confusing me fully. 

Camera receive the data, do her job BUT data "was not send?" this make no sense.

`Bluetooth is not a so simple protocol, Bluetooth Core Specification has more than 3000 pages and there are many other profile, don't guess its behaviors, please read document and search in internet.
`

Yes i learn this hardway... but still there is a companioes like ESP IDF or NIMBLE who create the Libs to use it without any "deep" knowledge i think. Just "read" and "write" scenario can not by so hard that i need to learn 3000 pages.... But i got your point.

gavalierm, 2024-01-23:
Ok I make a HUGE progress just with simple BYTE :)

I explore whole Arduino library. So much inspiration here. I found that "Subscribe" message can be:
- 0x01 0x00 for Notification (without ACK)
- 0x02 0x00 for Notification (with ACK) a.k.a Indication

Because Camera PROTOCOL do not say anything about 01/02 i tried the **02 and it works**! So i update the registry process like Arudino lib have - just add flag for 01/02 scenario.

So i m really close to the finish now.

Just resolve the problem with most 3 write operations (icluding the subscibe writes...)

@esp-lrh if you can help mi figure out this problem i will be done. And i can create some Example for my scenario.

gavalierm, 2024-01-23:
So i update my git repo

I add the `blecent_write_to_outgoing` function to test how many writes i can perform..

Just 3 including the registrations writes.

I think there is a problem with 
- my ESP config
- Callbacks
- om_buf usage (have to be freed or something?)

I realize that the writes without ACK (no_rsp_write) not working at all. So i think the writes get respone with but my callback (NULL) dont do anything and this can be problem?

It is essential to do something after write operation?

Because my CHRs is write only, when i perform READ on this the Camera reject the connection (it looks like that)

esp-lrh, 2024-01-24:
I say not good because your method response to `BLE_GAP_EVENT_PASSKEY_ACTION` only work in specific scenarios
The type of `io.action` is not up to you, you must keep it as `pkey.action = event->passkey.params.action`, perform the `action` pass from the event. Reference example I told you before, `blephrp`.
[image]

Anyway, no matter what io capability you set, just make sure you got the `BLE_GAP_EVENT_ENC_CHANGE` event and the event status code is 0.

If you still got error code 6, try add `CONFIG_BT_NIMBLE_GATT_MAX_PROCS=20` to your `sdkconfig.defaults` and rebuild, or edit this option in `menuconfig`, the default value is 4.

gavalierm, 2024-01-24:
Who is initiator and who is responder?

Because i think in my case the initiator is "CENT" device (my code), because i "Initiate security".

My PHPR camera is "Display only" for sure. Because this is written in protocol (camera display the code) there is no other option - on the page what you send is not option for "Display only" for responder so i think is "Display/Keyoard" so in this fact my device can be "display only" and Camera should show up the keyboard to type the pass. But this never happed, camera respond to this type of request with error. So i think the developers of camera jus implement the part "Display Only" and all others (Keyboard) just responde with error.

So my CENT device (initiator) have to be some sord of "passkey INPUT" becase camera is DISPLAY ONLY (or respond only for display type of inititation).

So my cant HAVE to be set to "KeyboardOnly" because whe i set it to "DisplayKeyboard" nothing happned... camera do not show the keyboard and i do not generate the pass ...

Like you said - I got ENC response 0 so all security is not the case for me - its working.

I will try to solve the operations...

Thank you for the tip `CONFIG_BT_NIMBLE_GATT_MAX_PROCS` i will try.

gavalierm, 2024-01-24:
@esp-lrh 

`CONFIG_BT_NIMBLE_GATT_MAX_PROCS` works!!! Thank you for pointing me.

So i need to try now send the command 100 times? because (logically) i need to write to the camera many times for long long time.

gavalierm, 2024-01-24:
And one more thing (and i hope it will be the last one)

My data is truncated to the 2 block of max 12 bytes you can see the log, that i have one log of 12 bytes and 1 log of remaining 4 bytes... 

Can you please ponit me to the code example where i can combine this data before sending int to MQTT? Thank you.

```
I (23067) DATA: ff 10 00 00 09 07 02 02 c7 02 12 03 00 00 01 00 
I (23068) DATA: 10 27 00 00 
I (23167) DATA: ff 05 00 00 01 08 01 02 03 
```

rahult-github, 2024-01-24:
@gavalierm  , 

Please check once in your application if the word "DATA" is printed as part of it.  There is no such print with word "DATA" in stack / example. 

As for the sequence, i think it may be due to print_mbuf(attr->om) call done from some place in the application. 

The data is actually in sequence only, but in chained format. So it gets printed in separate lines in a loop as the print introduces a new line after every chain iteration.  It is being sent as a whole only though.

gavalierm, 2024-01-24:
@rahult-github 

i use `ESP_LOG_BUFFER_HEX("DATA", data, data_len);`

in BLE_onReceive which is called every time i got RX

```
    case BLE_GAP_EVENT_NOTIFY_RX:
        BLE_onReceive(event->notify_rx.om);
        return 0;
```

```
void BLE_onReceive(const struct os_mbuf *om){
    if(esp_mqtt_state != STATE_CONNECTED){
        return;
    }
    uint16_t data_len = OS_MBUF_PKTLEN(om);
    uint8_t *data = malloc(data_len);
    
    if (data != NULL) {
        os_mbuf_copydata(om, 0, data_len, data);
        if(data[4] == 9 && data[5] == 0){ // 4 means operation type timecode 5 means thick
            //ESP_LOGW(BLE_TAG,"DATA Timecode Thick");
            return;
        }
        ESP_LOG_BUFFER_HEX("DATA", data, data_len);
        // send data to mqtt using enqueue whis is async-like behavior
        // sending from camera to the operator do not have priorty
        esp_mqtt_client_enqueue(mqtt_client, MQTT_DOWNSTREAM_TOPIC, (const char *)data, sizeof(data), 0, 0, true);

        // Free the allocated buffer
        free(data);
    } else {
        ESP_LOGE(BLE_TAG, "Failed to allocate memory for data");
    }
}
```

So this means that RX have to be triggeret twice with 12 bytes and with 4 remaining bytes...

rahult-github, 2024-02-06:
Closing this issue . Please feel free to reopen in case of further queries.

closed 2024-02-06 by rahult-github