espressif/esp-idf #12990: Combine file_server and tusb_msc

diplfranzhoepfinger, 2024-01-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

nopnop2002, 2024-01-17:
As far as I know, tusb_msc cannot be used with SPIFFS.

Quoted from README of tusb_msc

Data is read/written from/to SPI Flash through wear-levelling APIs.

This indicates that FAT will be used.

nopnop2002, 2024-01-17:
file_serving example only supports SD cards or SPIFFS.

TinyUSB only supports FAT.
Data is read/written from/to SPI Flash through wear-levelling APIs.

diplfranzhoepfinger, 2024-01-17:
look: 

[https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/examples/storage/wear_levelling/main/wear_levelling_example_main.c#L68C1-L80C6](https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/examples/storage/wear_levelling/main/wear_levelling_example_main.c#L68C1-L80C6)

and: 

[https://github.com/diplfranzhoepfinger/file_serving/blob/7abcde520c2cabd8233685884aca0893aaf84cf6/file_serving/main/mount.c#L133C1-L142C25](https://github.com/diplfranzhoepfinger/file_serving/blob/7abcde520c2cabd8233685884aca0893aaf84cf6/file_serving/main/mount.c#L133C1-L142C25)

same code.

nopnop2002, 2024-01-17:
If you mount the USB memory with tinyusb_msc_storage_mount(), I don't think any other mounting is necessary.

diplfranzhoepfinger, 2024-01-17:
yes, good idea ...

i will check it.

i understood it is only Mounted if it is plugged into PC.

so this will then not work. but i will check it ... OK. Thanks.

diplfranzhoepfinger, 2024-01-22:
> If you mount the USB memory with tinyusb\_msc\_storage\_mount(), I don't think any other mounting is necessary.
> 
> There is no need to perform example\_mount\_storage().

yes, approved, and working. 

the only thing is now: it seems the USB takes storage exclusively, not shared by the Application.

diplfranzhoepfinger, 2024-01-22:
[https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc)

well described here: will this change in Future ?

diplfranzhoepfinger, 2025-02-04:
@igrr you mentioned another Operation Mode of USB ?

tore-espressif, 2026-10-19:
@diplfranzhoepfinger USB must have exclusive access to the underlaying storage media, because USB works directly on memory blocks, bypassing File System APIs. Either the storage is exposed over USB _or_ it is exposed to the application (eg. to the file server)

> you mentioned another Operation Mode of USB ?

There is Media Transfer Protocol (MTP) that solves just this problem. Using this class, the USB device exposes _files_. However, it is not very well adopted; many smartphones or old-style photo cameras use this protocol. It usually works out-of-the-box on Windows but you might need extra drivers for MacOS or Linux.

We introduced MTP support with esp-tinyusb 2.3.0 https://github.com/espressif/esp-usb/pull/531

tore-espressif, 2026-10-23:
We'll close this issue, feel free to reopen if there are more questions

closed 2026-10-23 by tore-espressif