espressif/esp-idf #12991: System crash when disable BLE

Patrick-Bai, 2024-01-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

 ESP32-S3-DevKitC-1U-N8

### Power Supply used.

USB

### What is the expected behavior?

Disable BLE.

### What is the actual behavior?

System crash when I disable it.

### Steps to reproduce.

deinit flow: Stop beacon -> deregister gatt server -> disable Radio -> close session

### Debug Logs.

```plain
[1970-01-01 00:01Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
 
Core  0 register dump:
PC      : 0x40056eed  PS      : 0x00060630  A0      : 0x820696b5  A1      : 0x3fcbfbc0
A2      : 0x3fc9fb5c  A3      : 0x00000007  A4      : 0x00000007  A5      : 0x3fcbfc18
A6      : 0x3fc9f994  A7      : 0x00000000  A8      : 0x00000007  A9      : 0x00000000
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00060020  A13     : 0x00060023
A14     : 0x40003a8c  A15     : 0x4000fd6c  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000007  LBEG    : 0x42069926  LEND    : 0x42069934  LCOUNT  : 0x00000002
 
 
Backtrace: 0x40056eea:0x3fcbfbc0 |<-CORRUPTED
 
 
 
 
ELF file SHA256: 48bfb911fa5cef96

==================== CURRENT THREAD STACK =====================
#0  0x40056eed in ?? ()
#1  0x42064dd1 in ble_gap_find_reattempt_conn_idx (conn=0x0) at components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:194
#2  ble_hs_hci_evt_disconn_complete (event_code=<optimized out>, len=4, data=0x3fcbbc9a) at components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:194
#3  ble_hs_hci_evt_disconn_complete (event_code=<optimized out>, data=0x3fcbbc9a, len=<optimized out>) at components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:174
#4  0x42065068 in ble_hs_hci_evt_process (ev=0x3fcbbc98) at components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:901
#5  0x42064714 in ble_hs_event_rx_hci_ev (ev=<optimized out>) at components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:520
#6  0x4206c5dd in ble_npl_event_run (ev=0x3fca0af8 <ble_hs_hci_os_event_buf+12>) at components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:125
#7  nimble_port_run () at components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:78
```

### More Information.

I trace the code in ble_hs_hci_evt.c. I found the problem is this code. The conn will be NULL but it also run `memcmp()`
``` 
static int
ble_gap_find_reattempt_conn_idx(const struct ble_hs_conn *conn)
{
    int i;

    for (i = 0; i < MYNEWT_VAL(BLE_MAX_CONNECTIONS); i++) {
        if (memcmp(&reattempt_conn[i].peer_addr, &conn->bhc_peer_addr, sizeof(ble_addr_t)) == 0) {
            return i;
        }
    }
    /* No matching entry found. Return invalid index */
    return MYNEWT_VAL(BLE_MAX_CONNECTIONS);
}

static int
ble_hs_hci_evt_disconn_complete(uint8_t event_code, const void *data,
                                unsigned int len)
{
    ....   
    ble_hs_lock();
    conn = ble_hs_conn_find(le16toh(ev->conn_handle));
    if (conn != NULL) {
        ble_hs_hci_add_avail_pkts(conn->bhc_outstanding_pkts);
    }
    ble_hs_unlock();

#if CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT
    if (ev->reason == BLE_ERR_CONN_ESTABLISHMENT) {
        int rc, i, idx;

        idx = ble_gap_find_reattempt_conn_idx(conn);
```

rahult-github, 2024-01-17:
Hi @Patrick-Bai , 

v4.4.3 is older version. Request to use the latest version where this issue has been fixed.

Patrick-Bai, 2024-01-18:
Hi @rahult-github:
Thanks for your quick reply. I add the patch from v4.4.5 and I think the problem is fixed. I will close the issue.

closed 2024-01-18 by Patrick-Bai