espressif/esp-idf #12994: ESP32-C2 can not start ap in idf 5.1.2

heculess, 2024-01-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-C2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP8684-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

I can find ssid with name espnow

### What is the actual behavior?

Everything is OK, but I can not find any ssid with name espnow

### Steps to reproduce.

1. In the Example Configuration menu:
Set the Wi-Fi configuration.
Set WiFi SSID to espnow
Set WiFi Password to XXXXXXXX
2. Build and Flash

### Debug Logs.

```plain
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x16c4
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x2980
entry 0x403acb70
[0;32mI (21) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader[0m
[0;32mI (21) boot: compile time Jan 17 2024 16:02:42[0m
[0;32mI (21) boot: chip revision: v1.0[0m
[0;32mI (24) boot.esp32c2: MMU Page Size  : 32K[0m
[0;32mI (29) boot.esp32c2: SPI Speed      : 60MHz[0m
[0;32mI (34) boot.esp32c2: SPI Mode       : DIO[0m
[0;32mI (39) boot.esp32c2: SPI Flash Size : 2MB[0m
[0;32mI (43) boot: Enabling RNG early entropy source...[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (75) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (82) boot: End of partition table[0m
[0;32mI (86) esp_image: segment 0: paddr=00010020 vaddr=3c078020 size=1dc70h (121968) map[0m
[0;32mI (122) esp_image: segment 1: paddr=0002dc98 vaddr=3fcacd80 size=02380h (  9088) load[0m
[0;32mI (125) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=71ca8h (466088) map[0m

[2024-01-17 16:35:48.297]
RX：[0;32mI (232) esp_image: segment 3: paddr=000a1cd0 vaddr=3fcaf100 size=000f0h (   240) load[0m
[0;32mI (232) esp_image: segment 4: paddr=000a1dc8 vaddr=40380000 size=0cd78h ( 52600) load[0m
[0;32mI (255) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (255) boot: Disabling RNG early entropy source...[0m
[0;32mI (267) cpu_start: Unicore app[0m
[0;32mI (267) cpu_start: Pro cpu up.[0m
[0;32mI (275) cpu_start: Pro cpu start user code[0m
[0;32mI (275) cpu_start: cpu freq: 120000000 Hz[0m
[0;32mI (276) cpu_start: Application information:[0m
[0;32mI (278) cpu_start: Project name:     wifi_softAP[0m
[0;32mI (284) cpu_start: App version:      1[0m
[0;32mI (288) cpu_start: Compile time:     Jan 17 2024 16:33:40[0m
[0;32mI (294) cpu_start: ELF file SHA256:  cedd9f2a912b118e...[0m
[0;32mI (300) cpu_start: ESP-IDF:          v5.1.2-dirty[0m
[0;32mI (306) cpu_start: Min chip rev:     v1.0[0m
[0;32mI (310) cpu_start: Max chip rev:     v1.99 [0m
[0;32mI (315) cpu_start: Chip rev:         v1.0[0m
[0;32mI (320) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (327) heap_init: At 3FCB2F90 len 00029BE0 (166 KiB): D/IRAM[0m
[0;32mI (333) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM[0m
[0;32mI (341) spi_flash: detected chip: gd[0m
[0;32mI (344) spi_flash: flash io: dio[0m
[0;33mW (348) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (362) sleep: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (368) sleep: Enable automatic switching of GPIO sleep configuration[0m
[0;32mI (375) app_start: Starting scheduler on CPU0[0m
[0;32mI (380) main_task: Started on CPU0[0m
[0;32mI (380) main_task: Calling app_main()[0m
[0;32mI (390) wifi softAP: ESP_WIFI_MODE_AP[0m
[0;32mI (390) pp: pp rom version: de8c800[0m
[0;32mI (400) net80211: net80211 rom version: de8c800[0m
I (410) wifi:wifi driver task: 3fcb9114, prio:23, stack:6144, core=0
I (410) wifi:wifi firmware version: 91b9630
I (410) wifi:wifi certification version: v7.0
I (410) wifi:config NVS flash: enabled
I (410) wifi:config nano formating: enabled
I (420) wifi:Init data frame dynamic rx buffer num: 32
I (420) wifi:Init static rx mgmt buffer num: 5
I (430) wifi:Init management short buffer num: 32
I (430) wifi:Init dynamic tx buffer num: 32
I (440) wifi:Init static tx FG buffer num: 2
I (440) wifi:Init static rx buffer size: 1600
I (440) wifi:Init static rx buffer num: 10
I (450) wifi:Init dynamic rx buffer num: 32
[0;32mI (450) wifi_init: rx ba win: 6[0m
[0;32mI (460) wifi_init: tcpip mbox: 32[0m
[0;32mI (460) wifi_init: udp mbox: 6[0m
[0;32mI (460) wifi_init: tcp mbox: 6[0m
[0;32mI (470) wifi_init: tcp tx win: 5744[0m
[0;32mI (470) wifi_init: tcp rx win: 5744[0m
[0;32mI (480) wifi_init: tcp mss: 1440[0m
[0;32mI (480) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (480) wifi_init: WiFi RX IRAM OP enabled[0m
W (490) wifi:Affected by the ESP-NOW encrypt num, set the max connection num to 2
[0;32mI (500) phy_init: phy_version 270,39d177a,Oct 13 2023,16:18:10[0m
I (530) wifi:mode : softAP (34:98:7a:0e:cc:e5)

[2024-01-17 16:35:48.554]
RX：I (660) wifi:Total power save buffer number: 16
I (660) wifi:Init max length of beacon: 752/752
I (660) wifi:Init max length of beacon: 752/752
[0;32mI (670) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1[0m
[0;32mI (670) wifi softAP: wifi_init_softap finished. SSID:espnow password:espnow135246 channel:7[0m
[0;32mI (680) main_task: Returned from app_main()[0m
```

larryli, 2024-01-17:
`python build.py menuconfig -> Component config -> Hardware Settings -> Main XTAL Config -> Main XTAL frequency -> 40 MHz`

heculess, 2024-01-17:
thanks.
it is usefull

closed 2024-01-17 by heculess