espressif/esp-idf #12995: Esp32-C6 API for disabling RTC watchdog

txf-, 2024-01-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

As per the C6 feature support thread, I am now posting this as a general issue and whilst in my case it applies to C6, I think it may also be a problem on anything other than the ESP32 and S2. 

[image]

On menuconfig, there is is option to allow disabling the watchdog in user code. I can't actually find any documentation related to this for anything other than the above mentioned microcontrollers.

As far as I can tell there `rtc_wdt_disable()` and `rtc_wdt_feed()`, do not apply to anything but the ESP32 and ESP32-S2 as those functions seem to be guarded by ifdefs for those devices.

Do equivalent functions exist for other devices?

I am currently using the 5.2 release branch

KonstantinKondrashov, 2024-01-18:
Hi @txf-!
Thanks for noticing that.
Yes, these functions do exist only for ESP32 and S2. The API is obsolete so you need to use a new one (wdt_hal) which is available for all chips.
```cmake
    if(CONFIG_IDF_TARGET_ESP32 OR CONFIG_IDF_TARGET_ESP32S2)
        list(APPEND srcs "rtc_wdt.c")
    endif()
```
I will update the Kconfig description to mention to use these APIs instead.
```c
wdt_hal_context_t rtc_wdt_ctx = RWDT_HAL_CONTEXT_DEFAULT();
wdt_hal_write_protect_disable(&rtc_wdt_ctx);
wdt_hal_disable(&rtc_wdt_ctx);
wdt_hal_write_protect_enable(&rtc_wdt_ctx);
```

There is a comment about this option in the doc - https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html#watchdog

txf-, 2024-01-18:
Thanks, comment in the doc also needs to be changed, because it is talking about the same functions as the Kconfig.

closed 2024-02-04 by espressif-bot