espressif/esp-idf #12996: ESP32-P4 support status

igrr, 2024-01-17:
# ESP32-P4 support status in ESP-IDF

ESP-IDF v5.3, currently in development, will include initial support for the mass production version of the ESP32-P4 chip.

In the list below, supported features are marked with a checked checkbox, unsupported features have an unchecked checkbox. An internal issue reference (such as "IDF-1234") is listed at the end of the feature description to help us keep this list up to date:

- [ ] Unsupported feature (IDF-1234)
- [X] Supported feature

This list will be updated as the support for ESP32-P4 is being developed.

## Reporting issues about ESP32-P4

If you have an issue to report about any of the ESP32-P4 features, please create a separate issue. This issue is only intended for announcements of ESP32-P4 support status.

---

- Low Power System
  - [X] Low Power Core (RISC-V)
  - [ ] Low-power ADC (IDF-6875)
  - [ ] Low-power GPIO (IDF-7536)
  - [ ] Low-power I2C Master (IDF-7540)
  - [ ] Low-power I2S (IDF-7537)
    - [ ] Voice Activity Detection (IDF-6517)
  - [ ] Low-power SPI (IDF-7538)
  - [ ] Low-power Temperature Sensor (IDF-7535)
  - [ ] Low-power Timer (IDF-7532)
  - [ ] Low-power UART (IDF-7533)
  - [X] Low-power Watchdog Timer
- Peripherals
  - ADC
    - [ ] ADC Calibration (IDF-7467)
    - [ ] ADC continuous mode driver (IDF-6497)
    - [ ] ADC oneshot mode driver (IDF-6496)
    - [ ] Support Low Power ADC instance (IDF-7468)
  - DMA (Direct Memory Access)
    - [ ] 2DDMA (IDF-6498)
    - [X] DMA channel allocator
    - [X] GDMA
      - [ ] Bit Scrambler (IDF-6877)
      - [X] CRC Calculator
    - [X] GDMA2
  - Ethernet driver
    - [ ] SPI Ethernet driver (IDF-7488)
    - [ ] internal MAC driver (IDF-6514)
  - [X] Event Task Matrix
  - [X] GPIO driver
    - [X] Analog Comparator
    - [X] GPIO ETM
    - [ ] Glitch Filter (IDF-7481)
    - [X] RTC IO (LP IO) driver
  - [X] GPIO sigma delta driver
  - GPSPI
    - [X] SPI Master driver
    - [X] SPI Slave HD driver
    - [X] SPI Slave driver
    - [ ] Support Low Power SPI instance (IDF-7504)
  - [X] GPTimer driver
  - [ ] H264 Encoder (New) (IDF-6876)
  - I2C
    - [X] I2C master driver
      - [ ] Support Low Power I2C instance (IDF-7490)
    - [X] I2C slave driver
  - I2S
    - [X] I2S driver
      - I2S PDM rx mode
      - I2S PDM tx mode
      - I2S STD mode
      - I2S TDM mode
    - [ ] I2S legacy driver (IDF-8194)
  - I3C
    - [ ] I3C Master Driver (IDF-6871)
    - [ ] I3C Slave Driver (IDF-6872)
  - [ ] Image Signal Processor (IDF-6503)
  - JPEG
    - [ ] JPEG Decoder Driver (IDF-6513)
    - [ ] JPEG Encoder Driver (IDF-6512)
  - [ ] LCD driver (IDF-7465)
    - I2C LCD driver
    - [ ] MIPI DSI (New) (IDF-6499)
    - RGB LCD driver
    - SPI LCD driver
    - i80 LCD driver
  - [X] LEDC driver
    - Gamma Curve Generation
  - [X] MCPWM driver
  - [ ] MIPI CSI (New) (IDF-6500)
  - [X] PCNT driver
  - PSRAM
    - [ ] .bss PSRAM (IDF-7557)
    - [X] PSRAM Device Driver
    - [ ] XIP PSRAM (IDF-7556)
  - Parallel IO
    - [ ] Parallel IO RX Driver (IDF-7471)
    - [X] Parallel IO TX Driver
  - [ ] Pixel Processing Accelorator Driver (IDF-6878)
  - [X] RMT driver
  - SD/SDIO/MMC driver
    - [X] SDMMC Host driver
      - [ ] UHS-I support (IDF-6874)
    - [X] SDSPI Host driver
  - [X] SPI Flash driver
  - Systimer driver
    - [X] Esptimer implementation over systimer
    - [X] OS tick port over systimer
  - [X] Temperature Sensor driver
  - [ ] Touch Sensor driver (IDF-7477)
  - UART
    - [X] Support Low Power UART instance
    - [X] UART FIFO mode driver
    - [ ] UHCI (UART DMA mode) driver (IDF-7507)
  - USB OTG
    - [ ] USB Device (TinyUSB) driver (IDF-7474)
      - [ ] USB 2.0 Device Support (IDF-8774)
    - [ ] USB Host driver (IDF-7473)
      - [ ] USB 2.0 Host Support (IDF-6873)
  - [ ] USB Serial JTAG (IDF-7496)
  - [X] memory copy by DMA
- Power management
  - [ ] Auto Light Sleep (IDF-7530)
  - [ ] Deep Sleep (IDF-7529)
  - [ ] Dynamic Frequency Switch (IDF-7527)
  - [ ] Light Sleep (IDF-7528)
- Security Features
  - [ ] AES accelerator (IDF-6519)
  - [X] Digital Signature driver
  - [X] ECC accelerator
  - [X] ECDSA driver
  - [X] Flash encryption
  - [X] HMAC driver
  - [ ] HUK Generator (New) (IDF-8041)
  - [ ] Key Manager (New) (IDF-7925)
  - [ ] RNG (IDF-6522)
  - [X] RSA hardware accelerator driver
  - [ ] SHA accelerator (IDF-7541)
  - [X] Secure boot
- System Features
  - [X] Bootloader Support
  - [ ] Brownout Detector (IDF-7519)
  - Cache
    - [X] Cache Driver
  - [ ] Clock Glitch Detector (IDF-7508)
  - [ ] Console (IDF-8105)
  - [ ] Cxx Component (IDF-8108)
  - [X] ESP Event
  - [ ] ESP ROM (IDF-8109)
  - [X] ESP Ringbuffer
  - [X] ESP Timer
  - [X] Efuse controller driver
  - [X] Floating Point Unit
  - [X] FreeRTOS
  - [X] MMU
  - MSPI
    - [X] MSPI tuning (Flash & PSRAM DDR and/or over 80 MHz)
    - [ ] SPI Flash auto suspend (IDF-7518)
  - [X] POSIX Threads
  - Panic Handling Features
    - [X] Core Dump
    - [ ] Debug Watchpoint (IDF-8043)
    - [ ] GDB Stub (IDF-7510)
    - [ ] Panic Handler (IDF-7511)
    - [ ] Stack Guard (IDF-7565)
  - [X] Ram App
  - [ ] TCM (IDF-7667)
  - [X] Watch Dog Timers
  - [ ] app trace (IDF-8107)
  - [ ] newlib (IDF-7862)

diplfranzhoepfinger, 2024-01-26:
when do you expect first P4 Docs and Samples ?

Alvin1Zhang, 2024-02-02:
Update on Feb. 2nd, 2024: support of MIPI DSI, 2DDMA, Key Manager, HUK Generator, ESP-ROM, I2S legacy driver have been added.

Alvin1Zhang, 2024-02-02:
> when do you expect first P4 Docs and Samples ?

Thanks for asking and sorry for reply late, please help write to https://www.espressif.com/en/contact-us/sales-questions for more support. Thanks.

Alvin1Zhang, 2024-02-20:
Update on Feb. 20th, 2024: support of USB 2.0 Device Support, USB 2.0 Host Support, Low-power GPIO and Low-power Timer have been added.

Alvin1Zhang, 2024-03-12:
Update on March 12th, 2024: support of Console, as well as SPI Ethernet driver have been added.

TD-er, 2024-03-25:
> > when do you expect first P4 Docs and Samples ?
> 
> Thanks for asking and sorry for reply late, please help write to https://www.espressif.com/en/contact-us/sales-questions for more support. Thanks.

I have seen similar references to the Espressif site like these as a reply on questions to get samples, but you can't apply for non-released samples on that site and also not on the linked sample distributors.
The P4/C5/C61 are not present as an option, nor some generic "pre-production" label.

Alvin1Zhang, 2024-03-27:
Update on March 27th, 2024: support of Light Sleep, Dynamic Frequency Switch, AES Accelerator, Low-power UART, Stack Guard, Glitch Filter, Low-power I2C Master have been added.

Espressif-yangxi, 2024-03-29:
> I have seen similar references to the Espressif site like these as a reply on questions to get samples, but you can't apply for non-released samples on that site and also not on the linked sample [...]

@TD-er We are pleased to take your feedback into account. The new product options ESP32-P4/C5/C61 has been released on the page_[contact us](https://www.espressif.com/en/contact-us/sales-questions). If you are interested in these products and would like to request samples or obtain more information, please don't hesitate to reach out to Espressif customer support team through this[ link](https://www.espressif.com/en/contact-us/sales-questions). We are ready to provide support. Thank you.

edbek, 2024-04-12:
What will be the consumption of the ESP32-P4 processor in BLE mode? Will the same large 40mA remain or will it be within 5mA, like Nordic?
What consumption ESP32-P4 is planned in deep-sleep mode?

igrr, 2024-04-12:
@edbek ESP32-P4 doesn't include any wireless peripherals (Wi-Fi or BLE). Regarding the deep sleep current, please wait a bit for the details, they will be available when the datasheet is published.

Alvin1Zhang, 2024-04-15:
Update on April 15th, 2024: support of Panic Handler, MIPI CSI (New), JPEG Decoder Driver, Deep Sleep, SHA Accelerator, Debug Watchpoint, Auto Light Sleep and RNG have been added.

Alvin1Zhang, 2024-05-07:
Update on May 7th, 2024: support of JPEG Encoder Driver and TCM have been added.

jstoefer, 2024-05-14:
> Update on April 15th, 2024: ... MIPI CSI (New) ...
Does the ISP of the P4 have support for stitching HDR ?
Will there be support for HDR ?

ginkgm, 2024-05-29:
> > Update on April 15th, 2024: ... MIPI CSI (New) ...
> > Does the ISP of the P4 have support for stitching HDR ?
> > Will there be support for HDR ?

No I think the ISR doesn't support HDR.

Alvin1Zhang, 2024-06-14:
Update on June 15th, 2024: support of Parallel IO RX Driver, USB Serial JTAG, Brownout Detector, XIP PSRAM, Pixel Processing Accelerator Driver, USB 2.0 Device Support, SPI Flash auto suspend, ADC oneshot mode driver, ADC continuous mode driver, H264 Encoder (New) and Touch Sensor driver have been added.

Alvin1Zhang, 2024-06-25:
Update on June 25th, 2024: support of GDB Stub has been added.

Alvin1Zhang, 2024-07-03:
Update on July 3rd, 2024: support of Low-power SPI has been added.

Alvin1Zhang, 2024-07-22:
Update on July 22nd, 2024: support of Clock Glitch Detector and .bss PSRAM have been added.

AndyDevLat, 2024-08-14:
> Update on Feb. 20th, 2024: support of USB 2.0 Device Support, USB 2.0 Host Support, Low-power GPIO and Low-power Timer have been added.

Does it mean we can already create applications using USB high speed host but full speed host support is still pending?

AndyDevLat, 2024-08-14:
Noted, thank you. And USB_FS peripheral not support just in host mode? I mean can it be used for device mode via tinyusb?

tore-espressif, 2024-08-14:
> Noted, thank you. And USB_FS peripheral not supported just in host mode? I mean can it be used for device mode via tinyusb?

USB_FS not supported at all at the moment :/

Alvin1Zhang, 2024-08-15:
Update on August 15th, 2024: support of MIPI CSI, ISP DVP, LCD_CAM DVP, Low-power I2S and Wi-Fi Host have been added.

DatanoiseTV, 2024-08-16:
What state is the I2S TDM mode? The examples in 5.3/5.3 don't compile.
```
/Users/dtns/dev/p4/i2s_tdm/main/i2s_tdm_example_main.c:119:5: error: unknown type name 'i2s_tdm_config_t'; did you mean 'i2s_chan_config_t'?
  119 |     i2s_tdm_config_t tdm_cfg = {
      |     ^~~~~~~~~~~~~~~~
      |     i2s_chan_config_t
```

chegewara, 2024-08-30:
Hi,
i see that P4 DMA support is green, but i believe it is not working
https://esp32.com/viewtopic.php?f=12&t=41620

TiKoESoPe, 2024-09-09:
Helklo, is there any news when the RGB LCD driver will be ready for the controller?

sukesh-ak, 2024-09-09:
> Helklo, is there any news when the RGB LCD driver will be ready for the controller?

Have you checked the P4 BSP yet?
https://components.espressif.com/components/espressif/esp32_p4_function_ev_board/versions/3.0.1

TiKoESoPe, 2024-09-09:
Yes, I checked there, but there is only the MIPI interface in use for the board and no RGB LCD interface. In the ESP-IDF I did not see the SOC_LCD_RGB_SUPPORTED for the ESP32-P4 set and was confused whether it is not yet implemented or will not be supported at all. Since it is listed here, I guess it is just not implemented yet.

tom-borcin, 2024-09-11:
Hello @TiKoESoPe, support for RGB LCD Driver is in progress right now. Once the support is done, we will update the status

torntrousers, 2024-09-13:
> Update on Feb. 2nd, 2024: support of MIPI DSI, 2DDMA, Key Manager, HUK Generator, ESP-ROM, I2S legacy driver have been added.

Hello @Alvin1Zhang / @igrr  - interested in the Key Management Unit on the P4 and from that comment back in Feb about "Key Manager" support i hoped there was some support, but asking over here it seems not yet - https://esp32.com/viewtopic.php?f=2&t=41800

Could you give any update on the Key Management Unit ?

Thanks!

TiKoESoPe, 2024-09-13:
Hello @tom-borcin, thanks for the info. I will wait. 👍

AdityaHPatwardhan, 2024-09-16:
> Hello @Alvin1Zhang / @igrr - interested in the Key Management Unit on the P4 and from that comment back in Feb about "Key Manager" support i hoped there was some support, but asking over here it [...]

Hi @torntrousers,
There were some performance issues of the Key Management Unit on the currently available revisions of ESP32-P4 SoC, hence it has been marked as not supported for now.
Once a new revision that fixes the issue is available then the support shall be enabled in the ESP-IDF.

torntrousers, 2024-09-16:
> Hi @torntrousers, There were some performance issues of the Key Management Unit on the currently available revisions of ESP32-P4 SoC, hence it has been marked as not supported for now. Once a new [...]

Do you mean a new hardware revision? I bought a couple of the P4 EV kits especially to try the Key Management Unit so i guess they are no good for this now? Could maybe put a note on the shop page - all it says presently is that its all good except that the ADC is uncalibrated :( Could you post back here when the new rev boards are available?

TD-er, 2024-09-16:
Yep, which silicon revision should have it?
And which revisions can be found out there ?

AdityaHPatwardhan, 2024-09-16:
@torntrousers @TD-er, The work to fix is already on-going, I can only confirm when a new revision is available and has been tested. Ideally it should be the very next revision of ESP32-P4 (which is ECO2). Just to prevent users from we have disabled the support from ESP-IDF and also marked it as non-supported in the above status link.

AdityaHPatwardhan, 2024-09-21:
@ESP32DE Yes, The Key manager shall not be enabled for the ECO1 chip. Apologies for the inconvenience. Please note that no other feature of the SoC is affected due to the Key Manager being disabled.

torntrousers, 2024-09-21:
I bought a couple of P4's just to try the KM, any chance i can trade them in for eco2's when that comes out? 
;)

DatanoiseTV, 2024-09-23:
I have also 30 spare P4 chips, if anyone is interested. Shipping from Germany.

microfoundry, 2024-10-02:
Just curious - the Espressif site states the P4 will have a single I3**S** Interface (which I assume to be SoundWire), yet I see nothing here that would indicate that is in ESP-IDF development. It is a typo (i3c maybe?) or...

tom-borcin, 2024-10-02:
Hi @microfoundry, that indeed looks like a typo. Thank you for notifying us

Alvin1Zhang, 2024-10-08:
Update on October 8th, 2024: support of below features have been added.

* Low-power I2S
* Clock Output
* Support Low Power I2C instance
* I2C LCD driver
* SPI LCD driver
* i80 LCD driver
* External flash support
* ECDSA driver
* HUK Generator
* Key Manager.

torntrousers, 2024-10-08:
> Update on October 8th, 2024: support of below features have been added.
> 
>     * ECDSA driver

Is there any doc or example code for that ECDSA driver that i can see?

ginkgm, 2024-10-08:
> Just curious - the Espressif site** states the P4 will have a single I3**S** Interface (which I assume to be SoundWire), yet I see nothing here that would indicate that is in ESP-IDF development. It is a typo (i3c maybe?) or...
> 
> **Specifically on the Espressif Product Selector site/P4 Product Brief/Features/Communication Interfaces

Hi @microfoundry , thanks for your question.

It is supposed to be an I3C peripheral to work with camera sensors. Could you provide link to where you saw it on our official site, so that we can correct this error?

microfoundry, 2024-10-08:
[image]

HareshPrajapati, 2024-10-16:
> The Key manager shall not be enabled for the ECO1 chip

Is it possible to ship to india?

HareshPrajapati, 2024-10-16:
> I have also 30 spare P4 chips, if anyone is interested. Shipping from Germany.

Is it possible to ship to india?

tom-borcin, 2024-10-16:
> > Update on October 8th, 2024: support of below features have been added.
> > ```
> > * ECDSA driver
> > ```
> 
> Is there any doc or example code for that ECDSA driver that i can see?

Hi @torntrousers, here is:
1. ECDSA documentation: https://docs.espressif.com/projects/esp-idf/en/stable/esp32h2/api-reference/peripherals/ecdsa.html 
2. Our own test: https://github.com/espressif/esp-idf/blob/master/components/hal/test_apps/crypto/main/ecdsa/test_ecdsa.c

Emill, 2024-10-20:
Is there any information or reference about the so called "AI instructions"? Is it the same instruction set as the ESP32-S3 processor instruction extensions or a new one?

PhilippvK, 2024-10-21:
> Is there any information or reference about the so called "AI instructions"? Is it the same instruction set as the ESP32-S3 processor instruction extensions or a new one?

@Emill I couldn’t find any official documentation/specification of the ISA implemented on the P4, but the Espressif LLVM toolchain already supports it. Feel free to look at the implemented instructions: https://github.com/espressif/llvm-project/blob/e9c8e60ee8d58e21a7f3582a4695eeb99ec4c5d7/llvm/lib/Target/RISCV/RISCVInstrInfoESP32P4.td

Emill, 2024-10-21:
> @Emill I couldn’t find any official documentation/specification of the ISA implemented on the P4, but the Espressif LLVM toolchain already supports it. Feel free to look at the implemented [...]

Thanks! It seems they have mostly preserved the same instructions as in ESP32-S3 but added a few new instructions as well.

foxnux, 2024-11-15:
Is there any **ETA** for **RGB LCD driver (IDF-10962)**?
I got some ESP32-P4 samples and designed our own PCB, but I am not able to start testing a RGB565 LCD panel.

suda-morris, 2024-11-15:
@foxnux RGB_LCD driver is already available on release/v5.4 and master branch. https://github.com/espressif/esp-idf/commit/efcb91b47eb5a6a309d6f455e4955584be1a7f30

Alvin1Zhang, 2024-11-15:
Update on November 15th, 2024: support of below features have been added.

* Voice Activity Detection
* Image Signal Processor
* RGB LCD driver
* UHS-I support
* app trace

foxnux, 2024-11-15:
@suda-morris Thank you for the good news. Yes! It was already added almost three weeks ago.
I should keep checking the master branch. Sorry for the noise.

Yep, once we use up our stock of tens of thousands of RGB LCDs, we will switch to MIPI LCDs. ;)

Alvin1Zhang, 2024-11-28:
Thanks for subscribing to this ticket, only remaining items are:

	* Peripherals
		* Low-power Temperature Sensor (IDF-7535)
		* ADC Calibration (IDF-7467)
		* Support Low Power ADC instance (IDF-7468)
		* Bit Scrambler (IDF-6877)
		* Support Low Power SPI instance (IDF-7504)
		* I3C Master Driver (IDF-6871)
		* I3C Slave Driver (IDF-6872)
		* UHCI (UART DMA mode) driver (IDF-7507)
	* Power management
		* Sleep retention (IDF-11007)
	* Security
		* ECDSA driver
		* HUK Generator
		* Key Manager

There will be updates for these remaining items when they are supported. We are closing the issue and unpinning it. Please open new issue if you have a feature request, question or any other problem. Thanks.

AndyDevLat, 2024-11-28:
Hi!

This issue was closed, but as i understand USB FS is  still not working?
Is it software issue or some hardware problem with the current chip revision?

Regards,

Andrew

Closed #12996<https://github.com/espressif/esp-idf/issues/12996> as completed.

tore-espressif, 2024-11-28:
> This issue was closed, but as i understand USB FS is  still not working?

You're correct. We have tested internal FS device mode, which will be added after we upgrade to TinyUSB v0.17, by end of December.

FS Host is not yet fully tested, but there is no HW issue, only missing SW support

AndyDevLat, 2024-11-28:
Hi!

Thank you for information!
In fact i saw that TinyUSB already has some DWC2 support and to enable USB FS device (this is actually what we need in our project)
you need to set #define CFG_TUSB_RHPORT0_MODE       OPT_MODE_DEVICE | OPT_MODE_FULL_SPEED
Then it uses USB FS registers, defined as #define DWC2_FS_PERIPH_BASE  0x60080000UL.
But it did not work for some reason - dcd_init failed to initialize controller.
Thus I assumed there could be some hardware problem or you also need to also enable it somehow via HAL.
Additionally it is not clear to me  how to control pins for USB FS and USB Jtag/Serial: preliminary datasheet says they can be
GPIO24/25 or GPIO 26/2 but i could not figure out where it is controlled, so i thought i was missing something.
Hope this is sorted out soon.

Thank you!

Regards,

Andrew

PaulAnurag, 2024-12-23:
Hi
Is there any documentation with regards to the changes made from ESP32-P4_Function_EV_Board V1.0B to V1.2.