espressif/esp-idf #13000: Failed to create task using xTaskCreate() despite having sufficient heap available.

Josh-TapNoa, 2024-01-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### Espressif SoC revision.

ESP32S2

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S2 WROOM-I

### Power Supply used.

External 3.3V

### What is the expected behavior?

For xTaskCreate to push the newly created task onto the task list.

### What is the actual behavior?

xTaskCreate returns pdFAIL

### Steps to reproduce.

`bool ota_update_initiator_transfer_task_start(esp_partition_t *target_partition, uint8_t *target_mac, TaskHandle_t calling_task)
{
    initiator_task_params_t *task_params = (initiator_task_params_t *)malloc(sizeof(initiator_task_params_t));
    if (task_params == NULL)
    {
        ESP_LOGE(TAG, "Failed to allocate memory for ota initiator task parameters!");
        return false;
    }
    task_params->target_partition = target_partition;
    task_params->target_mac = target_mac;
    calling_task_initiator = calling_task;

    if(target_mac == NULL){
        printf("Received a null reference to the target mac! \n");
    }

    if(target_partition == NULL){
        printf("Received a null reference to the target partition! \n");
    }

    ESP_LOGI(TAG, "Starting initiator task!");
    
    TaskHandle_t created_task_handle = NULL;
    size_t total_heap_size = heap_caps_get_total_size((MALLOC_CAP_INTERNAL));
    printf("Free Heap Size: %d bytes\n", total_heap_size);

    BaseType_t task_creation_result = xTaskCreate(ota_update_initiator_transfer_task, "OTAInitiatorTask", INITIATOR_STACK_SIZE, task_params, configMAX_PRIORITIES -1, &created_task_handle);

    vTaskDelay(pdMS_TO_TICKS(75));

    printTaskList();

    if (task_creation_result == pdPASS)
    {
        ESP_LOGI(TAG, "Initiator task started successfully! Task handle: %p", created_task_handle);
        return true;
    }
    else
    {
        ESP_LOGE(TAG, "Initiator task failed to start!");
        free(task_params); // Free allocated memory in case of failure
        return false;
    }
}`

This is part of my "local OTA" system, that sends firmware updates locally over ESPNOW. 

It has the following serial output: 

I (56124) Local Coms OTA - Initiator: Starting initiator task!
Free Heap Size: 189136 bytes
Tasks:  Status: TaskID: Priority:       Stack HWM:      TaskNum:        CPU:    RunCount:
 hub_local_ota          X       24      1260    18
IDLE            R       0       1144    3
mqtt_task       B       5       3508    16
Tmr Svc         B       1       116     4
rmaker_queue_ta B       5       2036    10
Button (GPIO:6) B       0       1352    13
tiT             B       18      1476    6
btm_rrm_t       B       2       6332    9
Button (GPIO:34 S       0       2740    14
sys_evt         B       20      708     7
Main Controller S       0       1184    15
espnow_main     B       1       2288    21
esp_timer       S       22      1936    1
wifi            B       23      4040    8
Flow Sensor     S       0       1716    5
 
E (56235) Local Coms OTA - Initiator: Initiator task failed to start!
I (56243) Network: Local OTA Failed!

### More Information.

I am running on Ubuntu 20.04 LTS with a ESP IDF 5.1 Docker container that has been modified to carry the ESPRainmaker source code from a particular stable commit and manage a bunch of command Alias'

KonstantinKondrashov, 2024-01-17:
@Josh-TapNoa 
You use the total heap size function. You need to use another function to get free heap size.
heap_caps_get_total_size -> heap_caps_get_free_size

Josh-TapNoa, 2024-01-17:
@KonstantinKondrashov  Thanks for the quick reply. I do indeed call the total instead of free. That is my mistake, I have been experimenting with the memory and task debugging functions and left that in there by mistake. 

I would be right in concluding that xTaskCreate only returns pdFAIL in the case that the memory could not be allocated for the stack and/or buffer?

KonstantinKondrashov, 2024-01-17:
Yes, according to the doc (https://www.freertos.org/a00125.html), xTaskCreate returns pdPASS or errCOULD_NOT_ALLOCATE_REQUIRED_MEMORY (if memory is not enough to allocate the requested stack size).
You might need to decrease the INITIATOR_STACK_SIZE to get it to work.

Josh-TapNoa, 2024-01-17:
Many thanks @KonstantinKondrashov will close this as is a non issue!

closed 2024-01-17 by Josh-TapNoa