espressif/esp-idf #13004: Esp32 unable to receive ota firmware having connection issue. poll_read select error 113, errno = Software caused connection abort, fd = 54

USAMA-SADIQ73, 2024-01-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I expect it to run as i am following as provided readme file is instruction but unable to update the firmware.

### What is the actual behavior?

i am unable to receive the firmware update over the ota. i ma using the espidf ota example "simple_ota_example". Server is configured properly but unable to receive to receive the firmware update. getting the following error message. it gets stop here. need help in this regard.Thanks

### Steps to reproduce.

1. Step
   I am using simple_ota_example provided by espidf and configured is url and wifi ssid and passward
2. Step
 setting spi flash size to 4mb. and clock speed to 240MHz
3. Step
 generated cetificated and save them to esp example directory and flash the code
4. Step
   than run the http server as instructed by readme file
5. Step
 server is running and after booting the esp32 it get stop at while receiving firmware from the server anf after some time connection get lost as shown in build log 
[image]
[image]

### Build or installation Logs.

```plain
I (7175) simple_ota_example: Starting OTA example task
I (7185) simple_ota_example: Attempting to download update from https://192.168.10.19:8070/hello_world.bin
I (7195) main_task: Returned from app_main()
I (7465) esp-x509-crt-bundle: Certificate validated
I (9575) esp_https_ota: Starting OTA...
I (9585) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
E (321945) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (321945) HTTP_CLIENT: transport_read: error - 57347 | ERROR
E (321945) esp_https_ota: data read -1, errno 0
E (321955) simple_ota_example: Firmware upgrade failed
```

nileshkale123, 2024-01-22:
@USAMA-SADIQ73 Thank you for bringing this matter to our attention.

Could you please furnish the debug-enabled logs for the same issue? To enable debug-level logs, navigate to 'idf.py menuconfig -> component config -> log output -> Default log verbosity' and set it to 'Debug'.

Subsequently, rebuild the application and save the generated logs. Your cooperation is greatly appreciated.

Alternatively, consider initiating a Python-based HTTPS server instead of OpenSSL. You can find detailed instructions for launching a Python-based server using the pytest_simple_ota.py file in the corresponding README.md. Below are the specific steps:
 
Firstly, you have to enable pytest for starting server using this way. Steps for it as follows:

1. Navigate to the main directory where you have cloned esp-idf.
2. Run command to enable pytest
     - ' . ./install.sh --enable-pytest '
     - ' . ./export.sh '

Once the pytest is enabled, here are detailed steps for initiating the server: 
1. Navigate to the 'simple_ota_example' directory.
2. Create a directory named 'server'.
3. Go into the 'server' directory.
4. Generate server certificates using the command specified in the README file:
     - 'openssl req -x509 -newkey rsa:2048 -keyout ca_key.pem -out ca_cert.pem -days 365 -nodes'
5. Copy the generated 'ca_cert.pem' file to 'simple_ota_example/server_certs/ca_cert.pem' using command:
     - 'cp ./ca_cert.pem ./../../simple_ota_example/server_certs/ca_cert.pem'
6. Navigate back to the 'simple_ota_example' directory
7. Execute the following command to start the server:
     - 'python3 pytest_simple_ota.py <binary file path> 8070 ./server/'

KaeLL, 2024-02-02:
@nileshkale123 Step 7 results in `ModuleNotFoundError: No module named 'common_test_methods'`

nileshkale123, 2024-02-07:
Hello @KaeLL 

You should append the path **'/tools/ci/python_packages'** to the IDF_PATH environment variable.

You can achieve this by executing the following command. Before this, please ensure that the IDF_PATH environment variable is already set to the IDF installation directory.

`export IDF_PATH="$IDF_PATH:$IDF_PATH/tools/ci/python_packages"`

KaeLL, 2024-02-07:
Shouldn't `export.sh` do that automatically?

nileshkale123, 2024-02-08:
@KaeLL 
These packages are exported during the CI setup process. Since this issue is quite common, we've already addressed it and it's currently in progress. You can expect to see the fix implemented in the next updates.

nileshkale123, 2024-02-20:
@KaeLL 
The issue you mentioned has been resolved and backported. You can find the solution in commit ID: 318038a746ccb060f17e3c5ccac50060e7d67afa.

nileshkale123, 2024-02-20:
I am closing this issue now. Thank you.

closed 2024-02-20 by nileshkale123