espressif/esp-idf #13006: Onewire RMT multiple buses?

rpaulbeam, 2024-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have built and tested the onewire_example for the ESP32-S2, and it works fine for a single onewire bus.  I was hoping to implement 3 buses, since my understanding is the RMT peripheral has 6 channels and each onewire bus takes 2.  However, rmt_rx_register_to_group() fails on the 2nd bus.  Is it intended that a single onewire bus should occupy the entire RMT peripheral, or is it a bug that it cannot find an unused channel?

rpaulbeam, 2024-01-18:
After review, the S2 has only 4 RMT channels, so it could, at most, do 2 onewire buses. but 64  words/channel should still fit in the 256 word memory of the S2?

suda-morris, 2024-01-18:
Hi @rpaulbeam The one-wire bus won't take off all the RMT channels. On most of esp chips except the esp32 and esp32s2, it takes 1 TX and 1 RX channels to construct the Onewire bus.

You're using esp32s2, it has 4 channel in total. Each channel can be configured as TX or RX. One limitation in the ESP32S2 RMT is, the RX can't support "RX ping-pong". So we have to prepare a big enough FIFO memory. This memory size can be tuned by `onewire_bus_rmt_config_t::max_rx_bytes`. In our DS18B20 example, we're setting it to 10 bytes. Internally, this will consume 10*8 RMT symbols(words). But each RMT channel can only provide 64 words, so, we need one more channel. All in all, on esp32s2, one 1-wire bus usually consumes 1TX and 2 RX channels.

rpaulbeam, 2024-01-18:
That is helpful.   I didn’t realize 2 rx channels were involved.   So the S3 should be able to do 2 onewire buses since it has 8 rmt channels?

suda-morris, 2024-01-18:
ESP32S3 can build up to 4 onewire buses.

rpaulbeam, 2024-01-18:
I’ll look at making that switch.   Thanks.

closed 2024-01-18 by suda-morris