espressif/esp-idf #13007: SSLCertVerificationError

tao-a-man, 2024-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I try to build app with script idf.py build

### What is the actual behavior?

I have Error SSLCertVerificationError

### Steps to reproduce.

1. Step idf.py build
[CMakeLists.txt](https://github.com/espressif/esp-idf/files/13973318/CMakeLists.txt)

### Build or installation Logs.

```plain
CMake Error at C:/Espressif/frameworks/esp-idf-v5.1.2/tools/cmake/build.cmake:540 (message):
  WARNING: HTTPSConnectionPool(host='components-file.espressif.com',
  port=443): Max retries exceeded with url:
  /components/espressif/esp_lcd_touch_cst816s.json (Caused by
  SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED]
  certificate verify failed: self signed certificate in certificate chain
  (_ssl.c:992)')))
```

### More Information.

[image]
SSLCertVerificationError

tao-a-man, 2024-01-18:
@igrr

larryli, 2024-01-18:
Disable the local proxy, or check your router

tao-a-man, 2024-01-18:
> Disable the local proxy, or check your router

i did research so much, maybe solution is grant permissions to esp-idf pass verify 😘

kumekay, 2024-01-18:
Hello @tao-a-man 

Some of our clients experienced a similar problem on some corporate-managed computers with Windows and strict network policies. Can it be your case?

You can try installing the python package to use system root certificates. 
Open your ESP-IDF shell and run 

```
pip install pip_system_certs
```

tao-a-man, 2024-01-18:
> Hello @tao-a-man
> 
> Some of our clients experienced a similar problem on some corporate-managed computers with Windows and strict network policies. Can it be your case?
> 
> You can try installing the python package to use system root certificates. Open your ESP-IDF shell and run
> 
> ```
> pip install pip_system_certs
> ```
> 
> Please let us know if it solves your problem.

thank you so much for reply, You got understand the problem correctly, but it not work, i did try your script but then i run idf.py build issue still happened

tao-a-man, 2024-01-18:
> @tao-a-man Then we probably need more information about particular restrictions in your network.
> 
> If you are located in mainland China, you may try to use our mirror in Beijing. To do so set an environment variable IDF_COMPONENT_STORAGE_URL to `https://components-file.espressif.cn/` through the Windows settings. Or, to do it temporary, run `$Env:IDF_COMPONENT_STORAGE_URL = "https://components-file.espressif.cn/"` run this command in IDF PowerShell terminal.

hii @kumekay sorry but i live in VietNam, my network team say: your link have cert selfsign, that why your link had been auto block when i run idf.py build 😭😭 sorry for your time thank you so much, anyway if you have idea for issue pleass help me ❤️

larryli, 2024-01-19:
Please check components-file.espressif.cn cert:
`openssl s_client -showcerts -servername components-file.espressif.cn -connect components-file.espressif.cn:443 </dev/null`

```
CONNECTED(00000003)
depth=2 C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert Global Root CA
verify return:1
depth=1 C = US, O = DigiCert Inc, CN = DigiCert TLS RSA SHA256 2020 CA1
verify return:1
depth=0 C = CN, ST = \E4\B8\8A\E6\B5\B7\E5\B8\82, O = \E4\B9\90\E9\91\AB\E4\BF\A1\E6\81\AF\E7\A7\91\E6\8A\80\EF\BC\88\E4\B8\8A\E6\B5\B7\EF\BC\89\E8\82\A1\E4\BB\BD\E6\9C\89\E9\99\90\E5\85\AC\E5\8F\B8, CN = *.espressif.cn
verify return:1
---
Certificate chain
 0 s:C = CN, ST = \E4\B8\8A\E6\B5\B7\E5\B8\82, O = \E4\B9\90\E9\91\AB\E4\BF\A1\E6\81\AF\E7\A7\91\E6\8A\80\EF\BC\88\E4\B8\8A\E6\B5\B7\EF\BC\89\E8\82\A1\E4\BB\BD\E6\9C\89\E9\99\90\E5\85\AC\E5\8F\B8, CN = *.espressif.cn
   i:C = US, O = DigiCert Inc, CN = DigiCert TLS RSA SHA256 2020 CA1
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Mar 10 00:00:00 2023 GMT; NotAfter: Feb  5 23:59:59 2024 GMT
-----BEGIN CERTIFICATE-----
MIIG3DCCBcSgAwIBAgIQB5vbRc06vgDkZZlH6vl1XzANBgkqhkiG9w0BAQsFADBP
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMSkwJwYDVQQDEyBE
aWdpQ2VydCBUTFMgUlNBIFNIQTI1NiAyMDIwIENBMTAeFw0yMzAzMTAwMDAwMDBa
Fw0yNDAyMDUyMzU5NTlaMHUxCzAJBgNVBAYTAkNOMRIwEAYDVQQIDAnkuIrmtbfl
uIIxOTA3BgNVBAoMMOS5kOmRq+S/oeaBr+enkeaKgO+8iOS4iua1t++8ieiCoeS7
veaciemZkOWFrOWPuDEXMBUGA1UEAwwOKi5lc3ByZXNzaWYuY24wggEiMA0GCSqG
SIb3DQEBAQUAA4IBDwAwggEKAoIBAQC2NfZY+NmNcalec6QwmvlPjUh6Cba1clIf
IhwNMrWeKKHu1g+a6hkm3OWDvoGO13zRyEFXRAECwdk+SRiQoNboTqwIAjkJAtxT
VYiTNaxzzXIJ72O+iBNFZao7ZoyR9+XEIY0tA9ddbEzFQ8NE92x0DloZ9T9In1JV
GCtmfnNxO+3iWerRCSQcAY/oO2sQVEvDJSb5rV7BI0NSZKaxOTsfKbYYy65cX2e7
sJdQY6j7I8ZYo3ogHNfShpoz6j46CXuQyAywmiesRYxw7OIV6hA39YcZrkEgt8/1
LJpwOmibqtceef8J+b5JO12Ap8F/M+ZYs9CkcIUztBzyGlRSKOcTAgMBAAGjggOM
MIIDiDAfBgNVHSMEGDAWgBS3a6LqqKqEjHnqtNoPmLLFlXa59DAdBgNVHQ4EFgQU
M6cGaaJuB2XE42iHxyu9v3z+pFIwOQYDVR0RBDIwMIIOKi5lc3ByZXNzaWYuY26C
EGVoci5lc3ByZXNzaWYuY26CDGVzcHJlc3NpZi5jbjAOBgNVHQ8BAf8EBAMCBaAw
HQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMIGPBgNVHR8EgYcwgYQwQKA+
oDyGOmh0dHA6Ly9jcmwzLmRpZ2ljZXJ0LmNvbS9EaWdpQ2VydFRMU1JTQVNIQTI1
NjIwMjBDQTEtMi5jcmwwQKA+oDyGOmh0dHA6Ly9jcmw0LmRpZ2ljZXJ0LmNvbS9E
aWdpQ2VydFRMU1JTQVNIQTI1NjIwMjBDQTEtMi5jcmwwPgYDVR0gBDcwNTAzBgZn
gQwBAgIwKTAnBggrBgEFBQcCARYbaHR0cDovL3d3dy5kaWdpY2VydC5jb20vQ1BT
MH0GCCsGAQUFBwEBBHEwbzAkBggrBgEFBQcwAYYYaHR0cDovL29jc3AuZGlnaWNl
cnQuY29tMEcGCCsGAQUFBzAChjtodHRwOi8vY2FjZXJ0cy5kaWdpY2VydC5jb20v
RGlnaUNlcnRUTFNSU0FTSEEyNTYyMDIwQ0ExLmNydDAJBgNVHRMEAjAAMIIBfgYK
KwYBBAHWeQIEAgSCAW4EggFqAWgAdwB2/4g/Crb7lVHCYcz1h7o0tKTNuyncaEIK
n+ZnTFo6dAAAAYbKuk4jAAAEAwBIMEYCIQCZpnROPCzG3ZLDFRiYGM0SG5mE9j3r
Z11rmDYBx7CWWgIhANt2DzHMDKstKidPOkUDzHKH+C0JRaE0qBI6R4C9fQVFAHUA
c9meiRtMlnigIH1HneayxhzQUV5xGSqMa4AQesF3crUAAAGGyrpOYQAABAMARjBE
AiBC3YTFoP0X+cC/tSfI0JWqO6clNm6Wl6Dd/oLgmhSsjgIgbmqba6RKLzGhpCQ9
LynazsSzjZFnTPgRbTmi29bkLxcAdgBIsONr2qZHNA/lagL6nTDrHFIBy1bdLIHZ
u7+rOdiEcwAAAYbKuk4VAAAEAwBHMEUCIFr4ZwPFLhVmI2LK7NU4yf3IRNB7a7+z
QrwqBw36VGEvAiEAoXZB7pmAEu6wZ0rb9gplKlCPyOkbfuTWb1l+IfrpwCEwDQYJ
KoZIhvcNAQELBQADggEBACqf+wKijqQlKXL76I6jRNH+kqcfplvPMnOx+CGAYbFU
+15O13roc0kp6akbASyY6Ycv66doT8+3NnQ5sYoXHgsLvLIBS8N+1KPmb4ktyoOV
WXMmKi6sGpzmfg2Xq7bWhUG2dp1bmBYErhVlgKiBPEJWFdvzK0pzLzGLNzCo6l9q
gd/c4NlOh4edoxo2r08kJVabmJpzpYD+5mUyXXTQSSJZ2EWePb28d9OqiP0GmUah
Pv567p3j6dUq+gz1Q7OttkXxF75yMJnSRk7KU3s9nac0ddAW5i3HHSqJ3KHh/v1T
Pb7aDGuFYnP4cXS+DM1qp6suxeZx59AWBvPlHkZ2YPg=
-----END CERTIFICATE-----
 1 s:C = US, O = DigiCert Inc, CN = DigiCert TLS RSA SHA256 2020 CA1
   i:C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert Global Root CA
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Sep 24 00:00:00 2020 GMT; NotAfter: Sep 23 23:59:59 2030 GMT
-----BEGIN CERTIFICATE-----
MIIE6jCCA9KgAwIBAgIQCjUI1VwpKwF9+K1lwA/35DANBgkqhkiG9w0BAQsFADBh
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBD
QTAeFw0yMDA5MjQwMDAwMDBaFw0zMDA5MjMyMzU5NTlaME8xCzAJBgNVBAYTAlVT
MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxKTAnBgNVBAMTIERpZ2lDZXJ0IFRMUyBS
U0EgU0hBMjU2IDIwMjAgQ0ExMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKC
AQEAwUuzZUdwvN1PWNvsnO3DZuUfMRNUrUpmRh8sCuxkB+Uu3Ny5CiDt3+PE0J6a
qXodgojlEVbbHp9YwlHnLDQNLtKS4VbL8Xlfs7uHyiUDe5pSQWYQYE9XE0nw6Ddn
g9/n00tnTCJRpt8OmRDtV1F0JuJ9x8piLhMbfyOIJVNvwTRYAIuE//i+p1hJInuW
[... 18 lines cut ...]
-----END CERTIFICATE-----
---
Server certificate
subject=C = CN, ST = \E4\B8\8A\E6\B5\B7\E5\B8\82, O = \E4\B9\90\E9\91\AB\E4\BF\A1\E6\81\AF\E7\A7\91\E6\8A\80\EF\BC\88\E4\B8\8A\E6\B5\B7\EF\BC\89\E8\82\A1\E4\BB\BD\E6\9C\89\E9\99\90\E5\85\AC\E5\8F\B8, CN = *.espressif.cn
issuer=C = US, O = DigiCert Inc, CN = DigiCert TLS RSA SHA256 2020 CA1
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA-PSS
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 3571 bytes and written 394 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_128_GCM_SHA256
Server public key is 2048 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
DONE
```

tao-a-man, 2024-01-19:
> Please check components-file.espressif.cn cert: `openssl s_client -showcerts -servername components-file.espressif.cn -connect components-file.espressif.cn:443 </dev/null`
>
> ```
> CONNECTED(00000003) [...]

Hii @larryli, Thanks Sir for reply, but my directory structure like bellow, sorry for my stupid question, Where should I insert that code?
[image]

tao-a-man, 2024-01-19:
> components-file

hii @larryli Thanks Sir for reply, but I live in VietNam, Can i change link "components-file.espressif.cn" to "components-file.espressif.com"?

tao-a-man, 2024-01-19:
hii @larryli That script return like bellow 😔
[image]

larryli, 2024-01-19:
Please contact support@fortinet.com

Your network security provider replaces the espressif certificates(signed by digicert.com) to self-signed.

tao-a-man, 2024-01-19:
> Please contact [support@fortinet.com](mailto:support@fortinet.com)
> 
> Your network security provider replaces the espressif certificates(signed by digicert.com) to self-signed.

Yes you true @larryli, my network security did fix it, thank you so much for your help, Have a good day, Happy new Lunar Year 🐲

tao-a-man, 2024-01-22:
hii @larryli Excuse me, I have a issue, I think u know that, Plss help me idea for it, I did run idf.py build then get issue like bellow (picture 1), The problem lies in the line of code in file build.cmake (picture 2), I think my network security did block your link then i can't connect to it, Plss tell me all link need to run file build.cmake success, Because i need tell with my network security open that link, Thank youu ❤️
[image]
[image]

larryli, 2024-01-22:
@tao-a-man 

https://github.com/espressif/idf-component-manager/issues/27#issuecomment-1472769021

https://components-file.espressif.com/
https://api.components.espressif.com/
https://components-file.espressif.com/

kumekay, 2024-01-22:
@tao-a-man 

https://api.components.espressif.com/ is a legacy link and not used by the component manager.

When you only download components, it should access https://components-file.espressif.com/
Depending on configuration, it may also a call to https://components.espressif.com/api/

kumekay, 2024-01-30:
@tao-a-man If it works for you, I'm closing this issue, feel free to reopen if the problem persists

closed 2024-01-30 by kumekay