espressif/esp-idf #13008: Some problems about using emmc in ESP32

hibfh, 2024-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I tested emmc example in board [ESP32-DevKitC] module ：[ESP32-WROOM-32E] ,but fail.
here my code:

 ```
sdmmc_slot_config_t slot_config ;
sdmmc_host_t host = SDMMC_HOST_DEFAULT();
esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = false,
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };

   host.slot=SDMMC_HOST_SLOT_1;
    host.flags = SDMMC_HOST_FLAG_1BIT;
    host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Other fields will be initialized to zero
    slot_config.cd = SDMMC_SLOT_NO_CD;
    slot_config.wp = SDMMC_SLOT_NO_WP;

    slot_config.width = 1;

    // Set bus IOs
    slot_config.clk = GPIO_NUM_14;
    slot_config.cmd = GPIO_NUM_15;
    slot_config.d0 = GPIO_NUM_2;
    // slot_config.d1 = GPIO_NUM_4;
    // slot_config.d2 = GPIO_NUM_12;
    // slot_config.d3 = GPIO_NUM_13;

    // Enable internal pullups on enabled pins. The internal pullups
    // are insufficient however, please make sure 10k external pullups are
    // connected on the bus. This is for debug / example purpose only.
    slot_config.flags |= SDMMC_SLOT_FLAG_INTERNAL_PULLUP;

    //rst pull up
    gpio_set_level(GPIO_NUM_16, 1);

    ESP_LOGI(TAG, "Mounting filesystem");
    esp_err_t ret = esp_vfs_fat_sdmmc_mount(mount_point, &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                     "If you want the eMMC to be formatted, set the EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the eMMC (%s). "
                     "Make sure eMMC lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return;
    }
    ESP_LOGI(TAG, "Filesystem mounted");

    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);
```

here the log:

```
I (364) eMMC: Initializing eMMC
I (364) eMMC: Using SDMMC peripheral
I (374) eMMC: Mounting filesystem
E (374) sdmmc_periph: sdmmc_host_clock_update_command(150): sdmmc_host_start_command returned 0x105
E (384) sdmmc_periph: disabling clk failed
E (384) sdmmc_periph: sdmmc_host_set_card_clk: sdmmc_host_clock_update_command returned 0x105
E (394) sdmmc_periph: setting probing freq and 1-bit bus failed
E (404) sdmmc_periph: sdmmc_host_init_slot: sdmmc_host_set_card_clk returned 0x105
E (414) vfs_fat_sdmmc: slot init failed (0x105).
E (414) eMMC: Failed to initialize the eMMC (ESP_ERR_NOT_FOUND). Make sure eMMC lines have pull-up resistors in place.
I (434) main_task: Returned from app_main()
```
what may cause the fail? 
I use an  [emmc module](https://wiki.friendlyelec.com/wiki/index.php/Matrix_-_eMMC_Module).chip is KLM8G1GETF-B041
because no 1.8V VCC to use ，so i use 3.3V to be VCC_IO ,and pull up VCC is the same one ,
I 'm not sure that  4K pull-up resistance is suitable  or not, if not ,i 'll change it to 10K.

rrtandler, 2024-01-23:
Hi @hibfh ,
Could you provide details about the pin connection between the devkit and the card adapter ? And also please clarify the meaning of "because no 1.8V VCC to use ，so i use 3.3V to be VCC_IO".

hibfh, 2024-01-24:
@rrtandler I figure out that  1.8V no need，I just change VCC_IO to 3.3V，because 1.8V can not pull up esp32.but fail by the same return。the pin connect as follows：
EMMC                    ESP32
(PIN1) data0      ——  (PIN2)GPIO2
(PIN11)CMD      ——  (PIN15)GPIO15
(PIN12)CLK        ——  (PIN14)GPIO14
(PIN16)VCC_IO  ——  (PIN3V3)3V3
(PIN17)RST        ——  (PIN16)GPIO16
(PIN18)VCC_3V3——  (PIN3V3)3V3
(PIN19)GND       ——  (PINGND)GND

hibfh, 2024-01-24:
@rrtandler 
It seems that EMMC still detect card lead to the NOT_FOUND(0x105) Problem.
I wonder in which situation will [hw->cdetect.cards] become zero? The default value is 2,and seems never change .
My code set slot_config.cd = SDMMC_SLOT_NO_CD,
why it didn't work?
Any Idea?

pacucha42, 2024-01-24:
Hi @hibfh,
we will take a look and let you know asap - please, hold on, as we don't have exact hardware which you use so it may take few days to get something reasonably close.

hibfh, 2024-01-25:
Hello @pacucha42 ,
I wonder if it is necessary to pull up CLK since I was told that no  need to pull up CLK in emmc ，but in [this Document](https://github.com/espressif/esp-idf/blob/master/examples/storage/emmc/README.md) it mentions 10K pull up in PIN clk.

pacucha42, 2024-01-25:
Afaik, you do need to set pull-up for eMMC card as the clock signal is defined as input by the eMMC specs (ie driven by host, in this case ESP32)

hibfh, 2024-01-29:
hey, some good news for u @pacucha42 .this problem has been solved, thanks for your help!

closed 2024-01-29 by hibfh

pacucha42, 2024-01-29:
Nice, thanks for the update! What was the root cause, by the way?

hibfh, 2024-01-29:
Seems related to hardware connect, my colleague adds some pull-up resistor for unused data pin（d1、d2、d3） and so on. And I find that I defined CONFIG_SOC_SDMMC_USE_GPIO_MATRIX which is not suitable and necessary for esp32.

hibfh, 2024-01-29:
@pacucha42 btw, the clk pin connects to the module without pull up, and can still work. it seems that there is no need to pull up CLK.

pacucha42, 2024-01-29:
Thanks for explanation