espressif/esp-idf #13009: PSRAM access runs into wdt timeout

dimonysos, 2024-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My code runs into a wdt timeout in some cases when the external PSRAM is accessed.

`Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4008858c  PS      : 0x00060134  A0      : 0x8008b5d9  A1      : 0x3ffbee70  
0x4008858c: esp_cpu_compare_and_set at /IDF_BUILD/../esp-idf/components/esp_hw_support/cpu.c:464

A2      : 0x5f000000  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060723  
A6      : 0x00000001  A7      : 0x3ffb39c0  A8      : 0x1e827526  A9      : 0xffffff81  
A10     : 0x3ffb39c0  A11     : 0x003fffff  A12     : 0x00000000  A13     : 0x3ffb39c0  
A14     : 0x00000000  A15     : 0x00000af0  SAR     : 0x0000000c  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400835c0  LEND    : 0x400835c8  LCOUNT  : 0x00000027  
0x400835c0: esp_timer_impl_get_counter_reg at /IDF_BUILD/../esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:118

0x400835c8: esp_timer_impl_get_counter_reg at /IDF_BUILD/../esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:128

Backtrace: 0x40088589:0x3ffbee70 0x4008b5d6:0x3ffbeea0 0x40091892:0x3ffbeee0 0x40091900:0x3ffbef10 0x400824cd:0x3ffbef40 0x40082560:0x3ffbef80 0x40092d74:0x3ffbefb0 0x4010d439:0x3ffbefe0 0x400eecc1:0x3ffbf000 0x400ef6bf:0x3ffbf030 0x400efb0d:0x3ffbf070 0x400ea459:0x3ffbf0a0 0x400eace2:0x3ffbf0d0 0x400ec4d5:0x3ffbf110 0x400e0977:0x3ffbf170 0x400d8fc4:0x3ffbf1b0 0x4012160b:0x3ffbf1e0 0x4008b3ab:0x3ffbf210
0x40088589: esp_cpu_compare_and_set at /IDF_BUILD/../esp-idf/components/esp_hw_support/cpu.c:464

0x4008b5d6: spinlock_acquire at /IDF_BUILD/../esp-idf/components/esp_hw_support/include/spinlock.h:130
 (inlined by) xPortEnterCriticalTimeout at /IDF_BUILD/../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:473

0x40091892: xPortEnterCriticalTimeoutSafe at /IDF_BUILD/../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:585
 (inlined by) vPortEnterCriticalSafe at /IDF_BUILD/../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:592
 (inlined by) multi_heap_internal_lock at /IDF_BUILD/../esp-idf/components/heap/multi_heap.c:165

0x40091900: multi_heap_malloc_impl at /IDF_BUILD/../esp-idf/components/heap/multi_heap.c:209

0x400824cd: heap_caps_malloc_base at /IDF_BUILD/../esp-idf/components/heap/heap_caps.c:179

0x40082560: heap_caps_malloc_default at /IDF_BUILD/../esp-idf/components/heap/heap_caps.c:238

0x40092d74: malloc at /IDF_BUILD/../esp-idf/components/newlib/heap.c:24

0x4010d439: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:50
`

And this is the stack dump I am getting.
`
======================== THREADS INFO =========================
  Id   Target Id          Frame 
* 1    process 1073413592 0x4008858c in esp_cpu_compare_and_set (addr=0x5f000000, compare_value=3007315967, new_value=52685) at ../esp-idf/components/esp_hw_support/cpu.c:464
  2    process 1073478060 0x40088725 in esp_cpu_wait_for_intr () at ../esp-idf/components/esp_hw_support/cpu.c:145
  3    process 1073480220 0x40088725 in esp_cpu_wait_for_intr () at ../esp-idf/components/esp_hw_support/cpu.c:145
  4    process 1073529948 0x4008807b in esp_crosscore_int_send (core_id=1, reason_mask=<optimized out>) at ../esp-idf/components/esp_system/crosscore_int.c:168
  5    process 1073484684 0x4008807b in esp_crosscore_int_send (core_id=0, reason_mask=<optimized out>) at ../esp-idf/components/esp_system/crosscore_int.c:168
  6    process 1073609032 0x4000bff0 in _xtos_set_intlevel ()
  7    process 1073504128 0x4000bff0 in _xtos_set_intlevel ()
  8    process 1073410452 0x4000bff0 in _xtos_set_intlevel ()
  9    process 1073413164 0x4000bff0 in _xtos_set_intlevel ()
  10   process 1073610096 0x4000bff0 in _xtos_set_intlevel ()
  11   process 1073500688 0x4000bff0 in _xtos_set_intlevel ()
  12   process 1073411844 0x4000bff0 in _xtos_set_intlevel ()
  13   process 1073509216 0x4000bff0 in _xtos_set_intlevel ()
  14   process 1073513968 0x4000bff0 in _xtos_set_intlevel ()
  15   process 1073517992 0x4000bff0 in _xtos_set_intlevel ()

       TCB             NAME PRIO C/B  STACK USED/FREE
---------- ---------------- -------- ----------------
0x3ffafdd8             main      1/1        1056/7644
0x3ffbf9ac            IDLE0      0/0         480/1304
[... 99 lines cut ...]
#0  0x4000bff0 in _xtos_set_intlevel ()
#1  0x4008b6dd in vPortClearInterruptMaskFromISR (prev_level=<optimized out>) at ../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:564
#2  vPortExitCritical (mux=0x3ffb00b8 <xKernelLock>) at ../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:504
#3  0x4008e941 in xTaskGenericNotifyWait (uxIndexToWait=<optimized out>, ulBitsToClearOnEntry=<optimized out>, ulBitsToClearOnExit=4294967295, pulNotificationValue=0x3ffaf638, xTicksToWait=4294967295) at ../esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5769
#4  0x40088158 in ipc_task (arg=<optimized out>) at ../esp-idf/components/esp_system/esp_ipc.c:58
#5  0x4008b3ae in vPortTaskWrapper (pxCode=0x4008810c <ipc_task>, pvParameters=0x1) at ../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

==================== THREAD 13 (TCB: 0x3ffc7360, name: 'Tsk5') =====================
#0  0x4000bff0 in _xtos_set_intlevel ()
#1  0x4008b6dd in vPortClearInterruptMaskFromISR (prev_level=<optimized out>) at ../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:564
#2  vPortExitCritical (mux=0x3ffc62f0) at ../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:504
#3  0x4008b0ad in xQueueSemaphoreTake (xQueue=0x3ffc629c, xTicksToWait=<optimized out>) at ../esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1853
...
==================== THREAD 14 (TCB: 0x3ffc85f0, name: 'Tsk6') =====================
#0  0x4000bff0 in _xtos_set_intlevel ()
#1  0x4008b6dd in vPortClearInterruptMaskFromISR (prev_level=<optimized out>) at ../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:564
#2  vPortExitCritical (mux=0x3ffc7520) at ../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:504
#3  0x4008b0ad in xQueueSemaphoreTake (xQueue=0x3ffc74cc, xTicksToWait=<optimized out>) at ../esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1853
...
==================== THREAD 15 (TCB: 0x3ffc95a8, name: 'Tsk7') =====================
#0  0x4000bff0 in _xtos_set_intlevel ()
#1  0x4008b6dd in vPortClearInterruptMaskFromISR (prev_level=<optimized out>) at ../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:564
#2  vPortExitCritical (mux=0x3ffc8870) at ../esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:504
#3  0x4008afab in xQueueReceive (xQueue=0x3ffc881c, pvBuffer=0x3ffc9488, xTicksToWait=<optimized out>) at ../esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1632
...`

dimonysos, 2024-01-19:
To clarify I am using the ESP IDF v5.2 beta2 (tag).
In some cases I seem to get a LoadProhibited as well, when trying to access SPIRAM

`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40091f32  PS      : 0x00060f33  A0      : 0x800919e2  A1      : 0x3ffbee00  
0x40091f32: search_suitable_block at /IDF_BUILD/../esp-idf/components/heap/tlsf/tlsf.c:302
 (inlined by) block_locate_free at /IDF_BUILD/../esp-idf/components/heap/tlsf/tlsf.c:560
 (inlined by) tlsf_malloc at /IDF_BUILD/../esp-idf/components/heap/tlsf/tlsf.c:1005

A2      : 0x3f800014  A3      : 0x00002000  A4      : 0x3ffae884  A5      : 0x3ffbc370  
A6      : 0x3f800014  A7      : 0x3ffb39d0  A8      : 0x00000000  A9      : 0x00002000  
A10     : 0x00000007  A11     : 0xffffffff  A12     : 0xffffffff  A13     : 0xffffffff  
A14     : 0x0000001b  A15     : 0x0000cdcd  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001b  LBEG    : 0x400f85a6  LEND    : 0x400f85aa  LCOUNT  : 0x00000000  

Backtrace: 0x40091f2f:0x3ffbee00 0x400919df:0x3ffbee30 0x400824a9:0x3ffbee60 0x400824cc:0x3ffbeea0 0x400f6b04:0x3ffbeed0 0x400f85cc:0x3ffbef00 0x400f75f5:0x3ffbefe0 0x400efeef:0x3ffbf040 0x400f054b:0x3ffbf080 0x400e450b:0x3ffbf110 0x400e050b:0x3ffbf190 0x400e069a:0x3ffbf1c0 0x400e07c2:0x3ffbf1f0 0x400d8e64:0x3ffbf230 0x4012f523:0x3ffbf260 0x4008b483:0x3ffbf290
0x40091f2f: search_suitable_block at /IDF_BUILD/../esp-idf/components/heap/tlsf/tlsf.c:302
 (inlined by) block_locate_free at /IDF_BUILD/../esp-idf/components/heap/tlsf/tlsf.c:560
 (inlined by) tlsf_malloc at /IDF_BUILD/../esp-idf/components/heap/tlsf/tlsf.c:1005

0x400919df: multi_heap_malloc_impl at /IDF_BUILD/../esp-idf/components/heap/multi_heap.c:210

0x400824a9: heap_caps_malloc_base at /IDF_BUILD/../esp-idf/components/heap/heap_caps.c:179

0x400824cc: heap_caps_malloc at /IDF_BUILD/../esp-idf/components/heap/heap_caps.c:202
`

igrr, 2024-01-24:
On most ESP32 modules, GPIOs 16 and 17 are used to connect PSRAM.

Quoting the [ESP32-WROVER-E datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-wrover-e_esp32-wrover-ie_datasheet_en.pdf):

> External connections can be made to any GPIO except for GPIOs in the range 6-11, 16, or 17. GPIOs 6-11 are connected
to the module’s integrated SPI flash. GPIOs 16 and 17 are connected to the module’s integrated PSRAM. For details,
please see Section 7 Schematics.

When UART takes over GPIO16 and 17, PSRAM can no longer be accessed correctly.

dimonysos, 2024-01-24:
This is interessting. Actually I am porting code from old ESP-IDF v3.3.6 release. The PSRAM was used with the exact same board on that ESP-IDF without any issue and the UART1 was used in parallel with GPIO16 and 17 which also worked.

The exact same module that I use is the ESP32-WROVER-E (ESP32-D0WD-V3). Why was it working with ESP-IDF v3.3.6? Is it possible for me to adapt some parts of the ESP-IDF v5.2 to actually achieve the same results and if that is possible maybe you could give me a hint where to look for in the sources (probably some parts in components/esp_psram).

I am not able to change the HW as I need to keep the backwards compatibility for older boards. Therefor the solution has to be made in SW and if that is not possible that would mean I would have to remain on the old ESP-IDF version.

igrr, 2024-01-24:
I am not sure, perhaps the application on 3.3.6 didn't enable PSRAM at all? You can try scoping the signals on GPIO16 and 17 to see if there was any activity prior to UART being initialized. I don't have another explanation why this would work on an older version...

dimonysos, 2024-01-25:
No, I am very sure that both PSRAM 8MB (of which only the low banks 4MB were used) and the UART1 on GPIO16 and 17 are working in parallel in v3.3.6. Both are used in the application almost all of the time and very frequently. There must be some change in the v5.2 (and I also tested v5.1.2) which breaks this functionality. Besides the ESP-IDF, some driver adaptions such as spi/i2c and the build system the rest of the application remained the same.

Well thank you for your support so far. At least now I have a clue what is the cause of those crashes. I just tried the GPIOs 5 and 4 by calling the following:
uart_set_pin((uart_port_t)1, 5, 4, -1, -1);
Using those as the UART output does indeed not cause any crash with the PSRAM or an interrupt wdt timeout. Unfortunately I can't use those GPIOs as some other IC is already routed on those and I also have to keep compatibility for older boards. I will try to figure out what change in the PSRAM source component might be causing this. If I figure something out I will add it here. Let me know if you might think of something that could help me in the search.

closed 2024-01-25 by dimonysos