espressif/esp-idf #13011: Bluedroid SPP acceptor: `esp_spp_write`-en data comes back as `ESP_SPP_DATA_IND_EVT` after client closes serial port

LaXiS96, 2024-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2 release

### Espressif SoC revision.

ESP32-D0WDQ6 (revision v1.0)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC V4

### Power Supply used.

USB

### What is the expected behavior?

[See implementation here](https://github.com/LaXiS96/esp32-obd2/blob/f83b683084a7f1c2eaf316d761428c5ec7872a3a/main/bt.c) (my repo).
With the ESP32 accepting Bluetooth SPP connections, a Linux client connects to the RFCOMM serial port, opens it and then closes it without disconnecting from the ESP32.
The expected behavior is that `ESP_SPP_DATA_IND_EVT` events should only be received when the connected client has the serial port open and is actually sending data.

### What is the actual behavior?

What happens instead is that after the serial port is closed, `ESP_SPP_DATA_IND_EVT` events are notified with the same data that is concurrently being written with `esp_spp_write` from another task.
See the log below (`BT` tag):
- 14-byte `t55540D0E0A0D\r` messages are written to SPP every 2 seconds (after receiving them from CAN through SLCAN task)
- at 39691ms the client closes the serial port, no SPP "close" event is received but `ESP_SPP_DATA_IND_EVT` events start coming in, with the same data that was just written (see for example at 39831ms, the SLCAN command value is the same that was sent above)

### Steps to reproduce.

1. Connect to ESP32 Bluetooth serial port from PC (on Linux this creates the `/dev/rfcomm0` device)
2. Open serial port on PC with preferred software (e.g. `tio` or CuteCom)
3. (In my project's case, send configuration commands to start receiving data: `S6\r` and `O\r`)
4. Close serial port on PC (e.g. by closing the terminal or GUI), but do not disconnect the Bluetooth device
5. If the client opens the serial port again, everything returns to normal; if it closes it again, the issue occurs again

### Debug Logs.

```plain
--- esp-idf-monitor 1.3.4 on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7084
ho 0 tail 12 room 4
load:0x40078000,len:15584
load:0x40080400,len:4
[... 185 lines cut ...]
I (41741) SLCAN: serial received bytes:1
I (41751) SLCAN: serial received bytes:1
I (41741) BT: ESP_SPP_DATA_IND_EVT length:1
I (41761) SLCAN: serial received bytes:1
I (41761) BT: ESP_SPP_DATA_IND_EVT length:1
I (41771) SLCAN: serial received bytes:1
I (41771) BT: ESP_SPP_DATA_IND_EVT length:1
I (41771) SLCAN: serial received bytes:1
I (41811) BT: ESP_SPP_DATA_IND_EVT length:1
I (41811) SLCAN: serial received bytes:1
I (41811) BT: ESP_SPP_DATA_IND_EVT length:1
I (41811) SLCAN: serial received bytes:1
I (41811) BT: ESP_SPP_DATA_IND_EVT length:1
I (41821) SLCAN: serial received bytes:1
I (41821) BT: ESP_SPP_DATA_IND_EVT length:2
I (41831) SLCAN: serial received bytes:2
I (41831) SLCAN: command "^Gt55540D0E0A0D"
I (41841) SLCAN: respond error
I (41841) SLCAN: serial transmit bytes:1
I (41851) BT: write messages:1 bytes:1
I (41871) BT: ESP_SPP_DATA_IND_EVT length:1
I (41871) SLCAN: serial received bytes:1
I (41871) BT: ESP_SPP_DATA_IND_EVT length:1
I (41871) SLCAN: serial received bytes:1
```

boblane1, 2024-01-19:
Hi @LaXiS96,

It is unlikely that no ESP_SPP_CLOSE_EVT event comes when the client closes the serial port.  I have tried with tio or cutecom, but neither could reproduce the issue. Could you please simplify your project to reproduce the issue and detail the sending behavior on client side?

Thanks.

LaXiS96, 2024-01-19:
I can reproduce it outside my original project, but only on my Linux machine and not on Windows.

Based on code and logs below:
- at `3766` the Linux client connects via Bluetooth and written data starts looping back to `DATA_IND` events already
- at `7226` I open the rfcomm0 serial port in CuteCom
- at `15226` I close the connection in CuteCom

I am noticing that the `DATA_IND` data contains the entire test message plus an additional `^J` (which is the terminal representation of `\n`), it looks like the Linux client is actively sending back the data?

On Windows I have no such issue, closing the serial port logs this as expected:
```plain
W (47635) BT_RFCOMM: port_rfc_closed RFCOMM connection in server:1 state 2 closed: Closed (res: 19)
I (47645) spp: ESP_SPP_CLOSE_EVT
W (47655) BT_RFCOMM: rfc_find_lcid_mcb LCID reused LCID:0x40 current:0x0
W (47655) BT_RFCOMM: RFCOMM_DisconnectInd LCID:0x40
```

<details>
<summary>sdkconfig.defaults</summary>

```plain
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=y
CONFIG_BT_BLE_ENABLED=n
CONFIG_BT_CLASSIC_ENABLED=y
CONFIG_BT_SPP_ENABLED=y
CONFIG_BT_SSP_ENABLED=n
```
</details>

<details>
<summary>main.c</summary>

```c
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "esp_bt.h"
#include "esp_bt_main.h"
#include "esp_bt_device.h"
#include "esp_gap_bt_api.h"
#include "esp_spp_api.h"

static uint32_t sppHandle = 0;

static char *bda2str(uint8_t *bda, char *str, size_t size)
{
    if (bda == NULL || str == NULL || size < 18)
        return NULL;

    uint8_t *p = bda;
    sprintf(str, "%02x:%02x:%02x:%02x:%02x:%02x", p[0], p[1], p[2], p[3], p[4], p[5]);
    return str;
}

static void gapCallback(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
{
    switch (event)
    {
    case ESP_BT_GAP_AUTH_CMPL_EVT:
        ESP_LOGI("gap", "ESP_BT_GAP_AUTH_CMPL_EVT");
        if (param->auth_cmpl.stat == ESP_BT_STATUS_SUCCESS)
        {
            ESP_LOGI("gap", "authentication success: %s", param->auth_cmpl.device_name);
            esp_log_buffer_hex("gap", param->auth_cmpl.bda, ESP_BD_ADDR_LEN);
        }
[... 484 lines cut ...]
I (22266) spp: 0x3ffcfdc0   31                                                |1|
I (22266) spp: ESP_SPP_DATA_IND_EVT length:1
I (22276) spp: 0x3ffcfddc   32                                                |2|
I (22286) spp: ESP_SPP_DATA_IND_EVT length:1
I (22286) spp: 0x3ffcfdf8   33                                                |3|
I (22296) spp: ESP_SPP_DATA_IND_EVT length:1
I (22296) spp: 0x3ffcfe9c   34                                                |4|
I (22306) spp: ESP_SPP_DATA_IND_EVT length:1
I (22316) spp: 0x3ffcfeb8   35                                                |5|
I (22326) spp: ESP_SPP_DATA_IND_EVT length:1
I (22326) spp: 0x3ffcff7c   36                                                |6|
I (22336) spp: ESP_SPP_DATA_IND_EVT length:1
I (22336) spp: 0x3ffcff98   37                                                |7|
I (22346) spp: ESP_SPP_DATA_IND_EVT length:1
I (22356) spp: 0x3ffcffb4   38                                                |8|
I (22366) spp: ESP_SPP_DATA_IND_EVT length:1
I (22366) spp: 0x3ffcffd0   39                                                |9|
I (22406) spp: ESP_SPP_DATA_IND_EVT length:2
I (22406) spp: 0x3ffcfda4   0d 0a                                             |..|
I (22406) spp: ESP_SPP_DATA_IND_EVT length:1
I (22416) spp: 0x3ffcfdc0   5e                                                |^|
I (22416) spp: ESP_SPP_DATA_IND_EVT length:1
I (22426) spp: 0x3ffcfddc   4a                                                |J|
```
</details>

LaXiS96, 2024-01-21:
Huh, that was exactly it! I was using the `blueman` GUI to connect to the ESP32, and it looks like it does `connect` instead of `bind`.
I'll use the `rfcomm` CLI from now on; I don't understand why the `connect` implementation loops back data but I guess that's not a problem anymore.
Thank you for your help @boblane1

closed 2024-01-21 by LaXiS96