espressif/esp-idf #13014: bugreport: httpd_uri_match_wildcard to use http_server wildcard uri in combination with ws_server

onyx99, 2024-01-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-espidf @ 3.50102.0 (5.1.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

using http_server together with ws_server

### What is the actual behavior?

when i try to connect to ws://hostip/ws i got in the server log
(38123) httpd_parse: parse_block: request URI/header too long[0m
(38123) httpd_txrx: httpd_resp_send_err: 431 Request Header Fields Too Large - Header fields are too long[0m

and (so) :ws_handler(httpd_req_t *req) is not being called (got nothing from there in the log)

possible bug report using http_server with ws_server

when i request http://hostip/ http://hostip/test etc, i got a correct response
Hello, World!

### Steps to reproduce.

the problem occurs when enabling httpd_uri_match_wildcard and therefore probably somewhere in the handling of httpd_uri_match_wildcard
because when i remove that and move/change the http handler to 404 handler then websocket works together with http
but then i got ofcourse a uri not found error in the logs first before the http content is sent to the client.

below is my used code.

/*
tested with:
framework-espidf @ 3.50102.0 (5.1.2)
ws_server + http_server example
*/

//WebSocketServer.h
#ifndef WEBSOCKET_SERVER_H
#define WEBSOCKET_SERVER_H

#include <esp_http_server.h>
#include <string>

class WebSocketServer {
public:
    WebSocketServer();
    ~WebSocketServer();

    void start();
    void stop();

private:
    static httpd_handle_t server;

    // HTTP handlers
    static esp_err_t http_handler(httpd_req_t *req);
    static esp_err_t http_handler_static(httpd_req_t *req);
 
    // Web server functions
    static httpd_handle_t startWebServer();
    static esp_err_t stopWebServer(httpd_handle_t server);
 

    //websocket functions
    static esp_err_t ws_handler(httpd_req_t *req);
    static esp_err_t ws_handler_static(httpd_req_t *req);
    static esp_err_t trigger_async_send(httpd_handle_t handle, httpd_req_t *req);
    static void ws_async_send(void *arg);

    //static bool WebSocketServer::httpd_uri_match_wildcard(const char *uri_template, const char *uri_to_match, size_t match_upto);

};

#endif  // WEBSOCKET_SERVER_H

//WebSocketServer.cpp
#include "WebSocketServer.h"

#include <esp_http_server.h>
#include <esp_log.h>

#include <string>

struct async_resp_arg {
  httpd_handle_t hd;
  int fd;
};

const char* TAG = "http_server";
httpd_handle_t WebSocketServer::server;

WebSocketServer::WebSocketServer() {
}

WebSocketServer::~WebSocketServer() {
    stop();
}

void ws_async_send(void *arg)
{
    static const char * data = "Async data";
    struct async_resp_arg *resp_arg = (async_resp_arg*)arg;
    httpd_handle_t hd = resp_arg->hd;
    int fd = resp_arg->fd;
    httpd_ws_frame_t ws_pkt;
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    ws_pkt.payload = (uint8_t*)data;
    ws_pkt.len = strlen(data);
    ws_pkt.type = HTTPD_WS_TYPE_TEXT;

    httpd_ws_send_frame_async(hd, fd, &ws_pkt);
    free(resp_arg);
}

esp_err_t WebSocketServer::trigger_async_send(httpd_handle_t handle, httpd_req_t *req)
{
   // Now you can use the complete definition of async_resp_arg

struct async_resp_arg *resp_arg = (struct async_resp_arg *)malloc(sizeof(struct async_resp_arg));

    if (resp_arg == NULL) {
        return ESP_ERR_NO_MEM;
    }
    resp_arg->hd = req->handle;
    resp_arg->fd = httpd_req_to_sockfd(req);
    esp_err_t ret = httpd_queue_work(handle, ws_async_send, resp_arg);
    if (ret != ESP_OK) {
        free(resp_arg);
    }
    return ret;

}

esp_err_t WebSocketServer::ws_handler(httpd_req_t *req) {
   
     ESP_LOGI(TAG, "ws request");

   if (req->method == HTTP_GET) {
        ESP_LOGI(TAG, "Handshake done, the new connection was opened");
        return ESP_OK;
    }
    httpd_ws_frame_t ws_pkt;
    uint8_t *buf = NULL;
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    ws_pkt.type = HTTPD_WS_TYPE_TEXT;
    /* Set max_len = 0 to get the frame len */
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "httpd_ws_recv_frame failed to get frame len with %d", ret);
        return ret;
    }
    ESP_LOGI(TAG, "frame len is %d", ws_pkt.len);
    if (ws_pkt.len) {
        /* ws_pkt.len + 1 is for NULL termination as we are expecting a string */
        buf = (uint8_t*)calloc(1, ws_pkt.len + 1);
        if (buf == NULL) {
            ESP_LOGE(TAG, "Failed to calloc memory for buf");
            return ESP_ERR_NO_MEM;
        }
        ws_pkt.payload = buf;
        /* Set max_len = ws_pkt.len to get the frame payload */
        ret = httpd_ws_recv_frame(req, &ws_pkt, ws_pkt.len);
        if (ret != ESP_OK) {
            ESP_LOGE(TAG, "httpd_ws_recv_frame failed with %d", ret);
            free(buf);
            return ret;
        }
        ESP_LOGI(TAG, "Got packet with message: %s", ws_pkt.payload);
    }
    ESP_LOGI(TAG, "Packet type: %d", ws_pkt.type);
    if (ws_pkt.type == HTTPD_WS_TYPE_TEXT &&
        strcmp((char*)ws_pkt.payload,"Trigger async") == 0) {
        free(buf);
       // return trigger_async_send(req->handle, req);
    }

    ret = httpd_ws_send_frame(req, &ws_pkt);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "httpd_ws_send_frame failed with %d", ret);
    }
    free(buf);
    return ret;
}

esp_err_t WebSocketServer::http_handler(httpd_req_t *req) {
    std::string path(req->uri);

    if (path == "/") { // default page
        path = "/index.html";
    }

    std::string filePath = path.substr(1);

    // TODO: Implement logic to read the file and send its contents in the response

    // For now, let's just send a simple "Hello, World!" response
    const char* response_data = "Hello, World!";
    httpd_resp_send(req, response_data, strlen(response_data));
    return ESP_OK;
}

void WebSocketServer::start() {
    
    ESP_LOGI(TAG, "Start webserver");
    server = startWebServer();
    if (server) {
        ESP_LOGI(TAG, "Started webserver");
    } else {
        ESP_LOGE(TAG, "Failed to start web server");
    }
}

void WebSocketServer::stop() {
    if (server) {
        ESP_LOGI(TAG, "Stopping webserver");
        if (stopWebServer(server) == ESP_OK) {
            server = nullptr;
            ESP_LOGI(TAG, "Webserver stopped successfully");
        } else {
            ESP_LOGE(TAG, "Failed to stop web server");
        }
    }
}

/*
bool WebSocketServer::httpd_uri_match_wildcard(const char *uri_template, const char *uri_to_match, size_t match_upto) {
}
*/

httpd_handle_t WebSocketServer::startWebServer() {
    httpd_config_t config = HTTPD_DEFAULT_CONFIG();
    config.uri_match_fn = httpd_uri_match_wildcard; //accept wildcards as url

    ESP_LOGI(TAG, "Starting server on port: '%d'", config.server_port);

    if (httpd_start(&server, &config) == ESP_OK) {
        ESP_LOGI(TAG, "Registering URI handlers");

       
        httpd_uri_t ws = {
            .uri        = "/ws",
            .method     = HTTP_GET,
            .handler    = &ws_handler_static,
            .user_ctx   = NULL,
            .is_websocket = true,               // Mandatory: set to `true` to handler websocket protocol
            .handle_ws_control_frames = false,  // Optional: set to `true` for the handler to receive control packets, too
            .supported_subprotocol = "chat",    // Optional: set supported subprotocol for this handler
        };
        httpd_register_uri_handler(server, &ws);
           
        httpd_uri_t http = {
            .uri       = "/*",
            .method    = HTTP_GET,
            .handler   = &http_handler_static,
            .user_ctx  = NULL,
            .is_websocket = false,  
            .handle_ws_control_frames = false,
            .supported_subprotocol = "",
        };
        httpd_register_uri_handler(server, &http);

        return server;
    } else {
        ESP_LOGE(TAG, "Error starting server!");
        return nullptr;
    }
}

esp_err_t WebSocketServer::stopWebServer(httpd_handle_t server) {
    return httpd_stop(server);
}

// Static member function for HTTP handler
esp_err_t WebSocketServer::http_handler_static(httpd_req_t *req) {
    // Access instance variables using: WebSocketServer::server
    return WebSocketServer::http_handler(req);
}

// Static member function for ws handler
esp_err_t WebSocketServer::ws_handler_static(httpd_req_t *req) {
    // Access instance variables using: WebSocketServer::server
    return WebSocketServer::ws_handler(req);
}

onyx99, 2024-01-19:
Here is the full debug log, noticing:
-> D (223648) httpd_parse: httpd_req_new: New request, has WS? No, sd->ws_handler valid? No, sd->ws_close? No

D (223608) httpd: httpd_server: processing listen socket 54[0m
D (223608) httpd: httpd_accept_conn: newfd = 57[0m
D (223608) httpd_sess: httpd_sess_new: fd = 57[0m
D (223608) httpd_sess: httpd_sess_new: active sockets: 1[0m
D (223618) httpd: httpd_accept_conn: complete[0m
D (223618) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:2 posted to loop 0x3ffbcb04[0m
D (223628) httpd: httpd_server: doing select maxfd+1 = 58[0m
D (223638) httpd: httpd_process_session: processing socket 57[0m
D (223638) httpd_sess: httpd_sess_process: httpd_req_new[0m
D (223648) httpd_parse: httpd_req_new: New request, has WS? No, sd->ws_handler valid? No, sd->ws_close? No[0m
D (223658) httpd_txrx: httpd_recv_with_opt: requested length = 128[0m
D (223658) httpd_txrx: httpd_recv_with_opt: received length = 128[0m
D (223668) httpd_parse: read_block: received HTTP request block size = 128[0m
D (223678) httpd_parse: cb_url: message begin[0m
D (223678) httpd_parse: cb_url: processing url = /ws[0m
D (223688) httpd_parse: verify_url: received URI = /ws[0m
D (223688) httpd_parse: cb_header_field: headers begin[0m
D (223698) httpd_txrx: httpd_unrecv: length = 110[0m
D (223698) httpd_parse: pause_parsing: paused[0m
D (223698) httpd_parse: cb_header_field: processing field = Host[0m
D (223708) httpd_txrx: httpd_recv_with_opt: requested length = 128[0m
D (223718) httpd_txrx: httpd_recv_with_opt: pending length = 110[0m
D (223718) httpd_parse: read_block: received HTTP request block size = 110[0m
D (223728) httpd_parse: continue_parsing: skip pre-parsed data of size = 5[0m
D (223738) httpd_parse: continue_parsing: un-paused[0m
D (223738) httpd_parse: cb_header_value: processing value = 192.168.2.15[0m
D (223748) httpd_parse: cb_header_field: processing field = Connection[0m
D (223758) httpd_parse: cb_header_value: processing value = Upgrade[0m
D (223758) httpd_parse: cb_header_field: processing field = Pragma[0m
D (223768) httpd_parse: cb_header_value: processing value = no-cache[0m
D (223768) httpd_parse: cb_header_field: processing field = Cache-Control[0m
D (223778) httpd_parse: cb_header_value: processing value = no-cache[0m
D (223788) httpd_parse: cb_header_field: processing field = User-Agent[0m
D (223788) httpd_parse: cb_header_value: processing value = Mozilla/5.0 (W[0m
D (223798) httpd_parse: parse_block: parsed block size = 110[0m
D (223808) httpd_txrx: httpd_recv_with_opt: requested length = 128[0m
D (223808) httpd_txrx: httpd_recv_with_opt: received length = 128[0m
D (223818) httpd_parse: read_block: received HTTP request block size = 128[0m
D (223828) httpd_parse: cb_header_value: processing value = indows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/120.0.0.0 Safari/537.36[0m
D (223838) httpd_parse: cb_header_field: processing field = Upgrade[0m
D (223848) httpd_parse: cb_header_value: processing value = websocket[0m
D (223848) httpd_parse: cb_header_field: processing field = Origin[0m
D (223858) httpd_parse: cb_header_value: processing value = c[0m
D (223868) httpd_parse: parse_block: parsed block size = 238[0m
D (223868) httpd_txrx: httpd_recv_with_opt: requested length = 128[0m
D (223878) httpd_txrx: httpd_recv_with_opt: received length = 128[0m
D (223878) httpd_parse: read_block: received HTTP request block size = 128[0m
D (223888) httpd_parse: cb_header_value: processing value = hrome-extension://pfdhoblngboilpfeibdedpjgfnlcodoo[0m
D (223898) httpd_parse: cb_header_field: processing field = Sec-WebSocket-Version[0m
D (223908) httpd_parse: cb_header_value: processing value = 13[0m
D (223908) httpd_parse: cb_header_field: processing field = Accept-Encoding[0m
D (223918) httpd_parse: cb_header_value: processing value = gzip, deflate[0m
D (223928) httpd_parse: cb_header_field: processing field = Accept-Language[0m
D (223928) httpd_parse: parse_block: parsed block size = 366[0m
D (223938) httpd_txrx: httpd_recv_with_opt: requested length = 128[0m
D (223948) httpd_txrx: httpd_recv_with_opt: received length = 128[0m
D (223948) httpd_parse: read_block: received HTTP request block size = 128[0m
D (223958) httpd_parse: cb_header_value: processing value = nl-NL,nl;q=0.9,en-US;q=0.8,en;q=0.7[0m
D (223968) httpd_parse: cb_header_field: processing field = Sec-WebSocket-Key[0m
D (223978) httpd_parse: cb_header_value: processing value = FHX5uhAsmWxChc65RZ9Qag==[0m
D (223978) httpd_parse: cb_header_field: processing field = Sec-WebSocket-Extensions[0m
D (223988) httpd_parse: cb_header_value: processing value = permessage-deflate; [0m
D (223998) httpd_parse: parse_block: parsed block size = 494[0m
D (223998) httpd_txrx: httpd_recv_with_opt: requested length = 19[0m
D (224008) httpd_txrx: httpd_recv_with_opt: received length = 19[0m
D (224018) httpd_parse: read_block: received HTTP request block size = 19[0m
D (224018) httpd_parse: cb_header_value: processing value = client_max_window_b[0m
D (224028) httpd_parse: parse_block: parsed block size = 513[0m
[0;33mW (224038) httpd_parse: parse_block: request URI/header too long[0m
[0;33mW (224038) httpd_txrx: httpd_resp_send_err: 431 Request Header Fields Too Large - Header fields are too long[0m
D (224058) httpd_txrx: httpd_send_all: sent = 91[0m
D (224058) httpd_txrx: httpd_send_all: sent = 2[0m
[... cut ...]
D (224068) httpd_txrx: httpd_send_all: sent = 26[0m
D (224078) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3ffbcb04[0m
[0;31mE (224088) httpd_txrx: httpd_resp_send_err: error calling setsockopt : 22[0m
D (224098) httpd_sess: httpd_sess_delete: fd = 57[0m
D (224098) httpd_sess: httpd_sess_delete: active sockets: 0[0m
D (224098) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:7 posted to loop 0x3ffbcb04[0m
D (224108) httpd: httpd_server: doing select maxfd+1 = 56[0m

ESP-YJM, 2024-01-22:
@onyx99 Could you please follow this guide https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/simple#troubleshooting and retry?

ESP-YJM, 2024-03-06:
@onyx99 Any update for this issue? I am closing this issue. If you have, feel free to open again. Thanks.

closed 2024-03-06 by ESP-YJM